The-OpenROAD-Project/OpenROAD #1655: Routing congestion too high error

vijayank88, 2022-02-21:
I'm using OpenLane Tag: [2022.02.21_02.33.47](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.02.21_02.33.47)
wb_interconnect is mpw-3 shuttle precheck passed design.

If I use same design in current OR App, failed at global routing. Log as follows:
```
[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal     178080         70200          60.58%
met2       Vertical       133560         94466          29.27%
met3       Horizontal      89040         60345          32.23%
met4       Vertical        53424         33949          36.45%
met5       Horizontal      17808          7749          56.49%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 13726
[INFO GRT-0198] Via related Steiner nodes: 669
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 31730
[INFO GRT-0112] Final usage 3D: 205552
[ERROR GRT-0118] Routing congestion too high.
Error: groute.tcl, 69 GRT-0118
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

```
Issue reproducible tar file attached here. 

Expectation: Tool should complete routing stage

eder-matheus, 2022-03-02:
@vijayank88 Similar to https://github.com/The-OpenROAD-Project/OpenROAD/issues/1620, I was able to complete the global routing with no errors by reducing the global adjustment (`GLB_RT_ADJUSTMENT`) from `0.3` to `0.1`. What is the placement density of this design?

mousaq92, 2022-03-03:
Hello,

(GLB_RT_ADJUSTMENT) "0.1" does pass global routing. However, all I see are filler cells in the design. Please correct me if I am mistaken.

[image]

eder-matheus, 2022-03-03:
@mousaq92 In the `Display Control`, look at `Instances` and check what is selected to be shown. You probably have to add the "StdCells" option. Check the figure below:
[image]

mousaq92, 2022-03-03:
Got it thanks.

vijayank88, 2022-03-03:
@eder-matheus 
By reducing the global adjustment (GLB_RT_ADJUSTMENT) from 0.3 to 0.1 resolved congestion issue.

closed 2022-03-03 by vijayank88