The-OpenROAD-Project/OpenROAD #1659: openroad python bindings should cover sta

proppy, 2022-02-21:
Currently openroad python bindings only cover the `odb` module, it would be nice to be able to also access sta which seems to be already distributed as a standalone library:
https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/CMakeLists.txt#L436

With existing SWIG binding:
https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/app/StaApp.i

maliberty, 2022-02-22:
There is significant code in TCL around those binding (https://github.com/The-OpenROAD-Project/OpenSTA/tree/master/tcl).  It isn't as helpful as odb to expose the raw interface.  I think this falls under the general heading of dealing with meaningful python bindings.

proppy, 2022-02-22:
Is the API surface covered by the SWIG binding a subset of what's exposed thru `libOpenSTA`?
https://github.com/The-OpenROAD-Project/OpenSTA/tree/master/include/sta

maliberty, 2022-02-22:
I haven't check but most likely it is

maliberty, 2022-05-10:
Replaced by #1812

closed 2022-05-10 by maliberty