The-OpenROAD-Project/OpenROAD #1677: DRT fails with checkConnectivity break error

antonblanchard, 2022-02-28:
I'm seeing the following error during DRT of the attached design. Interestingly the error mentions layer 6 (`met5`), which should be disabled with `-top_routing_layer met4`. I don't see any met5 in the global route guides, so I presume it isn't coming from there.

Test case: [connectivity-issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8149727/connectivity-issue.tar.gz)

Just in case, I added the files required to redo global routing:

```
# GRT
openroad groute.tcl
# DRT
./run.sh
```

Note that this error appeared in the middle of the ECO flow, so there is a chance the buffer fixup code has contributed to this.

```
Error: booth_b52_m48 1 pin not visited #guides = 12
Error: checkConnectivity break, net booth_b52_m48
Objs not visited:
frPathSeg: begin (529690 276250 ) end ( 530150 276250 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (528770 276930 ) end ( 535210 276930 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (530150 276000 ) end ( 530150 276250 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (529690 276000 ) end ( 529690 276250 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (528770 276000 ) end ( 528770 276930 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (534290 263330 ) end ( 535210 263330 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (528770 276000 ) end ( 529690 276000 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (529690 276000 ) end ( 530150 276000 ) layerNum 4
	beginStyle: 1
	endStyle: 1
frPathSeg: begin (535210 263330 ) end ( 535210 276930 ) layerNum 6
	beginStyle: 1
	endStyle: 1
frVia: at ( 535210 276930 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
frVia: at ( 534290 263330 )
VIA DEF:
VIA L1M1_PR DEFAULT
  RECT -85 -85 85 85
  RECT -85 -85 85 85
  RECT -145 -115 145 115
frVia: at ( 535210 263330 )
VIA DEF:
VIA M1M2_PR DEFAULT
  RECT -160 -130 160 130
  RECT -75 -75 75 75
  RECT -130 -160 130 160
INSTTERM: (INST/CELL/TERM/NET) U$$3665 sky130_fd_sc_hd__xor2_1 X booth_b52_m48
```

Stephanommg, 2022-03-30:
Please test with this branch: https://github.com/The-OpenROAD-Project-private/OpenROAD/tree/secure-TR_APsWithShapes

I made a fix for other connectivity error so perhaps their cause was the same.

maliberty, 2022-03-30:
@Stephanommg he will not have access to the private repo.   If you make a PR and label it so sync public that will copy over the branch to the staging repo where he can see it.

maliberty, 2022-03-30:
@antonblanchard you can see it at The-OpenROAD-Project-staging:secure-TR_APsWithShapes

Stephanommg, 2022-03-30:
Link:  https://github.com/The-OpenROAD-Project-staging/OpenROAD/tree/secure-TR_APsWithShapes

mousaq92, 2022-03-31:
How can I checkout this branch? I can try to check this if that's ok. Do I need to re-install OpenROAD? Thanks.

Stephanommg, 2022-04-12:
Please check the master branch and report back if its solved.

antonblanchard, 2022-04-12:
It's still reproducing as of commit be66da7a5d3c18a69683b72e691c43d55d3e2fe7

Stephanommg, 2022-04-14:
FYI, I already am aware of what the problem is and I will be working on solving it in the next week.

Stephanommg, 2022-04-27:
FYI, the problem is solved, but it is still going through some tests.

closed 2022-05-13 by Stephanommg