The-OpenROAD-Project/OpenROAD #1684: PDN generation fails, unclear why

growly, 2022-03-04:
I'm running openroad through openlane (tag: 2022.02.08_01.48.52). The openroad version seems to be:

```
OpenLane Container (1860ec4):/openlane$ openroad --version
OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf
```

I am passing in an experimental DEF ([test.def.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8189108/test.def.zip)
). Openroad fails to run the PDN generation step, failing with this unhelpful error:

```
OpenROAD 3eafa6da2942ec22d27e2208c99c7a449af34fbf 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/build_lut/runs/RUN_2022.03.04_22.16.19/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/build_lut/runs/RUN_2022.03.04_22.16.19/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/build_lut/test.def
[INFO ODB-0128] Design: lut
[INFO ODB-0130]     Created 9 pins.
[INFO ODB-0131]     Created 111 components and 643 component-terminals.
[INFO ODB-0133]     Created 89 nets and 257 connections.
[INFO ODB-0134] Finished DEF file: /openlane/build_lut/test.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: 
[INFO PDN-0008] Design name is lut.
[INFO PDN-0009] Reading technology data.
[INFO PDN-0011] ****** INFO ******
Type: stdcell, stdcell_grid
    Stdcell Rails
      Layer: met1  -  width: 0.480  pitch: 5.440 
    Straps
      Layer: met4  -  width: 1.600  pitch: 153.600  offset: 16.320 
    Connect: {met1 met4}
Type: macro, CORE_macro_grid_1
    Macro orientation: R0 R180 MX MY R90 R270 MXR90 MYR90
    Straps
    Connect: {met4_PIN_ver met5}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - stdcell_grid.
can't read "stripe_locs(met4,GROUND)": no such element in array
```

I looked at what I think is the [source](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/pdn/src/PdnGen.tcl), but stripe_locs is used in several different procs so I'm not sure which is the problem. I'm not sure what I'm doing wrong.

The configuration is generated by [this openlane script](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/pdn_cfg.tcl), but it seems to spit out that power net is `VPWR` and the ground net is `VGND`.

vijayank88, 2022-03-05:
@growly 
are you running openlane for MPW?
Try to update latest OpenLane tag: [2022.03.03_02.36.44](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.03_02.36.44), if issue repeats attach openorad_issue_reproducible tar file generated by tool.

growly, 2022-03-08:
The problem has gone away with

1. Using the [2022.03.03_02.36.44](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.03.03_02.36.44) version of OpenLane
2. Setting the `FP_PDN_HPITCH` and `FP_PDN_VPITCH` parameters to something smaller than the size of the design (I got the hint from #1008)

closed 2022-04-15 by vijayank88