The-OpenROAD-Project/OpenROAD #1688: Trouble building LEF parser for macOS/Windows

nmoroze, 2022-03-09:
I noticed a couple errors in CI for the SiliconCompiler project when trying to build the si2 LEF parser out of `src/odb/src/lef` after updating to a recent version of OpenROAD. On macOS, the error is:

```
    /private/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T/pip-req-build-84dlzdxd/third_party/tools/openroad/tools/OpenROAD/src/odb/src/lef/lef/lefrReader.cpp:319:14: error: use of undeclared identifier 'lefyyparse'
        status = lefyyparse();
```
[source](https://github.com/siliconcompiler/siliconcompiler/runs/5473173740?check_suite_focus=true#step:9:228)

This seems to be related to [this](https://github.com/The-OpenROAD-Project/OpenROAD/commit/4016a522e2d50694e912d63828addc7a36bd1ef9#diff-7e4a5edd455cf5d0f7a5ab17c20ec3ee8db2f56e01330b0e3d8da85083342894L62) change, adding back the `extern` lefyyparse declaration fixes the error for me locally on my Macbook. Was there any reason for getting rid of this? 

I figure since the build succeeds on macOS in OpenROAD CI, there's likely some problem in our machine configuration, but any insight you could provide would be appreciated.

On Windows we're getting a linker error I have not yet had time to to dig into. I know this isn't a supported platform, but I'd appreciate if you have any insight there as well:

```
    lef.lib(lef_keywords.cpp.obj) : error LNK2019: unresolved external symbol "union LefDefParser::YYSTYPE LefDefParser::lefyylval" (?lefyylval@LefDefParser@@3TYYSTYPE@1@A) referenced in function "int __cdecl LefDefParser::lefamper_lookup(char *)" (?lefamper_lookup@LefDefParser@@YAHPEAD@Z)
```
[source](https://github.com/siliconcompiler/siliconcompiler/runs/5473173795?check_suite_focus=true#step:9:264)

Thank you!

jjcherry56, 2022-04-04:
What bison/flex are you using? I build using the homebrew versions with 
    # flex/bison override apple version
    export PATH="/opt/homebrew/opt/bison/bin:$PATH"
    export PATH="/opt/homebrew/opt/flex/bin:$PATH"

to ignore the broken ones in /usr/bin installed by xcode.

nmoroze, 2023-02-10:
Thank you! This does the trick. Sorry for not following up!

closed 2023-02-10 by nmoroze