The-OpenROAD-Project/OpenROAD #1690: `Min area of metal1 holes > 0.14um^2 (met1.7)` from detailed router

donn, 2022-03-11:
While the design passes detailed routing fine, magic later finds an issue with one of the metal1 holes:

[usb.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8233009/usb.tar.gz)

See https://github.com/RTimothyEdwards/open_pdks/issues/236 for Tim's image

This is a blocker for mpw-5.

vijayank88, 2022-03-12:
Relevant to another issue: https://github.com/The-OpenROAD-Project/OpenLane/issues/954
----------------------------------------
Metal3 minimum area < 0.24um^2 (met3.6)
----------------------------------------

maliberty, 2022-03-14:
Where in the layout is this hole to be found?

maliberty, 2022-03-14:
The linked issue says

`The attached screenshot was taken at position (181.675um, 880.160um), (181.860um, 880.315um) `

which makes no sense as the design isn't that large:
```
UNITS DISTANCE MICRONS 1000 ;
DIEAREA ( 0 0 ) ( 175225 185945 ) ;
```

That location is outside the DIEAREA.

maliberty, 2022-03-14:
@RTimothyEdwards would you respond to the above about error location

donn, 2022-03-14:
That's on me, I uploaded a different layout with the same issue. @maliberty @RTimothyEdwards

I'll upload a new one in a bit.

maliberty, 2022-03-14:
@donn whichever is smaller is better :)

maliberty, 2022-03-14:
I can reproduce it now and will take a look.

maliberty, 2022-03-14:
There is no MINENCLOSEDAREA rule in the LEF for met1 so there is no reason for the router to think this is illegal.  Why does magic think so?  @RTimothyEdwards @donn

maliberty, 2022-03-14:
@Stephanommg this is some ugly routing (even if not illegal per LEF):
[image]

Would you take a look when you have a chance.  It isn't urgent.  Be sure to use the reproducible.tar.gz above.

RTimothyEdwards, 2022-03-14:
@maliberty :  (1) There _is_ a MINENCLOSEDAREA rule in the LEF file for metal1 if you are using the technology LEF file from open_pdks.  (2) Regardless of whether or not the LEF file declares MINENCLOSEDAREA, it's still a SkyWater DRC rule, and magic flags it.

RTimothyEdwards, 2022-03-14:
@Donn:  (@maliberty ) :  The problem is that with the change to min/max/nom versions of the technology LEF file, the MINENCLOSEDAREA patch to the technology LEF file somehow vanished.  The issue is in open_pdks somewhere and I am investigating.  Please feel free to close this issue, as it is not an OpenROAD problem.

RTimothyEdwards, 2022-03-15:
@donn :  I'm still not sure why the patch failed, but I'm testing a fix now.

maliberty, 2022-03-15:
I'll leave this open for the routing quality concern but will not worry about the holes issue.

dineshannayya, 2022-07-22:
@maliberty I am also facing this issue with latest openlane tool set with PDK=sky130B and same thing passes with PDK=sky130A
My Repo: https://github.com/dineshannayya/riscduino_qcore
Failing Macro : make ycr4_iconnect
Failing Signature: 
----------------------------------------
Min area of metal1 holes > 0.14um^2 (met1.7)
----------------------------------------
 175.695um 265.440um 175.880um 265.595um
 142.115um 221.920um 142.300um 222.075um
 183.975um 205.600um 184.160um 205.755um
 91.055um 33.605um 91.240um 33.760um
 219.395um 1081.440um 219.580um 1081.595um
 193.635um 1418.720um 193.820um 1418.875um
 207.435um 1364.320um 207.620um 1364.475um
 211.575um 1339.205um 211.760um 1339.360um
 225.835um 1266.400um 226.020um 1266.555um
 207.895um 1817.925um 208.080um 1818.080um
 203.755um 1766.880um 203.940um 1767.035um
----------------------------------------
[INFO]: COUNT: 11
[INFO]: Should be divided by 3 or 4

Openlane Tool Version:

<pre>Kernel: Linux v5.13.0-52-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (HEAD -&gt; master, origin/master, origin/HEAD)
abb9d59 2022-07-01T13:46:02+02:00 Reimplement support for multiple power domains (#1175) - Marwan Abbas -  ()
83b6145 2022-06-26T18:19:11+02:00 Fix #1163 (#1164) - Mohamed Gaber -  ()
</pre>

dineshannayya, 2022-07-22:
@maliberty this block is same one which you have suggested fix in during below issue analysis 
https://github.com/The-OpenROAD-Project/OpenLane/issues/1106

maliberty, 2022-07-22:
@RTimothyEdwards last comment "I'm` still not sure why the patch failed, but I'm testing a fix now." - was this actually fixed?

I believe Stephano did fix a loop case in drt as well.

In any case you need to package a proper routing test case and not just point at a repo.

dineshannayya, 2022-07-22:
@maliberty As this does not fails at routing phase, but fails Magic DRC check phase, i need to see how to extract both stage.

maliberty, 2022-07-22:
@donn was going to deal with this but the standard workaround is to put a call to "exit -1" in the relevant script to force a failure.

Based on the slack discussion this is an open_pdk issue not an OR one.

antonblanchard, 2022-07-22:
@dineshannayya are you sure this is the same issue, and not: https://github.com/RTimothyEdwards/open_pdks/pull/262 ?

donn, 2022-07-22:
@maliberty Recent update: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

maliberty, 2022-07-23:
@antonblanchard based on a slack discussion this is an open_pdk issue and not an OR one.  I don't see any point in keeping this open anymore.

closed 2022-07-23 by maliberty