The-OpenROAD-Project/OpenROAD #1692: PDN Generation issue for some macros

donn, 2022-03-14:
Truthfully, I'm not 100% sure if this is just a config issue but, for some macros, for reasons I cannot discern, pdngen creates vertical straps that covers the macros, ignoring the halo setting.

[image]

* Reproducible:
[pdngen.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8243514/pdngen.tar.gz)

Curiously, just a different macro with the same config works fine:

[working.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8243567/working.tar.gz)

donn, 2022-03-14:
fyi @mattvenn

maliberty, 2022-03-14:
The straps are on met4 & met5 and the LEF for rgb_mixer has no OBS above met3 so why shouldn't there be a straps there?  

The more interesting question is why aren't the power pins connected.  I'll look at the config as I get a chance (just digging out from vacation).

mattvenn, 2022-03-14:
the interesting thing for me is that some designs work and others don't.
I didn't pickup on the OBS - what adds that in the flow/config?

mattvenn, 2022-03-14:
quite hard to see, but there are some 'via like' things on met4

[image]

[image]

that are on the met5, but they drop down to nothing on the macro.

another interesting thing is that the macro definitely has met4 power stripes

[image]

but then the lef doesn't put obs for met4. so maybe it's related to that.

Colin-Holehouse, 2022-03-15:
@donn - concerning the original issue you raised:
Straps are placed over the stdcell rows. For this testcase the stdcell rows are continuous across the entire design, hence the straps are across the entire design. In order to prevent this, the rows should be cut such that they do not overlap with the macro placements.

@maliberty - the connectivity issue you're seeing needs a little more explanation:

Comments on the testcase data:
- There is no connectivity specified for the vccd1/vssd1 pins on the rgb_mixer in the testcase.
- There are 2 add_global_connect commands. One connects VPWR and VPB pins to vccd1, one connects VGND and VNB pins to vssd1. But since the rgb_mixer block has pins vccd1 and vssd1 for power and ground, these have not been connected.
- Specifying power_pins and ground_pins as part of the macro definition merely serves to identify which pins are power and ground types [pdn now uses signalType information in the db to identify power and ground pins, so this is only necessary if the LEF being used does not have the correct USE definitions]. 
- The out.def supplied in the testcase creates a physical connection between the met4 pins of the macro and the met5 strap

I expect/assume that in the version of openroad used to create out.def of the testcase, pdn connects macro pins specified by power_pins in the macro definition to the POWER net being added by pdn. From a physical connectivity point of view, I believe this is the desired outcome in this case. However, I would argue that this behaviour is incorrect, since it does not take into account the logical connectivity when making this connection. This results in out.def which has inconsistent logical/physical netlist connections.

The pdn code has been updated since that snapshot

The latest pdn code looks at the database to determine which nets these pins are connected to in order to determine whether a connection needs to be made or not. In this case, the pins are not connected to any net, so no connections are made to the macro. This is consistent in that the physical connectivity matches the logical connectivity

After creating the logical connection between the vccd1 pins on rgb_mixer and the vccd1 net, the physical connection is made by pdn, once again resulting in a consistent DEF file, with matching logical and physical connectivity.

I was able to verify this by adding the logical connection into in.def

```
SPECIALNETS 2 ;
  - vccd1 ( rgb_mixer vccd1 ) + USE POWER ;
  - vssd1 ( rgb_mixer vssd1 ) + USE GROUND ;
END SPECIALNETS
```

I could also have used the following TCL commands:

```
add_global_connection -net vccd1 -inst_pattern rgb_mixer -pin_pattern vccd1
add_global_connection -net vssd1 -inst_pattern rgb_mixer -pin_pattern vssd1
```

donn, 2022-03-15:
@Colin-Holehouse The working example also has rows across the whole thing. Except the difference is the straps stop just fine there.

Colin-Holehouse, 2022-03-15:
@donn The macro rgb_mixer instantiated by pdngen.tar.gz has blockages up to met3, whereas the macro wrapped_rgb_mixer instantiated by working.tar.gz has blockages up to met4. 
The fact that there are any blockages on met4 causes the macro halo to be used to cut the met4 stdcell grid lines.
I can modify the code to also check the layers used by pins to determine which layers to cut - is this a change that you would like to see implemented?

mbalestrini, 2022-03-15:
> 
> but then the lef doesn't put obs for met4. so maybe it's related to that.

I think the LEF file is created in magic by openlane, using `lef write -hide` 
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/magic/lef.tcl

From magic: `Option -hide generates an abstract view that has all interior detail removed, leaving only pins and obstruction layers covering the entire cell with cut-out areas for the pins`
I think as rgb_mixer only has PINS on met4 it doesn't created the OBS

maliberty, 2022-03-16:
I don't see any OR issue to fix here.  Is there any reason to keep this open?

donn, 2022-03-17:
@maliberty @Colin-Holehouse Maybe add a warning for if pins are found? Don't auto-avoid, just add a warning.

mattvenn, 2022-05-11:
still hitting this on mpw-5c

mattvenn, 2022-05-11:
changing size 300x300 to 100x250 made the pdn bug go away

mattvenn, 2022-05-16:
How would I know if I was using the new pdn?
this is tested on mpw-6c (sorry my comment above was wrong when I said mpw-5c).

maliberty, 2022-05-16:
> How would I know if I was using the new pdn? this is tested on mpw-6c (sorry my comment above was wrong when I said mpw-5c).

It was turned on in commit 0c611a3422f0a9691c10776f951efd32887e85e5.   If you see the use of -secondary_power in scripts/openroad/pdn.cfg then you are  on the new one.

mattvenn, 2022-05-17:
then the answer is no

mattvenn, 2022-10-11:
> @VidyaChhabria <https://github.com/VidyaChhabria> check_power_grid should
> be reporting an error on this design but doesn't.

Apart from the non reported error, what is the fix?

maliberty, 2022-10-11:
This issue is quite old.  Has anyone tried this with the current OR and the rewritten pdngen?

mattvenn, 2022-10-11:
haven't seen this recently, so probably safe to close.

closed 2022-10-11 by maliberty