The-OpenROAD-Project/OpenROAD #1693: Platform Bringup KLayout technology file description

gkamendje, 2022-03-14:
The document `PlatformBringUp.md` describes how to generate a KLayout technology file `.lyt` . While doing so, it suggests to "load the previously generated `.lyp` file" before describing the steps to generate the KLayout property file. I think the order of the two chapters should be swapped. 
Having some screenshots in both the sections that describe how to generate` .lyt` and `.lyp` files would be very helpful.

maliberty, 2022-03-14:
If you feel inclined to improve it and submit a PR that would be great.  If not it will go in the work queue.

gkamendje, 2022-03-14:
I think, I could give it a trial.

vijayank88, 2023-06-14:
@gkamendje 
Are able to resolve it?

gkamendje, 2023-06-14:
I think so. I am able to generate a GDS file. I will work on a PR later on.

closed 2023-06-14 by gkamendje