The-OpenROAD-Project/OpenROAD #1702: resizer leaving max slew violations after placement

DarylK78, 2022-03-16:
In running global placement, the resizer is leaving large max slew violations.  I have tried setting the max wire length to 1000 which should cause these nets to be buffered, but they are still being left with large transitions
[openroad.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8266833/openroad.zip)
.

vijayank88, 2022-03-17:
@DarylK78 
Missing 
```
openroad> source run.tcl
can't read "::env(MERGED_LEF_UNPADDED)": no such variable
```
Also post log message.

DarylK78, 2022-03-17:
Can you please try run.sh?  That is the one I tested before I posted the tarball

vijayank88, 2022-03-18:
@DarylK78 
Are you talking about this slew violations?
```
===========================================================================
 report_check_types -max_slew -max_cap -max_fanout -violators
============================================================================
max slew

Pin                                    Limit    Slew   Slack
------------------------------------------------------------
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[9]    0.04    0.41   -0.37 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[8]    0.04    0.40   -0.36 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[3]    0.04    0.37   -0.33 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[9]    0.04    0.35   -0.31 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[10]    0.04    0.35   -0.31 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[7]    0.04    0.34   -0.30 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[7]    0.04    0.33   -0.29 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[3]    0.04    0.25   -0.21 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[8]    0.04    0.24   -0.20 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[6]    0.04    0.23   -0.19 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[7]    0.04    0.23   -0.19 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr1[4]    0.04    0.22   -0.18 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr0[4]    0.04    0.22   -0.18 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr1[0]    0.04    0.22   -0.18 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr1[2]    0.04    0.22   -0.18 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr0[0]    0.04    0.21   -0.17 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr1[3]    0.04    0.21   -0.17 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[8]    0.04    0.21   -0.17 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr0[3]    0.04    0.21   -0.17 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr0[5]    0.04    0.20   -0.16 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr1[10]    0.04    0.20   -0.16 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[10]    0.04    0.20   -0.16 (VIOLATED)
u_generateOutputs_u_Preamble_DUT_u_GeneratePreambleSpread_u_HDL_FIFO_skw_ram_re/addr0[1]    0.04    0.20   -0.16 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[6]    0.04    0.19   -0.15 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[3]    0.04    0.19   -0.15 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[9]    0.04    0.19   -0.15 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[6]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[5]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_Delimiter_DUT_u_GenerateDelimiterSpread/u_HDL_FIFO_skyw_ram/addr1[2]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[5]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[4]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_Delimiter_DUT_u_GenerateDelimiterSpread/u_HDL_FIFO_skyw_ram/addr0[0]    0.04    0.18   -0.14 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr0[4]    0.04    0.17   -0.13 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[2]    0.04    0.16   -0.12 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[5]    0.04    0.15   -0.11 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr1[1]    0.04    0.15   -0.11 (VIOLATED)
u_generateOutputs_u_Delimiter_DUT_u_GenerateDelimiterSpread/u_HDL_FIFO_skyw_ram/addr1[0]    0.04    0.15   -0.11 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr1[7]    0.04    0.15   -0.11 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_CRC_CONVENC_DUT1/u_HDL_FIFO_I_skyw_ram/addr1[4]    0.04    0.15   -0.11 (VIOLATED)
u_generateOutputs_u_Delimiter_DUT_u_GenerateDelimiterSpread/u_HDL_FIFO_skyw_ram/addr0[4]    0.04    0.15   -0.11 (VIOLATED)
[... cut ...]
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr1[0]    0.04    0.14   -0.10 (VIOLATED)
u_generateOutputs_u_Delimiter_DUT_u_GenerateDelimiterSpread/u_HDL_FIFO_skyw_ram/addr1[3]    0.04    0.14   -0.10 (VIOLATED)
u_generateOutputs_u_SingleRateFEC_Tx_u_GenerateConvEncSpread/u_HDL_FIFO_skw_ram_re/addr0[0]    0.04    0.14   -0.10 (VIOLATED)

===========================================================================
max slew violation count 68
max fanout violation count 0
max cap violation count 0
============================================================================
```

DarylK78, 2022-03-18:
Yes

vijayank88, 2022-03-22:
@DarylK78 
Its based manual macro placement right?
[image]
Place the macros close to core edges and based on net connection.

maliberty, 2022-03-22:
If the macros on the left have any pins on their left edge it will be impossible to buffer them as there is no space to do so.   I suggest you not place them hard against the edge and see if that helps.

DarylK78, 2022-03-29:
I will try moving the macro to the top edge to see if that helps.

The pins for the OpenRAM macros are on the top and bottom so being along the left/right edge shouldn't be an issue.

Is there a way to get the tool to recognize long wires and buffer them appropriately to fix these issues?

vijayank88, 2022-03-31:
@DarylK78 
Yes during repair_design tool try to find long wire length and try to optimize it. For your design its more than 2400um max wire length (Just close numbers). Check your logs

maliberty, 2022-03-31:
I've started to look at this and I think there is a potential resizer issue here.  The RAMs have a very tight input transition constraint but it should be achievable.

DarylK78, 2022-03-31:
I agree, given that there is plenty of room I would expect the re-sizer to add buffers or re-size until transition times are met.

jjcherry56, 2022-03-31:
The issue isn't about long wires.
The issue is the max slew, which is unrealistically small in the sram inputs.

This is another example of a completely unrealistic ram model from the [fake ram generator](https://github.com/bespoke-silicon-group/bsg_fakeram).
I noticed a lot of problems with the fake ram liberty files used by the openroad nangate45 regression
tests so I spent a lot of time fixing issues in the generator.  see https://github.com/jjcherry56/bsg_fakeram and https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/test/Nangate45/fakeram45.cfg

I'll look at the resizer for issues, but consider the fact that every other cell in the library uses 1.5ns for the max slew and the ram uses .04ns even though every path through the ram has a fixed delay and the output resistances are complete nonsense. It is not electrically commensurate with the rest of the library.

tspyrou, 2022-04-01:
@DarylK78 if you try openroad through openlane I believe that have manufacturable realistic rams in their sky130 setup.

tspyrou, 2022-04-04:
Please re-open if needed.

closed 2022-04-04 by tspyrou

jjcherry56, 2022-04-05:
I did find a number of issues in the resizer relevant to this issue. There are only a few buffers that will meet that tight of a requirement even right next to the ram input. It is not able to fix all the violations but it does a lot better. The PR is winding its way through the testing process.

DarylK78, 2022-04-05:
I looked at the rams from openroad and they have the same 0.04ns transition requirement on their inputs.

jjcherry56, 2022-04-06:
I have no idea where you were looking to find "rams from openroad" but the ones in openroad/test/nangate45/fakeram*.lib and openroad-flow-scripts/flow/designs/nangate45 use this:
    default_max_transition : 0.227;
because I regenerated them as I described above.
There are no sky130 rams in either repo that I am aware of.

DarylK78, 2022-04-06:
Sorry, I was referring to the skyw_130 RAMs included in the OpenLane distribution from EFabless, which are "released" by them and apparently suitable for production, but as I said they have the same issues outlined here.  Those aren't "openroad" RAMs, I apologize for the confusion.

I believe those were created using the OpenRAM memory compiler, and the models that my testcase has were generated by OpenRAM as well.  It seems the transition times come from "analytical" estimates which are clearly very pessimistic and unrealistic.  I will see if I can work with them to address that.

jjcherry56, 2022-04-06:
As I pointed out above I fixed the problems and posted the repo location. I have not bothered to try and integrate the changes with the bsg ram generator. At the very least the ram should look like a min size buffer, not something from a completely unrelated technology.