The-OpenROAD-Project/OpenROAD #1703: Unconnected PDN node on net vccd1

mattvenn, 2022-03-16:
```
INFO PSM-0064] Number of voltage sources = 88.                                                
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (367.200um, 1910.580um), layer: 6.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (367.200um, 1910.000um), layer: 5.
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1         
```                                                                                  
repo here: https://github.com/mattvenn/zero_to_asic_mpw5

run make user_project_wrapper should fail 
might need to run make install to get caravel/openlane scripts first

[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8271990/reproducible.tar.gz)

[image]

Colin-Holehouse, 2022-03-17:
Given that there are no rails (FP_PDN_ENABLE_RAILS='0'), then there are no transistors being supplied off these segments of metal. 

If rails are enabled, there is a connection to these stripes through the stdcell rails - when I set FP_PDN_ENABLE_RAILS='1' I get the following result:

```
[INFO PSM-0031] Number of PDN nodes on net vccd1 = 378519.
[INFO PSM-0064] Number of voltage sources = 88.
[INFO PSM-0040] All PDN stripes on net vccd1 are connected.
########## IR report #################
Worstcase voltage: 1.80e+00 V
Average IR drop  : 8.66e-15 V
Worstcase IR drop: 8.58e-13 V
######################################

```
So, whilst I agree that the report sections of the grid are disjoint, the fact that there are no connections to transistors mean that the power grid fully connects to all the transistors in the deisgn. The presence of this additional metal is fundamentally benign.

VidyaChhabria, 2022-03-17:
@Colin-Holehouse @maliberty  Is this something that check_power_grid should not be reporting as an error? Let me know if it needs a change.

vijayank88, 2023-06-19:
@mattvenn 
Is that resolved?

mattvenn, 2023-06-19:
Sorry, way to late for me to confirm / deny. Feel free to close this issue.

closed 2023-06-19 by maliberty