The-OpenROAD-Project/OpenROAD #1708: Resizer produces long chains of buffers

antonblanchard, 2022-03-18:
https://github.com/The-OpenROAD-Project/OpenLane/issues/960 mentions issues with resizer buffer insertion. Here is a test case, where the output of 1 DFF feeds the input of 24 others.

The buffers inserted by the resizer:

```
 sky130_fd_sc_hd__buf_12 repeater9 (.A(_000000_), .X(net9));
 sky130_fd_sc_hd__buf_12 repeater10 (.A(_000000_), .X(net10));

 sky130_fd_sc_hd__buf_12 repeater7 (.A(net10), .X(net7));
 sky130_fd_sc_hd__buf_12 repeater8 (.A(net10), .X(net8));

 sky130_fd_sc_hd__buf_12 repeater5 (.A(net8), .X(net5));
 sky130_fd_sc_hd__buf_12 repeater6 (.A(net8), .X(net6));

 sky130_fd_sc_hd__buf_12 repeater3 (.A(net6), .X(net3));
 sky130_fd_sc_hd__buf_12 repeater4 (.A(net6), .X(net4));

 sky130_fd_sc_hd__buf_12 repeater2 (.A(net4), .X(net2));
```

The DFFs have between 0 and 5 buffers inserted:

```
# DFF driving net
 sky130_fd_sc_hd__dfxtp_4 _50000_50000_ (.CLK(net1), .Q(_000000_));

# 0 buffers
 sky130_fd_sc_hd__dfxtp_4 _755000_50000_ (.CLK(net1), .D(_000000_));

# 1 buffer
 sky130_fd_sc_hd__dfxtp_4 _1460000_50000_ (.CLK(net1), .D(net9));
 sky130_fd_sc_hd__dfxtp_4 _2165000_50000_ (.CLK(net1), .D(net9));
 sky130_fd_sc_hd__dfxtp_4 _2870000_50000_ (.CLK(net1), .D(net9));

# 1 buffer
 sky130_fd_sc_hd__dfxtp_4 _50000_905000_ (.CLK(net14), .D(net10));
 sky130_fd_sc_hd__dfxtp_4 _755000_905000_ (.CLK(net14), .D(net10));

# 2 buffers
 sky130_fd_sc_hd__dfxtp_4 _1460000_905000_ (.CLK(net14), .D(net7));
 sky130_fd_sc_hd__dfxtp_4 _2165000_905000_ (.CLK(net14), .D(net7));
 sky130_fd_sc_hd__dfxtp_4 _2870000_905000_ (.CLK(net14), .D(net7));

# 2 buffers
 sky130_fd_sc_hd__dfxtp_4 _50000_1760000_ (.CLK(net13), .D(net8));
 sky130_fd_sc_hd__dfxtp_4 _755000_1760000_ (.CLK(net13), .D(net8));

# 3 buffers
 sky130_fd_sc_hd__dfxtp_4 _50000_2615000_ (.CLK(net12), .D(net6));
 sky130_fd_sc_hd__dfxtp_4 _755000_2615000_ (.CLK(net12), .D(net6));

# 3 buffers
 sky130_fd_sc_hd__dfxtp_4 _1460000_1760000_ (.CLK(net13), .D(net5));
 sky130_fd_sc_hd__dfxtp_4 _2165000_1760000_ (.CLK(net13), .D(net5));
 sky130_fd_sc_hd__dfxtp_4 _2870000_1760000_ (.CLK(net13), .D(net5));

# 4 buffers
 sky130_fd_sc_hd__dfxtp_4 _1460000_2615000_ (.CLK(net12), .D(net3));
 sky130_fd_sc_hd__dfxtp_4 _2165000_2615000_ (.CLK(net12), .D(net3));
 sky130_fd_sc_hd__dfxtp_4 _2870000_2615000_ (.CLK(net12), .D(net3));

# 4 buffers
 sky130_fd_sc_hd__dfxtp_4 _50000_3470000_ (.CLK(net11), .D(net4));
 sky130_fd_sc_hd__dfxtp_4 _755000_3470000_ (.CLK(net11), .D(net4));

# 5 buffers
 sky130_fd_sc_hd__dfxtp_4 _1460000_3470000_ (.CLK(net11), .D(net2));
 sky130_fd_sc_hd__dfxtp_4 _2165000_3470000_ (.CLK(net11), .D(net2));
 sky130_fd_sc_hd__dfxtp_4 _2870000_3470000_ (.CLK(net11), .D(net2));
```

A 3:3:3 tree of buffers would reduce the max depth by 2.

[resizer-buffer-insertion.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8307959/resizer-buffer-insertion.tar.gz)

maliberty, 2022-03-23:
I see it creating 
[image]

How would your 3:3:3 tree look and would it avoid any max cap/slew/length violations?

antonblanchard, 2022-04-05:
I need to rerun the tests now that we've got to the bottom of https://github.com/The-OpenROAD-Project/OpenROAD/issues/1736

antonblanchard, 2022-07-11:
I took another look at this. This test case:

[fanout.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9080972/fanout.tar.gz)

disables capacitance, slew and wire repair and just focuses on how `repair_design` fixes max fan out violations. There is a net with 8000 FFs on it and we set a max fan out of 20, which we should be able to repair with 3 levels of buffers.

`test-nofanout.v` is the baseline, where we use the resizer to fix the fan out issues. While we do fix the fan out issues, we've created some very long chains of buffers:

```
Startpoint: source (rising edge-triggered flip-flop clocked by clk)
Endpoint: sink767 (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                  0.00    0.00    0.00 ^ source/CLK (sky130_fd_sc_hd__dfxtp_4)
                  2.47    1.86    1.86 ^ source/Q (sky130_fd_sc_hd__dfxtp_4)
     9    1.35                           net (net)
                  2.47    0.00    1.86 ^ repeater8482/A (sky130_fd_sc_hd__buf_6)
                  0.12    0.30    2.16 ^ repeater8482/X (sky130_fd_sc_hd__buf_6)
    15    0.04                           net8482 (net)
                  0.12    0.00    2.16 ^ repeater8480/A (sky130_fd_sc_hd__buf_4)
                  0.10    0.18    2.34 ^ repeater8480/X (sky130_fd_sc_hd__buf_4)
    13    0.03                           net8480 (net)
                  0.10    0.00    2.34 ^ repeater8291/A (sky130_fd_sc_hd__clkbuf_4)
                  0.14    0.22    2.56 ^ repeater8291/X (sky130_fd_sc_hd__clkbuf_4)
    19    0.04                           net8291 (net)
                  0.14    0.00    2.56 ^ repeater8290/A (sky130_fd_sc_hd__clkbuf_4)
                  0.14    0.24    2.80 ^ repeater8290/X (sky130_fd_sc_hd__clkbuf_4)
    19    0.04                           net8290 (net)
                  0.14    0.00    2.80 ^ repeater8289/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.22    3.02 ^ repeater8289/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8289 (net)
                  0.14    0.00    3.02 ^ repeater8288/A (sky130_fd_sc_hd__buf_4)
                  0.13    0.21    3.23 ^ repeater8288/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8288 (net)
                  0.13    0.00    3.23 ^ repeater8287/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.21    3.44 ^ repeater8287/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8287 (net)
                  0.14    0.00    3.44 ^ repeater8286/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.22    3.66 ^ repeater8286/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8286 (net)
                  0.14    0.00    3.66 ^ repeater8285/A (sky130_fd_sc_hd__buf_4)
                  0.15    0.22    3.89 ^ repeater8285/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8285 (net)
                  0.15    0.00    3.89 ^ repeater8284/A (sky130_fd_sc_hd__buf_6)
                  0.09    0.16    4.05 ^ repeater8284/X (sky130_fd_sc_hd__buf_6)
    16    0.04                           net8284 (net)
                  0.09    0.00    4.05 ^ repeater8282/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.20    4.25 ^ repeater8282/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8282 (net)
                  0.14    0.00    4.25 ^ repeater8281/A (sky130_fd_sc_hd__buf_4)
                  0.15    0.22    4.47 ^ repeater8281/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8281 (net)
                  0.15    0.00    4.47 ^ repeater8280/A (sky130_fd_sc_hd__buf_6)
                  0.09    0.17    4.64 ^ repeater8280/X (sky130_fd_sc_hd__buf_6)
    19    0.04                           net8280 (net)
                  0.09    0.00    4.64 ^ repeater8279/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.20    4.85 ^ repeater8279/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8279 (net)
                  0.14    0.00    4.85 ^ repeater8278/A (sky130_fd_sc_hd__buf_4)
                  0.14    0.21    5.06 ^ repeater8278/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8278 (net)
                  0.14    0.00    5.06 ^ repeater8277/A (sky130_fd_sc_hd__buf_4)
                  0.15    0.22    5.28 ^ repeater8277/X (sky130_fd_sc_hd__buf_4)
    19    0.05                           net8277 (net)
                  0.15    0.00    5.28 ^ repeater8276/A (sky130_fd_sc_hd__buf_6)
[... 256 lines cut ...]
                  0.35    0.52    0.52 v source/Q (sky130_fd_sc_hd__dfxtp_4)
    20    0.68                           net (net)
                  2.36    1.16    1.68 v buffer_1_net/A (sky130_fd_sc_hd__clkbuf_16)
                  0.68    1.24    2.92 v buffer_1_net/X (sky130_fd_sc_hd__clkbuf_16)
    20    2.62                           net_1_net (net)
                  4.42    2.15    5.07 v buffer_18_net_1_net/A (sky130_fd_sc_hd__clkbuf_16)
                  0.63    1.85    6.92 v buffer_18_net_1_net/X (sky130_fd_sc_hd__clkbuf_16)
    20    1.65                           net_18_net_1_net (net)
                  8.15    4.09   11.01 v sink761/D (sky130_fd_sc_hd__dfxtp_1)
[... cut ...]
-----------------------------------------------------------------------------
                                  2.37   data required time
                                -11.01   data arrival time
-----------------------------------------------------------------------------
                                 -8.65   slack (VIOLATED)
```

@jjcherry56 I can see where we split the net in `RepairDesign::repairNetJunc()`, but I'm wondering if we can do it at more optimal points. I see a lot of places where we split where `fanout_left == 1` and `fanout_right == 19` for example.

jjcherry56, 2022-07-31:
With commit c525def45 the timing looks like this:

```
Startpoint: source (rising edge-triggered flip-flop clocked by clk)
Endpoint: sink3018 (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout      Cap     Slew    Delay     Time   Description
---------------------------------------------------------------------------------
                   0.000    0.000    0.000   clock clk (rise edge)
                            0.000    0.000   clock network delay (ideal)
                   0.000    0.000    0.000 ^ source/CLK (sky130_fd_sc_hd__dfxtp_4)
                   0.899    0.957    0.957 v source/Q (sky130_fd_sc_hd__dfxtp_4)
     4    1.371                              net (net)
                   0.899    0.000    0.957 v fanout8254/A (sky130_fd_sc_hd__clkbuf_16)
                   0.392    0.735    1.692 v fanout8254/X (sky130_fd_sc_hd__clkbuf_16)
     4    1.344                              net8254 (net)
                   0.392    0.007    1.699 v wire8255/A (sky130_fd_sc_hd__clkbuf_16)
                   0.425    0.594    2.292 v wire8255/X (sky130_fd_sc_hd__clkbuf_16)
     7    1.456                              net8255 (net)
                   9.869    4.943    7.235 v fanout8126/A (sky130_fd_sc_hd__buf_6)
                   0.313    2.592    9.827 v fanout8126/X (sky130_fd_sc_hd__buf_6)
    18    0.081                              net8126 (net)
                   0.313    0.002    9.829 v fanout8119/A (sky130_fd_sc_hd__buf_2)
                   0.089    0.294   10.123 v fanout8119/X (sky130_fd_sc_hd__buf_2)
    10    0.033                              net8119 (net)
                   0.089    0.001   10.124 v fanout8117/A (sky130_fd_sc_hd__buf_2)
                   0.079    0.191   10.315 v fanout8117/X (sky130_fd_sc_hd__buf_2)
    10    0.030                              net8117 (net)
                   0.079    0.001   10.316 v fanout8116/A (sky130_fd_sc_hd__buf_4)
                   0.078    0.205   10.521 v fanout8116/X (sky130_fd_sc_hd__buf_4)
    20    0.054                              net8116 (net)
                   0.078    0.001   10.522 v sink3018/D (sky130_fd_sc_hd__dfxtp_1)
                                    10.522   data arrival time

                   0.000    5.000    5.000   clock clk (rise edge)
                            0.000    5.000   clock network delay (ideal)
                            0.000    5.000   clock reconvergence pessimism
                                     5.000 ^ sink3018/CLK (sky130_fd_sc_hd__dfxtp_1)
                           -0.136    4.864   library setup time
                                     4.864   data required time
---------------------------------------------------------------------------------
                                     4.864   data required time
                                   -10.522   data arrival time
---------------------------------------------------------------------------------
                                    -5.657   slack (VIOLATED)
```

The best part is that repair_timing can improve the timing further if necessary.

```
Startpoint: source (rising edge-triggered flip-flop clocked by clk)
Endpoint: sink6339 (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout      Cap     Slew    Delay     Time   Description
---------------------------------------------------------------------------------
                   0.000    0.000    0.000   clock clk (rise edge)
                            0.000    0.000   clock network delay (ideal)
                   0.000    0.000    0.000 ^ source/CLK (sky130_fd_sc_hd__dfxtp_4)
                   0.079    0.369    0.369 v source/Q (sky130_fd_sc_hd__dfxtp_4)
     4    0.046                              net (net)
                   0.080    0.003    0.372 v wire8511/A (sky130_fd_sc_hd__clkbuf_16)
                   0.424    0.456    0.828 v wire8511/X (sky130_fd_sc_hd__clkbuf_16)
     4    1.478                              net8511 (net)
                  10.227    5.122    5.950 v fanout8344/A (sky130_fd_sc_hd__buf_8)
                   0.231    1.998    7.948 v fanout8344/X (sky130_fd_sc_hd__buf_8)
    12    0.079                              net8344 (net)
                   0.231    0.008    7.956 v fanout8333/A (sky130_fd_sc_hd__clkbuf_4)
                   0.092    0.276    8.232 v fanout8333/X (sky130_fd_sc_hd__clkbuf_4)
    10    0.041                              net8333 (net)
                   0.092    0.000    8.232 v fanout8331/A (sky130_fd_sc_hd__buf_4)
                   0.078    0.211    8.443 v fanout8331/X (sky130_fd_sc_hd__buf_4)
    20    0.054                              net8331 (net)
                   0.078    0.001    8.444 v sink6339/D (sky130_fd_sc_hd__dfxtp_1)
                                     8.444   data arrival time

                   0.000    5.000    5.000   clock clk (rise edge)
                            0.000    5.000   clock network delay (ideal)
                            0.000    5.000   clock reconvergence pessimism
                                     5.000 ^ sink6339/CLK (sky130_fd_sc_hd__dfxtp_1)
                           -0.136    4.864   library setup time
                                     4.864   data required time
---------------------------------------------------------------------------------
                                     4.864   data required time
                                    -8.444   data arrival time
---------------------------------------------------------------------------------
                                    -3.580   slack (VIOLATED)
```

closed 2022-07-31 by jjcherry56

antonblanchard, 2022-08-01:
Thanks @jjcherry56, this looks great! I have tested it out on a few designs and these long chains have gone away.