The-OpenROAD-Project/OpenROAD #1718: pdngen not read successfully

donn, 2022-03-25:
Despite the met3 resistance values being specified in the merged lef file being read, pdngen fails with "met3 resistance not found in DB. Check the LEF or set it using the 'set_layer_rc' command.".

* Logs:
```
OpenROAD v2.0-3168-g5b0698f91 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: ./in.def
[INFO ODB-0128] Design: PPU
[INFO ODB-0130]     Created 101 pins.
[INFO ODB-0131]     Created 22972 components and 113199 component-terminals.
[INFO ODB-0133]     Created 12494 nets and 41087 connections.
[INFO ODB-0134] Finished DEF file: ./in.def
[INFO PDN-0016] Power Delivery Network Generator: Generating PDN
  config: openlane/scripts/openroad/pdn_cfg.tcl
[INFO PDN-0008] Design name is PPU.
[INFO PDN-0009] Reading technology data.
[INFO PDN-0011] ****** INFO ******
Type: stdcell, stdcell_grid
    Stdcell Rails
      Layer: met1  -  width: 0.480  pitch: 5.440 
    Straps
      Layer: met4  -  width: 1.600  pitch: 153.600  offset: 16.320 
      Layer: met5  -  width: 1.600  pitch: 153.180  offset: 16.650 
    Connect: {met4 met5} {met1 met4}
Type: macro, CORE_macro_grid_1
    Macro orientation: R0 R180 MX MY R90 R270 MXR90 MYR90
    Straps
    Connect: {met4_PIN_ver met5}
[INFO PDN-0012] **** END INFO ****
[INFO PDN-0013] Inserting stdcell grid - stdcell_grid.
[INFO PDN-0015] Writing to database.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net VPWR is not explicitly set.
[WARNING PSM-0022] Using voltage 1.800V for VDD network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[WARNING PSM-0030] VSRC location at (5.520um, 10.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 27.290um).
[WARNING PSM-0030] VSRC location at (285.520um, 150.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (329.040um, 180.470um).
[WARNING PSM-0030] VSRC location at (565.520um, 290.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (636.240um, 333.650um).
[WARNING PSM-0030] VSRC location at (5.520um, 430.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 486.830um).
[WARNING PSM-0030] VSRC location at (285.520um, 570.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (329.040um, 640.010um).
[INFO PSM-0031] Number of PDN nodes on net VPWR = 26892.
[INFO PSM-0064] Number of voltage sources = 5.
[INFO PSM-0040] All PDN stripes on net VPWR are connected.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net VGND is not explicitly set.
[WARNING PSM-0021] Using voltage 0.000V for ground network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[WARNING PSM-0030] VSRC location at (5.520um, 10.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 103.880um).
[WARNING PSM-0030] VSRC location at (285.520um, 150.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (252.240um, 103.880um).
[WARNING PSM-0030] VSRC location at (565.520um, 290.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (559.440um, 257.060um).
[WARNING PSM-0030] VSRC location at (5.520um, 430.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 410.240um).
[WARNING PSM-0030] VSRC location at (285.520um, 570.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (252.240um, 563.420um).
[INFO PSM-0031] Number of PDN nodes on net VGND = 26404.
[INFO PSM-0064] Number of voltage sources = 5.
[INFO PSM-0040] All PDN stripes on net VGND are connected.
[INFO]: You shouldn't be using set_rc
[INFO PSM-0002] Output voltage file is specified as: ./tmp/floorplan/6-pdn.pga.rpt.
[WARNING PSM-0016] Voltage pad location (VSRC) file not specified, defaulting pad location to checkerboard pattern on core area.
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0019] Voltage on net VPWR is not explicitly set.
[WARNING PSM-0022] Using voltage 1.800V for VDD network.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[WARNING PSM-0030] VSRC location at (5.520um, 10.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 27.290um).
[WARNING PSM-0030] VSRC location at (285.520um, 150.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (329.040um, 180.470um).
[WARNING PSM-0030] VSRC location at (565.520um, 290.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (636.240um, 333.650um).
[WARNING PSM-0030] VSRC location at (5.520um, 430.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (5.520um, 486.830um).
[WARNING PSM-0030] VSRC location at (285.520um, 570.880um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (329.040um, 640.010um).
[INFO PSM-0031] Number of PDN nodes on net VPWR = 26892.
[ERROR PSM-0035] met3 resistance not found in DB. Check the LEF or set it using the 'set_layer_rc' command.
Error: pdn.tcl, 50 PSM-0035
```

* Reproducible:
[rep.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8351597/rep.tar.gz)

donn, 2022-03-25:
cc @marwaneltoukhy

tspyrou, 2022-03-25:
@osamahammad21 if you have time to look at this over the weekend please take a look. I think Matt is our until Monday.

maliberty, 2022-03-26:
I took a quick look and the message is just wrong.  It is via3 that is missing resistance and that is true.

closed 2022-03-26 by maliberty