The-OpenROAD-Project/OpenROAD #1729: sta: set_min_delay -from -to doesn't work

antonblanchard, 2022-03-30:
I'm trying to add constraints for a hard macro created by OpenROAD/Openlane and integrated into my design. I need constraints because we don't support liberty file creation yet, so the hard macro has no timing information.

This works:

```
set ram512_delay 5
set_min_delay -to microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0* $ram512_delay
```

But that requires you to estimate the clock tree delay to the CLK pin on the macro. I tried to specify a delay referenced to the CLK pin of the macro:

```
set_ram512_delay 0.5
set_min_delay -from microwatt_0.soc0.bram.bram0.ram_0.memory_0/CLK -to microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0* $ram512_delay
```

But that doesn't work. I also tried adding `-ignore_clock_latency`, but that didn't seem to help:

```
set_ram512_delay 0.5
set_min_delay -ignore_clock_latency -from microwatt_0.soc0.bram.bram0.ram_0.memory_0/CLK -to microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0* $ram512_delay
```

I'm not sure if I am doing the wrong thing, or if this is an issue. Feel free to close if I'm misusing the API.

Test case is too large to attach, so I put it here: https://ozlabs.org/~anton/junk/set_min_delay.tar.gz

jjcherry56, 2022-04-01:
The problem is that your -from is not a valid exception point. If it is a pin it has to be and endpoint (output port or clocked setup/hold constrained input). Even if the constraints were supported on random pins (which would not be compatible with bigco) it would not do what I think you want it do do. I can't tell what you are trying to do from the descripion, but I think you have a black box (the ram) with no liberty but you want it to act like it has timing paths and timing checks set_min_delay does not specify a path delay between pins, it specifies a constraint on paths between the start/end points. You should consider writing an sdf file for the rams with IOPATH and SETUP HOLD statements using TCL or python.

antonblanchard, 2022-04-02:
Thanks @jjcherry56 that is exactly the issue - OpenROAD has no way to produce timing data for hard macros, so DFFRAMs as well as my gate level multiplier and register file are all black boxes.

Thanks for the suggestion, I'll try constructing SDF files instead.

closed 2022-04-02 by antonblanchard

antonblanchard, 2022-04-02:
I just realised we are producing SDF files, but we have no way to use them in Openlane. I'll look into that issue.

jjcherry56, 2022-04-02:
That isn't the SDF you are looking for. It is gate level annotation. You want it at the block level. It is a hack to use sdf because it is not context sensitive (input slew/ouput load dependent). But it does capture goes into goes otta relationships that is better than nothing.

Once upon a time OpenSTA had model generation capabilities, but in the general case it is a nightmare to get right (parasitics from routing over the macro, transparent latches inside the block that can't be modeled properly with liberty and barely with ILMs, exceptions to/from internal pins etc). One of these days maybe something will reappear.

antonblanchard, 2022-04-07:
@jjcherry56 does OpenSTA support wildcards on port names? I couldn't find anything concrete in the spec itself but the changelog says:

```
2.1
The use of the wildcard instance specification restricted to cells at the ASIC
physical primitive level, no longer allowed in PATH or port_path.

3.0
The restriction of the use of the wildcard instance specification to cells at
the ASIC physical primitive level added in SDF 2.1 was removed.
```

So I think wildcards on port names is legal in SDF v3.0. This is my attempt at a block level SDF for a macro with 3 64 bit inputs (a, b, c) and one 64 bit output (o).

```
(DELAYFILE
 (SDFVERSION "3.0")
 (DESIGN "multiply_add_64x64")
 (DATE "Tue Apr  5 23:23:26 2022")
 (VENDOR "Manual")
 (PROGRAM "STA")
 (VERSION "2.3.1")
 (DIVIDER .)
 (VOLTAGE 1.800::1.800)
 (PROCESS "1.000::1.000")
 (TEMPERATURE 25.000::25.000)
 (TIMESCALE 1ns)
 (CELL
  (CELLTYPE "multiply_add_64x64")
  (INSTANCE)
  (DELAY
   (ABSOLUTE
    (IOPATH a* o* (8:8:8) (8:8:8))
    (IOPATH b* o* (8:8:8) (8:8:8))
    (IOPATH c* o* (8:8:8) (8:8:8))
   )
  )

  (TIMINGCHECK
   (SETUP a* (posedge clk) (-1.9:-1.9:-1.9))
   (HOLD  a* (posedge clk) (2.1:2.1:2.1))

   (SETUP b* (posedge clk) (-1.9:-1.9:-1.9))
   (HOLD  b* (posedge clk) (2.1:2.1:2.1))

   (SETUP c* (posedge clk) (-1.9:-1.9:-1.9))
   (HOLD  c* (posedge clk) (2.1:2.1:2.1))
  )
 )
)
```

It works with a[0] a[1] etc, but not a*

antonblanchard, 2022-04-12:
Here is the test case (`resizer_timing.tcl` and `resizer_timing-5.tcl`): [liberty-macro-hold-test.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8469429/liberty-macro-hold-test.tar.gz)

jjcherry56, 2022-04-12:
If you want to see hold time checks you have to use report_checks -path_delay min (you are using max).
Removing all the -from/-to arg that prevent you from seeing the critical path, before resizing it looks like this:

Startpoint: _118386_ (rising edge-triggered flip-flop clocked by user_clock2)
Endpoint: microwatt_0.soc0.bram.bram0.ram_0.memory_0
          (rising edge-triggered flip-flop clocked by user_clock2)
Path Group: user_clock2
Path Type: min

   Delay     Time   Description
-----------------------------------------------------------
   0.000    0.000   clock user_clock2 (rise edge)
   2.427    2.427   clock network delay (propagated)
   0.000    2.427 ^ _118386_/CLK (sky130_fd_sc_hd__dfxtp_1)
   0.338    2.764 v _118386_/Q (sky130_fd_sc_hd__dfxtp_1)
   0.243    3.007 v _058073_/X (sky130_fd_sc_hd__a22o_1)
   0.133    3.140 ^ _058074_/Y (sky130_fd_sc_hd__nand2_2)
   0.101    3.240 v _058075_/Y (sky130_fd_sc_hd__inv_2)
   0.187    3.428 v _059616_/X (sky130_fd_sc_hd__and2_1)
   0.171    3.598 v _059617_/X (sky130_fd_sc_hd__buf_12)
   0.043    3.641 v microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0 (RAM512)
            3.641   data arrival time

   0.000    0.000   clock user_clock2 (rise edge)
   3.155    3.155   clock network delay (propagated)
   0.250    3.405   clock uncertainty
  -0.141    3.264   clock reconvergence pessimism
            3.264 ^ microwatt_0.soc0.bram.bram0.ram_0.memory_0/CLK (RAM512)
   5.000    8.264   library hold time
            8.264   data required time
-----------------------------------------------------------
            8.264   data required time
           -3.641   data arrival time
-----------------------------------------------------------
           -4.623   slack (VIOLATED)

after resizing it shows

Startpoint: _112874_ (rising edge-triggered flip-flop clocked by user_clock2)
Endpoint: _112887_ (rising edge-triggered flip-flop clocked by user_clock2)
Path Group: user_clock2
Path Type: min

   Delay     Time   Description
-----------------------------------------------------------
   0.000    0.000   clock user_clock2 (rise edge)
   2.427    2.427   clock network delay (propagated)
   0.000    2.427 ^ _112874_/CLK (sky130_fd_sc_hd__dfxtp_2)
   0.367    2.794 v _112874_/Q (sky130_fd_sc_hd__dfxtp_2)
   0.134    2.927 ^ _056809_/Y (sky130_fd_sc_hd__nand3_2)
   0.097    3.024 v _058132_/Y (sky130_fd_sc_hd__o21ai_1)
   0.068    3.092 ^ _058133_/Y (sky130_fd_sc_hd__nand2_1)
   0.094    3.186 v _058134_/Y (sky130_fd_sc_hd__a31oi_1)
   0.000    3.186 v _112887_/D (sky130_fd_sc_hd__dfxtp_1)
            3.186   data arrival time

   0.000    0.000   clock user_clock2 (rise edge)
   3.027    3.027   clock network delay (propagated)
   0.250    3.277   clock uncertainty
  -0.141    3.136   clock reconvergence pessimism
            3.136 ^ _112887_/CLK (sky130_fd_sc_hd__dfxtp_1)
  -0.050    3.086   library hold time
            3.086   data required time
-----------------------------------------------------------
            3.086   data required time
           -3.186   data arrival time
-----------------------------------------------------------
            0.100   slack (MET)

which is exactly the slack margin.
Adding slack to other hold paths is irrelevant, and using headroom in setup paths to fix hold paths is intentional.

antonblanchard, 2022-04-12:
@jjcherry56 Of course! I was trying to improve my Fmax and noticed these long paths, but I guess they are unavoidable, because as you say we are just meeting the slack margin.

jjcherry56, 2022-04-12:
Not entirely unavoidable. If you crank down the cycle time it will look for other places to add the hold buffers and not disturb the setup (when possible).

antonblanchard, 2022-04-18:
I've taken another look at this, and I think the resizer is not picking the best place to fix the hold violation. Here's a test case with a huge hold constraint of 100ns between the RAM macro CLK and EN0 pins, a similarly huge setup constraint of 100ns and a clock period of 250ns, just to exaggerate the effect.

The baseline (`base.tcl`) ends up fixing about half the hold violation, at which time we've run out of the 150ns of setup slack and give up.

In the modified version (`hacked.tcl/hacked.def`) I manually added ~100ns of hold buffers to the EN0 pin of the macro. I run it through the resizer to fix up any remaining hold issues, and this version not only fixes all of the 100ns hold violation, it does so with ~30ns of spare setup slack.

So what is going on? If we look at the path the resizer works on first:

```
                  0.11    0.00    2.43 ^ _118386_/CLK (sky130_fd_sc_hd__dfxtp_1)
                  0.07    0.34    2.76 v _118386_/Q (sky130_fd_sc_hd__dfxtp_1)
     2    0.01                           microwatt_0.soc0._309_ (net)
                  0.07    0.00    2.76 v _058073_/A1 (sky130_fd_sc_hd__a22o_1)
                  0.08    0.24    3.01 v _058073_/X (sky130_fd_sc_hd__a22o_1)
     1    0.01                           _011293_ (net)
                  0.08    0.00    3.01 v _058074_/B (sky130_fd_sc_hd__nand2_2)
                  0.12    0.13    3.14 ^ _058074_/Y (sky130_fd_sc_hd__nand2_2)
     2    0.02                           _011294_ (net)
                  0.12    0.00    3.14 ^ _058075_/A (sky130_fd_sc_hd__inv_2)
                  0.08    0.10    3.24 v _058075_/Y (sky130_fd_sc_hd__inv_2)
     4    0.03                           _011295_ (net)
                  0.08    0.00    3.24 v _059616_/A (sky130_fd_sc_hd__and2_1)
                  0.08    0.19    3.43 v _059616_/X (sky130_fd_sc_hd__and2_1)
     2    0.01                           _012646_ (net)
                  0.08    0.00    3.43 v _059617_/A (sky130_fd_sc_hd__buf_12)
                  0.08    0.17    3.60 v _059617_/X (sky130_fd_sc_hd__buf_12)
    10    0.15                           microwatt_0.soc0.bram.bram0.ram_0._2_ (net)
                  0.12    0.04    3.64 v microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0 (RAM512)
```

All the outputs of the instances on that path are the top candidates to fix:

```
Sorted fanins
     hold_slack  slack_gap  level
_118386_/Q -99.623 248.180
_058075_/Y -99.623 246.273
_058074_/Y -99.623 246.273
_058073_/X -99.623 246.273
_059617_/X -99.623 245.730
_059616_/X -99.623 245.730
_058071_/X -99.456 247.511
_112874_/Q -99.456 246.115
```

Having said that, `_059617_/X` which feeds ` microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0` is not at the top of the list. Also, we really don't want to fix the output of `_059617_/X`, since it has a fanout of 10. We want instead to fix the input of `microwatt_0.soc0.bram.bram0.ram_0.memory_0/EN0`.

Is it possible to teach the resizer to create a new net to avoid hurting the other 9 things on this net and operate on it?

Here's the updated test case: [liberty-macro-hold-test-2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8503517/liberty-macro-hold-test-2.tar.gz)

antonblanchard, 2022-04-24:
That makes sense. Perhaps the issue is the resizer doesn't look for other places in the path to fix the hold violation once it has used up all the available setup slack.

jjcherry56, 2022-04-24:
I am working on it. I can get your example to "work" but of course there is fallout and getting it all to work is taking a while.

antonblanchard, 2022-04-24:
Thanks @jjcherry56!

jjcherry56, 2022-04-30:
it should work now
f790f0d31 master origin/master Merge pull request #1819 from The-OpenROAD-Project-staging/secure-rsz_hold

antonblanchard, 2022-05-01:
Thanks @jjcherry56, I've tested it and it does resolve the issue. It's made a huge difference to the fmax we can achieve with Microwatt on sky130hd. Nice work!