The-OpenROAD-Project/OpenROAD #1730: SEGV in sta::ConcreteCell::findPort during CTS

antonblanchard, 2022-03-30:
The following test case hits a SEGV in the STA code during CTS:

```
(gdb) backtrace 
#0  0x00000000011f2a6a in sta::ConcreteCell::findPort(char const*) const ()
#1  0x0000000001787865 in cts::TechChar::computeTopologyResults(cts::TechChar::SolutionData, sta::Vertex*, float, unsigned int) ()
#2  0x0000000001792aa3 in cts::TechChar::create() ()
#3  0x00000000017590b7 in cts::TritonCTS::runTritonCts() ()
#4  0x0000000001764770 in _wrap_run_triton_cts ()
#5  0x00007ffff7e2db52 in TclNRRunCallbacks () from /lib64/libtcl8.6.so
#6  0x00007ffff7e32bea in ?? () from /lib64/libtcl8.6.so
#7  0x00007ffff7e33417 in Tcl_EvalEx () from /lib64/libtcl8.6.so
#8  0x00007ffff7e3343a in Tcl_Eval () from /lib64/libtcl8.6.so
#9  0x00000000011b679c in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#10 0x0000000000fa5614 in ord::tclAppInit(Tcl_Interp*) ()
#11 0x00007ffff7ee5a59 in Tcl_MainEx () from /lib64/libtcl8.6.so
#12 0x0000000000f507e3 in main ()
```

[cts-segv.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8378470/cts-segv.tar.gz)

antonblanchard, 2022-03-30:
I realise what is going on - I was working on the cell exclude lists for openroad and somehow excluded clock cells. So it's definitely an invalid configuration, but might be nice to error out rather than SEGV.

maliberty, 2022-03-30:
There was a check that each buffer exists in LEF but not one for Liberty.  I'm adding that so you'll get an error instead of a crash.

antonblanchard, 2022-03-30:
Thanks @maliberty!