The-OpenROAD-Project/OpenROAD #1736: sta: issues with set_max_fanout

antonblanchard, 2022-03-31:
I'm trying to set a max fanout for all pins in a design. My understanding is this should do it:

```
set_max_fanout 5 [ current_design ]
```

But that doesn't appear to work since I end up with nets much higher fanouts (ie the resizer didn't repair them), and STA also doesn't report any max fanout violations.

Working with @maliberty and @tspyrou, we tried a number of other possible ways to add a fanout constraint:

```
set_max_fanout 5 [ get_nets -hier * ]
set_max_fanout 5 [ get_cells -of_object [get_nets -hier * ] ]
set_max_fanout 5 [ get_cells -hier * ]
set_max_fanout 5 [ get_cells ]
set_max_fanout 5 [ get_lib_cells * ]
set_max_fanout 5 [ get_lib_cells -of_objects [ get_cells -hier * ] ]
```

All seemed to fail.

Test case: [set_max_fanout.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8386757/set_max_fanout.tar.gz)

This runs STA against a design with some large fan outs, and set_max_fanout set to 5:

```Startpoint: _621_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _498_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock source latency
                  0.98    0.72    0.72 ^ clk (in)
     1    0.22                           clk (net)
                  0.98    0.00    0.72 ^ clkbuf_0_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.16    0.41    1.13 ^ clkbuf_0_clk/X (sky130_fd_sc_hd__clkbuf_16)
     2    0.14                           clknet_0_clk (net)
                  0.16    0.00    1.13 ^ clkbuf_1_0_0_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.05    0.17    1.31 ^ clkbuf_1_0_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
     1    0.02                           clknet_1_0_0_clk (net)
                  0.05    0.00    1.31 ^ clkbuf_1_0_1_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.20    0.25    1.56 ^ clkbuf_1_0_1_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.11                           clknet_1_0_1_clk (net)
                  0.20    0.00    1.56 ^ clkbuf_2_0_0_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.21    0.32    1.88 ^ clkbuf_2_0_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.12                           clknet_2_0_0_clk (net)
                  0.22    0.01    1.89 ^ clkbuf_3_1_0_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.90    0.81    2.70 ^ clkbuf_3_1_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
     8    0.54                           clknet_3_1_0_clk (net)
                  0.90    0.04    2.74 ^ clkbuf_leaf_93_clk/A (sky130_fd_sc_hd__clkbuf_16)
                  0.06    0.29    3.03 ^ clkbuf_leaf_93_clk/X (sky130_fd_sc_hd__clkbuf_16)
     2    0.01                           clknet_leaf_93_clk (net)
                  0.06    0.00    3.03 ^ _621_/CLK (sky130_fd_sc_hd__dfxtp_4)
                  1.56    1.48    4.51 ^ _621_/Q (sky130_fd_sc_hd__dfxtp_4)
    26    0.57                           a_registered[5] (net)
                  1.56    0.02    4.53 ^ U$$416/A2 (sky130_fd_sc_hd__a22o_4)
                  4.22    3.09    7.63 ^ U$$416/X (sky130_fd_sc_hd__a22o_4)
    65    1.56                           sel_1$4647 (net)
                  4.35    0.61    8.24 ^ U$$543/B2 (sky130_fd_sc_hd__a22o_1)
                  0.18    0.55    8.79 ^ U$$543/X (sky130_fd_sc_hd__a22o_1)
     1    0.01                           t$4711 (net)
                  0.18    0.00    8.79 ^ U$$544/A (sky130_fd_sc_hd__xor2_2)
                  0.22    0.15    8.94 v U$$544/X (sky130_fd_sc_hd__xor2_2)
     1    0.02                           booth_b6_m63 (net)
```

And yet:

```
max fanout violation count 0
```

tspyrou, 2022-03-31:
Note the manual says list of ports or cells are the objects supported but
set_max_fanout 5 [get_cells -hier *] gives this error.
[ERROR STA-0467] unsupported object type Instance.
Error: base.sdc, 13 STA-0467

jjcherry56, 2022-03-31:
The problem is the liberty file. "fanout" is not the number of loads, it is defined by the liberty fanout_load attribute on input pins and in this library the default value is zero:

    default_fanout_load : 0.0000000000;

change it to 1 and you will see the expected behavior

  default_fanout_load : 1.0;

closed 2022-03-31 by jjcherry56

tspyrou, 2022-03-31:
@antonblanchard can you file an issue on openlane to fix this liberty?

antonblanchard, 2022-03-31:
Thanks @jjcherry56, I verified that fixing the liberty file fixed the issue.

I'm not sure if we have modify the liberty files in open_pdks (so we can fix this ourselves), but I submitted a PR against the sky130 repo at https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/pull/7

antonblanchard, 2022-04-05:
@tspyrou FYI the fix is now in open_pdks, and I also submitted a PR for ORFS. I see a commit in a PR from @jjcherry56 to take care of the OpenROAD tests.