The-OpenROAD-Project/OpenROAD #1747: [Feature request] Routing only specific nets

armleo, 2022-04-03:
Can OpenRoad implement a TCL command for routing only specific nets? Looks like FastRoute will route the nets it has been asked to route. So this should be a fairly simple feature to implement.

The reason for this is that I wanted to extend OpenLane to route some PDN/Mixed signals. However, it seems that the only reasonable way to do this is by asking FastRoute to route these signals before Place. Any suggestions?

maliberty, 2022-04-03:
PDN signals are routed by pdngen not fastroute.  Would you describe further what you are trying to do.  Note that you would also have to enhance the detailed router as well.

armleo, 2022-04-03:
I am trying to route mixed signals from PADs to GPIOs located inside a Caravan template without any manual work. For this I made a custom NONDEFAULTRULE in LEF and I can observe that these rules are followed by both global and detailed router.

However, since the timings for these nets are disabled in SDC the observed routes are not the shortest paths, since there signals between PADs and GPIO module which are routed with shortest path, as they have timings enabled.

So I am currently trying to implement GPIO routing in the floorplanning stage, before placement (as the placement of GPIO is done manually) and then fixing them, preventing routers from modifying these nets. This would also allow user to see the nets and analyze the optimality before fully completing the flow.

I could just remove all nets from the DEF, do the routing, then add the routed nets back to DEF, but I thought if the router allowed to select the nets to route or prioritize this would solve this issue.

vijayank88, 2023-06-19:
@maliberty 
Any update on this request?

maliberty, 2023-06-19:
Another option would be a user defined high priority set of nets.

closed 2023-09-18 by maliberty