The-OpenROAD-Project/OpenROAD #1753: openroad build can't find_package(Qt5)

WS-Snyder, 2022-04-05:
cmake ..
indicates that Qt5 can't be found; tho Qt5 and qtcreator are working fine.
There is no findqt5.cmake but there is a findqt4.cmake from a previous install.
research indicates that qt5 no longer uses findxxx.cmake.
Is there a way to force the build to create a gui version, maybe using
the qtpaths shell command?

maliberty, 2022-04-05:
What version of cmake do you have?  Is your qt 5in the default location?

WS-Snyder, 2022-10-11:
The failure of cmake to build a GUI version is 100 % caused by
find_packages(Qt5) is return the paths(s).

I used cmake version 3.18.4 on Debian 11 (bullseye).

I will rebuild the entire tree on an Ubuntu 20.04.4 version and
see if the problem is resolved.

Thanks
Warren

maliberty, 2022-10-11:
Where is your QT installed?  If it is in a non-default location it may be that find_packages can't find it.

closed 2023-06-19 by vijayank88