The-OpenROAD-Project/OpenROAD #1758: Meta: Top Five Task You Wish Were Done Today

QuantamHD, 2022-04-06:
What are the top five tasks you wish were done today? i.e. 5 most impactful tasks for the project

maliberty, 2022-04-06:
My list
- Restructuring (rmp)
- Post-drt final fixing flow which will require incremental detailed routing.  This would replace the 'eco' flow in OL.
- Turn on the post-grt fixing flow in ORFS
- Enable RTL-MP and hierarchical synthesis with selective flattening
- Improved CTS (better clustering, GH-tree, improved buffer selection, smart NDRs)

QuantamHD, 2022-04-06:
@antonblanchard @arlpetergadfort Would love to hear from you as well

arlpetergadfort, 2022-04-07:
My list:

- CTS, the tree built something makes fixing setup/hold worse by poorly selecting buffers, improving CTS would help with that I think.
- Resizer setup repair gives up too soon when fixing setup violations, that could be improved with a better stop criteria, etc.
- Detail placement allowing specification of minimum gap (not just padding).

I'm sure there are more, but these are the ones at the front of my mind at the moment.

arlpetergadfort, 2022-04-07:
> @arlpetergadfort is the minimum gap to deal with the lack of 1 site filler in some pdks or something else?

@maliberty yes, exactly. Without it increasing density would become extremely difficult.

antonblanchard, 2022-04-08:
- +1 for "Post-drt final fixing flow which will require incremental detailed routing."
- +1 for "Improved CTS"
- VHDL/GHDL support. This is being worked on and I unfortunately haven't had time to test and offer feedback yet.