The-OpenROAD-Project/OpenROAD #1764: OpenROAD regressions with boundary checks [-Wp,-D_GLIBCXX_ASSERTIONS]

cbalint13, 2022-04-10:
There are no failing [regressions](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/test/regression) on latest a1cd122fe9bf2e6cb7b2b05be43c50dc5599d8f2 (2022/04/10) **unless** boundary checks are enabled.

**Suggestions:**
* for the betterment we could try enable ```-Wp,-D_GLIBCXX_ASSERTIONS``` compile flags and lets start investigate each failure.

* boundary check enabled environment also help daily development process acting as an immediate guard against odd things.

* as releng practice, such boundary check flag is **mandatory** & **enforced** by default on **all packages** in distros like RHEL / Fedora.

**Observations:**
* In contrast with OpenROAD, OpenSTA's [regressions](https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/test/regression.tcl) with boundary enabled **are fine** (as for now).

---

(x) Enabling boundary checks reveals these:
```
$ cat openroad-cxx-on.log | grep -e ERROR 
pd1 *ERROR* 
gcd_mem1_01 *ERROR* 
gcd_mem1_02 *ERROR* 
gcd_mem1_03 *ERROR* 
gcd_mem3_01 *ERROR* 
gcd_mem3_02 *ERROR* 
gcd_mem3_03 *ERROR* 
gcd_mem5_01 *ERROR* 
level3_01 *ERROR* 
level3_02 *ERROR* 
east_west1 *ERROR* 
east_west2 *ERROR* 
snap_layer1 *ERROR* 
repair_slew12 *ERROR* 
repair_antennas1 *ERROR* 
repair_antennas2 *ERROR* 
repair_antennas3 *ERROR* 
sw130_random *ERROR* 
sw130_random_simple *ERROR* 
gcd *ERROR* 
45_gcd *ERROR* 
```
(x) All the above fails due to boundary issues, e.x:
```
$ cat src/rcx/test/results/45_gcd.log  | grep Asser
/usr/include/c++/12/bits/stl_vector.h:1122: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>:
:operator[](size_type) [with _Tp = odb::tmg_rc; _Alloc = std::allocator<odb::tmg_rc>; 
reference = odb::tmg_rc&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

---

The [documentation](https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_macros.html) is way too summary, it covers more than enlisted there.
 A little sample (e.x on a generic std::vector) on how boundary checks works:

```
$ cat test1.cpp 
#include <vector>
#include <cstdio>

int main()
{
    std::vector<int> v = { 7, 5, 16, 8 };
    printf("v[-1] = %i\n", v[-1]);
}
```
```
$ cat test2.cpp 
#include <vector>
#include <cstdio>

int main()
{
    std::vector<int> v = { 7, 5, 16, 8 };
    printf("v[4] = %i\n", v[4]);
}
```

#### It runs, doesn't fail:

```
$ c++ test1.cpp -o test1; ./test1
v[-1] = 0

$ c++ test2.cpp -o test2; ./test2
v[4] = 0
```

#### Now failures are guaranteed:
```
$ c++ test1.cpp -Wp,-D_GLIBCXX_ASSERTIONS -o test1; ./test1
/usr/include/c++/12/bits/stl_vector.h:1122: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>
::operator[](size_type) [with _Tp = int; _Alloc = std::allocator<int>; reference = int&; 
size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```
```
$ c++ test2.cpp -Wp,-D_GLIBCXX_ASSERTIONS -o test2; ./test2
/usr/include/c++/12/bits/stl_vector.h:1122: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>
::operator[](size_type) [with _Tp = int; _Alloc = std::allocator<int>; reference = int&; 
size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```
---
Cc @maliberty 

Thanks,
~cristian.

maliberty, 2022-04-10:
We should definitely fix the issues this flag catches.  However it will have a runtime performance cost.  Are you saying every release package has this flag on all the time, not just during development?

cbalint13, 2022-04-10:
> We should definitely fix the issues this flag catches. 

* I will try involve and PR some of enumerated regression fixes when time permits.

> However it will have a runtime performance cost. 

* IMHE  for ```GLIBCXX_ASSERTIONS``` the cost isn't noticeable, compilers do a great job "hiding it".
* There is a discuss on it's performance concerns here: https://www.spinics.net/lists/fedora-devel/msg257877.html

> Are you saying every release package has this flag on all the time, not just during development?

* **YES** absolutely **all released** production .rpm packages in rhel/fedora land contains it (ever since such flags exists).

Using it even in final releases allows catching **complete new uncovered bugs** from the very users.

---

### Some more (offtopic) details:

* In RHEL land these are the **mandatory** flags (even many more) involved:
```
$ rpm --eval %{optflags}
-O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security 
-Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS  -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 
-fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic
-fasynchronous-unwind-tables  -fstack-clash-protection -fcf-protection
```

* A documentation of them is [here](https://src.fedoraproject.org/rpms/redhat-rpm-config//blob/rawhide/f/buildflags.md) .
* Beyond  ```GLIBCXX_ASSERTIONS``` there are: **stack-smash**, **origin-annotations**, **lto** and some more.
* The very ```GLIBCXX_ASSERTIONS``` is a **built-in** type, it's absence cannot be "negociated" with ```releg team``` :)

Thanks,
~cristian.

maliberty, 2022-04-11:
I've fixed a bunch of these (https://github.com/The-OpenROAD-Project/OpenROAD/pull/1765).  The last all seem to come down the tmg* code which is really bad code in odb.  Maybe its time to finally clean it up as fixing anything in there is excruciating.

proppy, 2022-04-12:
@cbalint13 do you recommend that we also enable this for https://github.com/hdl/conda-eda/tree/master/pnr/openroad? maybe that could be something enable for CMake for all Release build?

cbalint13, 2022-04-12:
> @cbalint13 do you recommend that we also enable this for https://github.com/hdl/conda-eda/tree/master/pnr/openroad? maybe that could be something enable for CMake for all Release build?

@proppy ,

You can for all else but not for ```openroad``` ,I suggest not yet, until all regression tests pass.

* For RHEL / Fedora ```opernroad``` [native packages](https://copr.fedorainfracloud.org/coprs/rezso/VLSI) I had to disable it.
* See disabling receipt line 81: https://copr-dist-git.fedorainfracloud.org/cgit/rezso/VLSI/openroad.git/tree/openroad.spec#n81

With it, ```openroad``` will fail on simple designs (e.g. the spm form openlane) at antenna-checking step.
It's enabled everywhere else (in that repo) except the ```openroad``` package.

maliberty, 2022-04-12:
@cbalint13 cvc issues should be filed at https://github.com/d-m-bailey/cvc

maliberty, 2022-04-13:
All designs pass but some of the ISPD routing benchmarks have issues that need looking into.

cbalint13, 2022-04-13:
> All designs pass but some of the ISPD routing benchmarks have issues that need looking into.

 * Nice ! That was very fast :)
 * Maybe offtopic (again), IMHO it't worth watching progress on ```-fanalyser``` in [GCC Land](https://developers.redhat.com/articles/2022/04/12/state-static-analysis-gcc-12-compiler) .

closed 2022-04-13 by maliberty