The-OpenROAD-Project/OpenROAD #1766: [ERROR GRT-0061] Layer NW is less than the min routing layer (METAL1)

gkamendje, 2022-04-11:
I the technology LEF file of my  platform, I do see the following layer definition:
```
LAYER NW
   TYPE MASTERSLICE ;
END NW
```

In the `pdn.cfg` file I  set a global connection to VDD on all VNW pins using:
```
set pdngen::global_connections {
  VDD {
    {inst_name .* pin_name VPWR}
    {inst_name .* pin_name VDD}
    {inst_name .* pin_name VNW}
  }
  VSS {
    {inst_name .* pin_name VGND}
    {inst_name .* pin_name VSS}
    {inst_name .* pin_name VPW}
  }
}
```
in the  platform `config.mk` file I do have the following 
```
export MIN_ROUTING_LAYER = METAL1
export MAX_ROUTING_LAYER = METAL5
```
and in `fastroute.tcl` I do have the following line
`set_routing_layers -signal $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER)`
I am getting the following error during placement:
```
[ERROR GRT-0061] Layer NW is less than the min routing layer (METAL1)
Error: fastroute.tcl, 1 GRT-0061.
```
It looks like layer NW is seen as a routing layer, whereas it should be seen as a power layer. How could I prevent the tool to see NW as a routing layer?  All the routing layers (metal1 to metal5) in the LEF file have `TYPE ROUTING;` in their definition.

vijayank88, 2022-04-11:
@gkamendje 
Can you please attach reproducible test case?
Are running the project through OpenROAD or OpenLane and share which version OpenROAD do you tried?

gkamendje, 2022-04-11:
@vijayank88 
I am running the project through OpenRoad. The LEF files I am using are proprietary. Is there a way to share the LEF files without publicly disclosing them?

vijayank88, 2022-04-11:
@gkamendje 
For proprietary files please check with @maliberty. They will help you on fixing this issue.

maliberty, 2022-04-11:
Where in your config is NW mentioned outside the LEF itself?  Are you using OpenROAD-flow-scripts or OpenLane or something else to run the tool?

gkamendje, 2022-04-11:
@maliberty 
I am using OpenROAD-flow-scripts. NW is not mentioned anywhere except in my tech LEF file. In the `pdn.cfg`, I did add the line  `{inst_name .* pin_name VNW} `  given that the macro LEF file defines the pins `VNW` which are used as power.

maliberty, 2022-04-11:
That error only comes from the define_layer_range, define_clock_layer_range, or set_global_routing_layer_adjustment commands based on the layer name(s) given.   You must be issuing one of those commands with NW as the argument.

I'm guessing NW is nwell and that is a strange place to have power pins.  Doesn't the macro have tapcells internally?  Are the VNW pins marked as power pins in the LEF?

Can you post your config.mk & pdn.cfg ?

maliberty, 2022-04-11:
Do you have anything else in your fastroute.tcl ?  I don't really think the pdn.cfg relates to this issue.
@tspyrou for thoughts on NDA

gkamendje, 2022-04-11:
I do not think that is related to pdn.cfg either. 
Here is the content of fastroute.tcl. set_routing_layers is commented out as you can see as a workaround the issue.  
```
set_global_routing_layer_adjustment $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER) 0.5
#set_routing_layers -signal $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER)
if {[info exist env(MACRO_EXTENSION)]} {
  set_macro_extension $::env(MACRO_EXTENSION)
}
```

maliberty, 2022-04-11:
would you add a puts "$::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER)" just to make sure the right values are set here and not NW.

gkamendje, 2022-04-11:
Put the suggested debug information at the beginning of fastroute.tcl show that the correct values are passed. I wonder how I could query the value of all layers found by the tool. 
```
[INFO ODB-0133]     Created 337 nets and 1094 connections.
[INFO ODB-0134] Finished DEF file: ./results/smic018/gcd/base/2_floorplan.def
1-5
[ERROR GRT-0061] Layer NW is less than the min routing layer (METAL1).
Error: fastroute.tcl, 2 GRT-0061

```

tspyrou, 2022-04-11:
@gkamendje if you post a public key I can give you access to an sftp server where you can upload private data. If you need us to sign an nda  we can do that.

maliberty, 2022-04-11:
What version of OR are you using?  That command takes layer names not integers for a long time now.

gkamendje, 2022-04-11:
@maliberty 
Good catch! The settings in the design config.mk (with the integers) was overriding what I had in the platform config.mk with strings.   
Thanks again

closed 2022-04-11 by gkamendje