The-OpenROAD-Project/OpenROAD #1779: Resizer/repair_design should Multi-VT Repair Flow.

QuantamHD, 2022-04-14:
This issue is for discussing the costs and benefits of a multi-Vt flow.

After some discussion internally at Google it seems like it would be relatively easy to swap in multi-Vt cells during repair design in order to improve PPA on the critical path.

The flow would be as follows

1. Identify the critical path. Replace all cells in critical path with LVT cells.
2. Rerun STA, if there's a new critical path (repeat step 1)
3. Identify the least critical path. Replace all cells with HVT cells. Do not replace cells if it would make the true critical path worse.
4. Rerun STA, if there's a new least critical path (repeat step 3).

End.

@maliberty @tspyrou Thoughts on such a flow?

tspyrou, 2022-04-14:
This flow should be executed after detailed routing with the VT swapping occurring between cells with identical physical footprints.  It can be done earlier in the flow but the critical path may change with the detailed routing so this is best.

QuantamHD, 2022-04-14:
That seems a bit more complicated. Could we start with the simple path based approach, and see what happens? I suspect there will still be a performance improvement albeit a little bit more muted.

tspyrou, 2022-04-14:
The Net based approach is actually simpler. Its a loop.
while there is a net with negative slack take the worst slack net and VT swap the net's driver until timing passes or all failing nets are at the fastest VT. No need to look at paths. The slack of all nets on the critical path is the same.

maliberty, 2022-04-14:
You'll have to make sure you aren't hurting hold while fixing setup.  Both algorithms look to be greedy but maybe that is a good place to start.

QuantamHD, 2022-04-14:
Okay sweet. The only other feedback I've gotten is that this is a nice first approach, but in the future it would be good to get power numbers in the optimization hull, and formulate it as an ILP problem, but that's a step for another day...

If we do add this optimization we should also make it the default flow for sky130 so all the MPW's benefit from the new flow.

maliberty, 2025-08-25:
We support multi-vt

closed 2025-08-25 by maliberty