The-OpenROAD-Project/OpenROAD #1797: Build hangs on Ubuntu20.04

varakely, 2022-04-19:
Everything went find until the build:
./etc/Build.sh
Instructions used: https://github.com/The-OpenROAD-Project/OpenROAD

It hangs on 58%.

Any advice please.

vvbandeira, 2022-04-19:
Can you provide the complete log?

varakely, 2022-04-19:
Thanks for prompt response. Unfortunately cant copy all log at the moment, here is the last part:
Can it be from limited available memory ? 

[ 58%] Linking CXX static library liblef.a
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lef && /usr/local/bin/cmake -P CMakeFiles/lef.dir/cmake_clean_target.cmake
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lef && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/lef.dir/link.txt --verbose=1
/usr/bin/ar qc liblef.a  CMakeFiles/lef.dir/lef/crypt.cpp.o CMakeFiles/lef.dir/lef/lefiArray.cpp.o CMakeFiles/lef.dir/lef/lefiCrossTalk.cpp.o CMakeFiles/lef.dir/lef/lefiDebug.cpp.o CMakeFiles/lef.dir/lef/lefiEncryptInt.cpp.o CMakeFiles/lef.dir/lef/lefiLayer.cpp.o CMakeFiles/lef.dir/lef/lefiMacro.cpp.o CMakeFiles/lef.dir/lef/lefiMisc.cpp.o CMakeFiles/lef.dir/lef/lefiNonDefault.cpp.o CMakeFiles/lef.dir/lef/lefiProp.cpp.o CMakeFiles/lef.dir/lef/lefiPropType.cpp.o CMakeFiles/lef.dir/lef/lefiTBExt.cpp.o CMakeFiles/lef.dir/lef/lefiUnits.cpp.o CMakeFiles/lef.dir/lef/lefiVia.cpp.o CMakeFiles/lef.dir/lef/lefiViaRule.cpp.o CMakeFiles/lef.dir/lef/lef_keywords.cpp.o CMakeFiles/lef.dir/lef/lefrCallbacks.cpp.o CMakeFiles/lef.dir/lef/lefrData.cpp.o CMakeFiles/lef.dir/lef/lefrReader.cpp.o CMakeFiles/lef.dir/lef/lefrSettings.cpp.o CMakeFiles/lef.dir/lef/lefwWriterCalls.cpp.o CMakeFiles/lef.dir/lef/lefwWriter.cpp.o CMakeFiles/lef.dir/lef_parser.cpp.o
/usr/bin/ranlib liblef.a
make[2]: Leaving directory '/home/ubuntu/bin/OpenROAD/build'
[ 58%] Built target lef
/usr/bin/make -f src/odb/src/lefin/CMakeFiles/lefin.dir/build.make src/odb/src/lefin/CMakeFiles/lefin.dir/depend
make[2]: Entering directory '/home/ubuntu/bin/OpenROAD/build'
cd /home/ubuntu/bin/OpenROAD/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/ubuntu/bin/OpenROAD /home/ubuntu/bin/OpenROAD/src/odb/src/lefin /home/ubuntu/bin/OpenROAD/build /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin/CMakeFiles/lefin.dir/DependInfo.cmake --color=
Scanning dependencies of target lefin
make[2]: Leaving directory '/home/ubuntu/bin/OpenROAD/build'
/usr/bin/make -f src/odb/src/lefin/CMakeFiles/lefin.dir/build.make src/odb/src/lefin/CMakeFiles/lefin.dir/build
make[2]: Entering directory '/home/ubuntu/bin/OpenROAD/build'
[ 58%] Building CXX object src/odb/src/lefin/CMakeFiles/lefin.dir/lefin.cpp.o
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin && /usr/bin/g++  -DBOOST_ALL_NO_LIB -DSPDLOG_COMPILED_LIB -I/home/ubuntu/bin/OpenROAD/src/odb/include -I/usr/include/tcl -I/home/ubuntu/bin/OpenROAD/src/odb/include/odb -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lef -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lefzlib -I/home/ubuntu/bin/OpenROAD/src/utl/include  -O3 -DNDEBUG -fPIC   -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -std=c++17 -o CMakeFiles/lefin.dir/lefin.cpp.o -c /home/ubuntu/bin/OpenROAD/src/odb/src/lefin/lefin.cpp
[ 58%] Building CXX object src/odb/src/lefin/CMakeFiles/lefin.dir/reader.cpp.o
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin && /usr/bin/g++  -DBOOST_ALL_NO_LIB -DSPDLOG_COMPILED_LIB -I/home/ubuntu/bin/OpenROAD/src/odb/include -I/usr/include/tcl -I/home/ubuntu/bin/OpenROAD/src/odb/include/odb -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lef -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lefzlib -I/home/ubuntu/bin/OpenROAD/src/utl/include  -O3 -DNDEBUG -fPIC   -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -std=c++17 -o CMakeFiles/lefin.dir/reader.cpp.o -c /home/ubuntu/bin/OpenROAD/src/odb/src/lefin/reader.cpp
[ 58%] Building CXX object src/odb/src/lefin/CMakeFiles/lefin.dir/lefTechLayerSpacingEolParser.cpp.o
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin && /usr/bin/g++  -DBOOST_ALL_NO_LIB -DSPDLOG_COMPILED_LIB -I/home/ubuntu/bin/OpenROAD/src/odb/include -I/usr/include/tcl -I/home/ubuntu/bin/OpenROAD/src/odb/include/odb -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lef -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lefzlib -I/home/ubuntu/bin/OpenROAD/src/utl/include  -O3 -DNDEBUG -fPIC   -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -std=c++17 -o CMakeFiles/lefin.dir/lefTechLayerSpacingEolParser.cpp.o -c /home/ubuntu/bin/OpenROAD/src/odb/src/lefin/lefTechLayerSpacingEolParser.cpp

vvbandeira, 2022-04-19:
The build script run in parallel, the error will be before the section you pasted. Please re-run as follows:
```
./etc/Build.sh -clean | tee build.log
```
Then upload the build.log file here.

vijayank88, 2022-04-21:
@varakely 
Try this way, it may helps:
If you installed `dependencies` with https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies
follow `Build by hand` https://github.com/The-OpenROAD-Project/OpenROAD#build-by-hand

varakely, 2022-04-26:
It still stops on 58%
[ 58%] Building CXX object src/odb/src/lefin/CMakeFiles/lefin.dir/lefTechLayerSpacingEolParser.cpp.o
cd /home/ubuntu/bin/OpenROAD/build/src/odb/src/lefin && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DSPDLOG_COMPILED_LIB -I/home/ubuntu/bin/OpenROAD/src/odb/include -I/usr/include/tcl -I/home/ubuntu/bin/OpenROAD/src/odb/include/odb -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lef -I/home/ubuntu/bin/OpenROAD/src/odb/src/lef/lefzlib -I/home/ubuntu/bin/OpenROAD/src/utl/include  -O3 -DNDEBUG -fPIC   -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -std=c++17 -o CMakeFiles/lefin.dir/lefTechLayerSpacingEolParser.cpp.o -c /home/ubuntu/bin/OpenROAD/src/odb/src/lefin/lefTechLayerSpacingEolParser.cpp

vvbandeira, 2022-04-26:
> The build script runs in parallel, the error will be before the section you pasted. Please re-run as follows:
> 
> ```
> ./etc/Build.sh -clean 2>&1 | tee build.log
> ```
> 
> Then upload the build.log file here.

@varakely please run as instructed and provide the full log file, without it we do not have enough information to understand the problem.

varakely, 2022-04-27:
@vvbandeira I am using AWS EC2 Ubuntu20.04 
grep MemTotal /proc/meminfo
MemTotal:         992204 kB

df -k .
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        8065444 6079276   1969784  76% /

varakely, 2022-05-05:
Yes the problem came from small server resources. Successfully built on .large server.

closed 2022-05-05 by vvbandeira