The-OpenROAD-Project/OpenROAD #1805: OpenROAD GUI SEGVs when loading DEF

antonblanchard, 2022-04-25:
The DEF in question likely has issues, but we shouldn't SEGV when trying to load it. (The DEF came out of the caravel project, I need to look at how it was generated.)

There is a lot of complaining before the SEGV:

```
[WARNING ODB-0245] error: Net VGND: Can not determine which direction to continue path,
[INFO ODB-0276]        via (via2_280_280) spans above and below the current layer (met2).
...
[WARNING ODB-0110] error: invalid VIA layers, cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers, cannot determine exit layer of path
```

The backtrace of the SEGV:

```
(gdb) backtrace 
#0  0x000000000150d480 in odb::dbTechLayer::getWidth() const ()
#1  0x00000000014bcbe8 in odb::dbWireShapeItr::next(odb::dbShape&) ()
#2  0x000000000149950c in odb::dbWire::getBBox(odb::Rect&) ()
#3  0x00000000013e1520 in odb::dbBlock::ComputeBBox() ()
#4  0x00000000013e168d in odb::dbBlock::getBBox() ()
#5  0x00000000018a6c62 in gui::LayoutViewer::getBounds() const ()
#6  0x00000000018ab591 in gui::LayoutViewer::hasDesign() const ()
#7  0x00000000018ab9f1 in gui::LayoutViewer::updateScaleAndCentering(QSize const&) ()
#8  0x00000000018abcdc in gui::LayoutViewer::setBlock(odb::dbBlock*) ()
#9  0x00007ffff6d38430 in void doActivate<false>(QObject*, int, void**) () at /lib64/libQt5Core.so.5
#10 0x000000000188b962 in gui::Search::newBlock(odb::dbBlock*) ()
#11 0x00007ffff6d38430 in void doActivate<false>(QObject*, int, void**) () at /lib64/libQt5Core.so.5
#12 0x000000000188b91f in gui::MainWindow::designLoaded(odb::dbBlock*) ()
#13 0x0000000000fc781d in ord::OpenRoad::readDef(char const*, bool, bool, bool) ()
```

Test case: [openroad-gui.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8554408/openroad-gui.tar.gz)

antonblanchard, 2022-04-25:
I'm guessing it was created by magic.

maliberty, 2022-04-25:
It has bad vias, e.g.:
   - via2_280_280
      + RECT via2 ( -140 -140 ) ( 140 140 )
      + RECT via3 ( -140 -140 ) ( 140 140 ) ;

two cuts and no enclosure.  Still we need a better error message instead.

PR #1807 (2022-04-25): odb: add sanity checks to definVia::viaEnd for malformed via
Fixes #1805
commit ed495608 (2022-04-25): odb: add sanity checks to definVia::viaEnd for malformed via

Fixes #1805
closed 2022-04-25 by maliberty