The-OpenROAD-Project/OpenROAD #1809: Microwatt gets SEGV in antenna repair

antonblanchard, 2022-04-26:
I tried enabling antenna repair on the Microwatt design and hit a SEGV. A test case: [antenna-fail.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8558902/antenna-fail.tar.gz)

Backtrace of the SEGV:

```
#0  0x00000000017ff47a in grt::GlobalRouter::addLocalConnections(std::map<odb::dbNet*, std::vector<grt::GSegment, std::allocator<grt::GSegment> >, grt::cmpById, std::allocator<std::pair<odb::dbNet* const, std::vector<grt::GSegment, std::allocator<grt::GSegment> > > > >&) ()
#1  0x00000000018131aa in grt::GlobalRouter::repairAntennas(sta::LibertyPort*, int) ()
#2  0x00000000018249a2 in _wrap_repair_antennas ()
#3  0x00007ffff7e2db52 in TclNRRunCallbacks () from /lib64/libtcl8.6.so
#4  0x00007ffff7e32bea in ?? () from /lib64/libtcl8.6.so
#5  0x00007ffff7e33417 in Tcl_EvalEx () from /lib64/libtcl8.6.so
#6  0x00007ffff7e3343a in Tcl_Eval () from /lib64/libtcl8.6.so
#7  0x00000000011f8a6c in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) ()
#8  0x0000000000fd7904 in ord::tclAppInit(Tcl_Interp*) ()
#9  0x00007ffff7ee5a59 in Tcl_MainEx () from /lib64/libtcl8.6.so
#10 0x0000000000f82015 in main ()
```

maliberty, 2022-04-26:
Partial fix in https://github.com/The-OpenROAD-Project/OpenROAD/pull/1811

jjcherry56, 2022-06-29:
I did a lot of work on antenna repair lately and it does not error any more.
The sky130hd LEF has some problems with antenna support.
I updated the OR versions but not ORFS or openlane.
Specifically, MACRO sky130_ef_sc_hd__fakediode_2 should be
 CLASS CORE ANTENNACELL 
and the PIN DIODE is missing
ANTENNADIFFAREA  0.434700 ;
sky130hs has issues also.

But with those changes OR still struggles with repairing the antennas. I haven't looked into exactly why but the diodes don't push the ratios into safety like they should.

antonblanchard, 2022-06-29:
@jjcherry56 thank you!

`sky130_ef_sc_hd__fakediode_2` is a bit weird in that it is used by other (non openroad) antenna fixup strategies in openlane, and is eventually either removed or replaced with a real diode. Having said that, `sky130_fd_sc_hd__diode_2` does need the `CLASS CORE ANTENNACELL` fix.

jjcherry56, 2022-06-29:
sky130_fd_sc_hd__diode_2 should have class CLASS CORE ANTENNACELL and NOT have ANTENNAGATEAREA.

closed 2022-07-01 by jjcherry56

jjcherry56, 2022-07-03:
BTW, pull request #2010 fixes a major bug in antenna repair that was causing it to add excessive diodes. it actually works on microwatt now. you will have to move filler placement to after antenna repair though.