The-OpenROAD-Project/OpenROAD #1834: gui::show crashes

varakely, 2022-05-05:
Re-install OpenRoad doesn't help

OpenROAD v2.0-3517-gc519b10f8 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> gui::show
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Stack trace:
 0# 0x000055CF3D6AF0A1 in build/src/openroad
 1# 0x00007FBD7D7290C0 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x00007FBD7E201AAD in /lib/x86_64-linux-gnu/libQt5Core.so.5
 5# QGuiApplicationPrivate::createPlatformIntegration() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 6# QGuiApplicationPrivate::createEventDispatcher() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 7# QCoreApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Core.so.5
 8# QGuiApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
 9# QApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
10# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in build/src/openroad
11# gui::Gui::showGui(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in build/src/openroad
12# show(char const*, bool) in build/src/openroad
13# 0x000055CF3E044785 in build/src/openroad
14# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
15# 0x00007FBD7F466924 in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
18# ord::tclAppInit(Tcl_Interp*) in build/src/openroad
19# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
20# main in build/src/openroad
21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
22# _start in build/src/openroad

vvbandeira, 2022-05-05:
On #1797 you mentioned you were using AWS cloud, is this still the case? What is your X environment? Are you using VNC?

varakely, 2022-05-05:
@vvbandeira, yes I am running in AWS and ssh from RealVNC session with proper DISPLAY set. Thanks.

ilesser, 2022-05-22:
I am having the same issue after I built the open road flow with docker, using master updated as of today.

For reproducing the error I simply load the docker image and then open openroad enablign the gui.
The setup_env.tcl has no changes from master. This is the log I get and this is the run_docker.sh:

```
docker \
   run \
   --tty \
   --interactive \
   --user $(id -u ${USER}):$(id -g ${USER}) \
   --net=host \
   --ipc=host \
   --env DISPLAY=${DISPLAY} \
   --volume $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
   --volume $(pwd)/flow/scripts:/OpenROAD-flow-scripts/flow/scripts:ro \
   --volume ${HOME}:/home/${USER} \
   openroad/flow-scripts \
   bash
```

```
ilesser@zenbook:OpenROAD-flow > ./run_docker.sh 
bash-4.2$ source setup_env.sh 
OPENROAD: /OpenROAD-flow-scripts/tools/OpenROAD
bash-4.2$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
xkbcommon: ERROR: failed to add default include path /usr/share/X11/xkb
Qt: Failed to create XKB context!
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
process 10: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Stack trace:
 0# 0x0000000001032356 in openroad
 1# 0x00007F445B183400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F44507F5655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007F4453AABBFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007F445C4B4A55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007F445C503A03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007F445811B478 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F4453AA93B5 in /lib64/libQt5DBus.so.5
19# 0x00007F445C30AE71 in /lib64/libQt5Core.so.5
20# 0x00007F445DCC8EA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
QThread::wait: Thread tried to wait on itself
QMutex: destroying locked mutex
QThread: Destroyed while thread is still running
Stack trace:
 0# 0x0000000001032356 in openroad
 1# 0x00007F445B183400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F445C2F7CE1 in /lib64/libQt5Core.so.5
 5# QThread::~QThread() in /lib64/libQt5Core.so.5
 6# 0x00007F4453AA95D9 in /lib64/libQt5DBus.so.5
 7# 0x00007F445B186CE9 in /lib64/libc.so.6
 8# 0x00007F445B186D37 in /lib64/libc.so.6
 9# 0x000000000103237B in openroad
10# 0x00007F445B183400 in /lib64/libc.so.6
11# gsignal in /lib64/libc.so.6
12# abort in /lib64/libc.so.6
13# 0x00007F44507F5655 in /lib64/libdbus-1.so.3
14# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
15# 0x00007F4453AABBFB in /lib64/libQt5DBus.so.5
16# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
17# 0x00007F445C4B4A55 in /lib64/libQt5Core.so.5
18# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
19# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
20# 0x00007F445C503A03 in /lib64/libQt5Core.so.5
21# g_main_context_dispatch in /lib64/libglib-2.0.so.0
22# 0x00007F445811B478 in /lib64/libglib-2.0.so.0
23# g_main_context_iteration in /lib64/libglib-2.0.so.0
24# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
25# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
26# QThread::exec() in /lib64/libQt5Core.so.5
27# 0x00007F4453AA93B5 in /lib64/libQt5DBus.so.5
28# 0x00007F445C30AE71 in /lib64/libQt5Core.so.5
29# 0x00007F445DCC8EA5 in /lib64/libpthread.so.0
30# clone in /lib64/libc.so.6
```

I checked /usr/share/X11 and the xkb directory is missing.

Could this be a problem with the docker image generation? Could something be missing from the way the image is generated?

ilesser, 2022-05-22:
I don't think the error is related to X server because I can succesfully open klayout without issues:

[image]

So the problem is definetely related to Qt 5

maliberty, 2022-05-22:
Try install xkb or using the openlane docker image.

ilesser, 2022-05-22:
@maliberty I didn't know what to install so I installed this:

```
yum install libxkbcommon libxkbcommon-devel libxkbfile libxkbfile-devel xorg-x11-xkb-extras xorg-x11-xkb-utils xorg-x11-xkb-utils-devel xfce4-xkb-plugin.x86_64
```

After that the error is slightly different:

```
[root@zenbook OpenROAD-flow-scripts]# openroad  -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
process 296: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Stack trace:
 0# 0x0000000001032356 in openroad
 1# 0x00007FBA7F31A400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007FBA7498C655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007FBA77C42BFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007FBA8064BA55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007FBA8069AA03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007FBA7C2B2478 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007FBA77C403B5 in /lib64/libQt5DBus.so.5
19# 0x00007FBA804A1E71 in /lib64/libQt5Core.so.5
20# 0x00007FBA81E5FEA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
QThread::wait: Thread tried to wait on itself
QMutex: destroying locked mutex
QThread: Destroyed while thread is still running
Stack trace:
 0# 0x0000000001032356 in openroad
 1# 0x00007FBA7F31A400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007FBA8048ECE1 in /lib64/libQt5Core.so.5
 5# QThread::~QThread() in /lib64/libQt5Core.so.5
 6# 0x00007FBA77C405D9 in /lib64/libQt5DBus.so.5
 7# 0x00007FBA7F31DCE9 in /lib64/libc.so.6
 8# 0x00007FBA7F31DD37 in /lib64/libc.so.6
 9# 0x000000000103237B in openroad
10# 0x00007FBA7F31A400 in /lib64/libc.so.6
11# gsignal in /lib64/libc.so.6
12# abort in /lib64/libc.so.6
13# 0x00007FBA7498C655 in /lib64/libdbus-1.so.3
14# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
15# 0x00007FBA77C42BFB in /lib64/libQt5DBus.so.5
16# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
17# 0x00007FBA8064BA55 in /lib64/libQt5Core.so.5
18# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
19# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
20# 0x00007FBA8069AA03 in /lib64/libQt5Core.so.5
21# g_main_context_dispatch in /lib64/libglib-2.0.so.0
22# 0x00007FBA7C2B2478 in /lib64/libglib-2.0.so.0
23# g_main_context_iteration in /lib64/libglib-2.0.so.0
24# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
25# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
26# QThread::exec() in /lib64/libQt5Core.so.5
27# 0x00007FBA77C403B5 in /lib64/libQt5DBus.so.5
28# 0x00007FBA804A1E71 in /lib64/libQt5Core.so.5
29# 0x00007FBA81E5FEA5 in /lib64/libpthread.so.0
30# clone in /lib64/libc.so.6
```

I am using a docker image I built using the build_openroad.sh with the lastest master from today.

Is there a way to download a docker image already built by OpenRoad or by someone else?

arlpetergadfort, 2022-05-22:
You could try to get more of the QT error information by setting `export QT_DEBUG_PLUGINS=1`
This might help you see if something is missing.

ilesser, 2022-05-22:
Here are my logs obtained with `openroad -gui 2>&1 |tee qt_debug.log`:

* `qt_debug.without_xkb.log`: Log before running the `yum install` command
* `qt_debug.with_xkb.log`: Log after running the `yum install` command

The install command is the same I showed earlier: `yum install libxkbcommon libxkbcommon-devel libxkbfile libxkbfile-devel xorg-x11-xkb-extras xorg-x11-xkb-utils xorg-x11-xkb-utils-devel xfce4-xkb-plugin.x86_64`

If I do a diff between both files they are identical  except the log before installing xkb has these three extra lines:

```
xkbcommon: ERROR: failed to add default include path /usr/share/X11/xkb
Qt: Failed to create XKB context!
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
```

[image]

This confirms the xkb will not affect the result.

ilesser, 2022-05-22:
I tried recompiling the docker images from the tag `v2.0` becauser I thought that would be more stable, but there are many errors during the image building so I couldn't do it.

Just to clarify, this issue is only related to the GUI. The whole flow works well until it wants to use the GUI

ilesser, 2022-05-22:
After some debugging I see that the crash leaves TCL in [line 139 of src/gui/src/gui.tcl](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gui/src/gui.tcl#L139):

```tcl
  gui::save_image $path {*}$area $resolution $options
```
After that it goes to [line 276 src/gui/src/gui.i](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gui/src/gui.i#L276):

```c
void save_image(const char* filename, double xlo, double ylo, double xhi, double yhi, double dbu_per_pixel = 0, const std::map<std::string, bool>& display_settings = {})
{
  auto gui = gui::Gui::get();
  gui->saveImage(filename, make_rect(xlo, ylo, xhi, yhi), dbu_per_pixel, display_settings);
}
```
An finally to [line 479 of src/gui/src/gui.cpp](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/gui/src/gui.cpp#L479):

```c
void Gui::saveImage(const std::string& filename, const odb::Rect& region, double dbu_per_pixel, const std::map<std::string, bool>& display_settings)
{
  if (db_ == nullptr) {
    logger_->error(utl::GUI, 15, "No design loaded.");
  }
  odb::Rect save_region = region;
  const bool use_die_area = region.dx() == 0 || region.dy() == 0;
  const bool is_offscreen = main_window->testAttribute(Qt::WA_DontShowOnScreen) /* if not interactive this will be set */ || !enabled();
```

And when it calls the `testAttribute` method it crashes for some reason.

I am no expert in C++ and I have never worked with Qt, so I don't think I can debug it anymore.

Is there anyone with Qt experience who could continue debugging this issue?

### How to reproduce this error
1. Clone the [OpenRoad-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts) commit   5583993563718c17ce94bc251a666dc433ca3fa2
2. Change directory `cd OpenROAD-flow-scripts`
3. Run `./build_openroad.sh` to build the docker images
4. Run `./run_docker.sh` provided in my previous commit
5. Run `cd flow`
6. Run `make`

```
ilesser@zenbook:OpenROAD-flow-scripts > cat /etc/os-release
NAME="Linux Mint"
VERSION="19.1 (Tessa)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.1"
VERSION_ID="19.1"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tessa
UBUNTU_CODENAME=bionic
ilesser@zenbook:OpenROAD-flow-scripts > docker --version
Docker version 18.09.5, build e8ff056
```

maliberty, 2022-05-22:
There was very recently a bug that turned up in save_images.  Please try with the latest code to make sure you have that fix.

arlpetergadfort, 2022-05-22:
@maliberty I don't think it has anything to do with save_image since @ilesser said at first this issue arose when calling `openroad -gui` which does not call any of that code referenced.
I believe, but correct me if I'm wrong, kLayout bundles Qt so the system Qt installation would not matter.

Looking around online there appears to be similar issues (https://forum.qt.io/topic/93247/qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found/21) one possible solution would be to ensure you have `sudo apt-get install libxcb-xinerama0`

ilesser, 2022-05-23:
@maliberty I am using the latest version of OpenRoad-flow-scripts. I will try later forcing OpenROAD app to the latest version.

@arlpetergadfort Thanks for the tip, I tried intalling `libxcb-xinerama0` but couldn't find it. Later I will try to add more sources to yum to see if I can install it.

maliberty, 2022-05-23:
@vvbandeira can you build a docker image with build_openroad.sh and try this out?

vvbandeira, 2022-05-23:
Yes. I will give it a try @maliberty

vvbandeira, 2022-05-23:
@ilesser can you try installing the following packages in the docker image and then running `openroad -gui` again?
```
yum install -y xorg-x11-server-Xorg xorg-x11-xauth mesa-dri-drivers
```

ilesser, 2022-05-23:
@vvbandeira I tried installing that but I still get a similar error.

Here is the log I get: 
[error.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/8757468/error.log)

ilesser, 2022-05-23:
I also tied `yum distribution-synchronization` but it didn't help. I get the same error

ilesser, 2022-05-23:
@vvbandeira  Adding those lines to the docker command work!

Now I can open the GUI without any errors and I didn't have to install anything, it works with the image I compiled.

```
ilesser@zenbook:OpenROAD-flow-scripts > ./run_docker.sh 
non-network local connections being added to access control list
[root@zenbook OpenROAD-flow-scripts]# source setup_env.sh 
OPENROAD: /OpenROAD-flow-scripts/tools/OpenROAD
[root@zenbook OpenROAD-flow-scripts]# openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
xkbcommon: ERROR: failed to add default include path /usr/share/X11/xkb
Qt: Failed to create XKB context!
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
libGL error: unable to load driver: i965_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: i965
libGL error: unable to load driver: i965_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: i965
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
```

This is my run_docker.sh:

```
xhost +local:docker
docker \
   run \
   --tty \
   --interactive \
   --net=host \
   --ipc=host \
   --env DISPLAY \
   --volume $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
   --volume $(pwd)/flow/scripts:/OpenROAD-flow-scripts/flow/scripts:ro \
   --volume /tmp/.X11-unix:/tmp/.X11-unix \
   --volume ${HOME}/.Xauthority:/.Xauthority \
   --security-opt seccomp=unconfined \
   --privileged \
   --volume ${HOME}:/home/${USER} \
   openroad/flow-scripts \
   bash

xhost -local:docker
```

vvbandeira, 2022-05-23:
@ilesser that is great. You will probably want to install those packages, in my testing I was not able to use the keyboard before installing them. You can also try to remove one by one of the options I gave to you to see which are really needed. @varakely can you try the proposed solutions here to see if they also apply to you?

ilesser, 2022-05-23:
That is correct @vvbandeira keyboard was not working at first, but after running `yum install -y xorg-x11-server-Xorg xorg-x11-xauth mesa-dri-drivers` it worked!

I want to say that I have been using Cadence tools for a decade now and I am pretty impressed by the openROAD flow.
Kudos to everyone that contributed! I would like to try to contribute to the development.

maliberty, 2022-05-23:
I'm glad you are up and running.  You are most welcome to contribute in whatever fashion interests you.

vvbandeira, 2022-06-05:
@varakely were you able to use the GUI so we can close the issue?

closed 2022-06-07 by vvbandeira