The-OpenROAD-Project/OpenROAD #1836: initFloorplan() always snaps to grid

antonblanchard, 2022-05-05:
I was looking through warnings during a tape out and saw:

```
[WARNING IFP-0028] Core area lower left (5.520, 10.880) snapped to (5.520, 10.880).
```

I printed out the values, and it does look like the size of my macro is not a multiple of site:

```
clx 5520
core_lx 5.5199999860633405e-06
cly10880 
core_ly 1.0879999972530641e-05
```

But it also seems like the comparison will never be true, because we are comparing a value that has been converted to Dbu:

```
      int clx = divCeil(metersToDbu(core_lx), site_dx) * site_dx;
      int cly = divCeil(metersToDbu(core_ly), site_dy) * site_dy;

      if (clx != core_lx || cly != core_ly) {
```

I presume this is harmless, but it would be nice to fix so the warning goes away.

maliberty, 2022-05-19:
I was fixing up ifp for python and I think I've eliminated this problem in the process (both values are dbu now).  Would you verify this is ok to close?  There is no test case.

antonblanchard, 2022-10-03:
Thanks @maliberty, I'm no longer seeing this.

closed 2022-10-03 by antonblanchard