The-OpenROAD-Project/OpenROAD #1838: Differentiable Placement Technique

QuantamHD, 2022-05-06:
Top Line Performance Numbers:
33% improvement in WNS

Have we looked at this technique? https://guozz.cn/publication/tdpdac-22/tdpdac-22.pdf

maliberty, 2022-05-06:
I'll look at the paper but note that dreamplace is not timing driven so WNS is not something they directly target.  Replace in OR has a timing driven mode so it may not show the same delta.

maliberty, 2022-05-07:
It relies on elmore delay which is less accurate.  It makes no mention of Ceff so I don't know if they just assume it is a second order effect or what.  It doesn't look extensible or CCS or SI.  With all that said it might be sufficient for global placement where accuracy is less crucial.  Dreamplace 4.0 is not public so it is hard to say what they are really comparing against in their table of results.

vijayank88, 2023-06-19:
@maliberty 
Can we close this  as `not planned`?

closed 2023-06-19 by maliberty