The-OpenROAD-Project/OpenROAD #1840: dpl: standard cells flipped

antonblanchard, 2022-05-06:
I tried taping out Microwatt/Caravel using Openlane with a recent OpenROAD and https://github.com/The-OpenROAD-Project/OpenLane/pull/1059 applied. I'm seeing a strange problem where all standard cells appear to be flipped, such that they are powered backwards.

Test case is at:  [reverse-dpl.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8643804/reverse-dpl.tar.gz)

antonblanchard, 2022-05-06:
I started looking through dpl, and noticed a few things:

- `Opendp::checkPowerLine` doesn't appear to check any height=1 cells at all
- We go to a lot of trouble to work out where the power and ground rails are (eg `Opendp::findRowPower`). Do we need to do that or just look at the orientation out of odb?

antonblanchard, 2022-05-07:
I'm not sure if it matters but there are an odd number of met1 rails in this design

maliberty, 2022-05-07:
findRowPower is just wrong as it is going to look at the ring not the first row in your test case.  Probably it worked before by luck depending on which ring is first.

maliberty, 2022-05-07:
> findRowPower is just wrong as it is going to look at the ring not the first row in your test case. Probably it worked before by luck depending on which ring is first.

Its more than just that.  I agree orientation should be enough.

closed 2022-05-08 by maliberty

antonblanchard, 2022-05-08:
Thanks @maliberty, I can confirm it fixed the original issue.

PR #1842 (2022-05-08): dpl: use row orientation and drop bogus multi-height tests
Dpl shouldn't be trying to figure out power connections - it should just
obey the row's orientation.  That is accomplished by storing it on the Pixel.

There is quite incompete and incorrect multi-height cell support.  The
test cases are invalid as they only have a single site which makes no
sense for multi-height support.  I have removed them and we should
address this  separately when there is demand.

Fixes #1840
commit 30900f44 (2022-05-08): dpl: use row orientation and drop bogus multi-height tests

Dpl shouldn't be trying to figure out power connections - it should just
obey the row's orientation.  That is accomplished by storing it on the Pixel.

There is quite incompete and incorrect multi-height cell support.  The
test cases are invalid as they only have a single site which makes no
sense for multi-height support.  I have removed them and we should
address this  separately when there is demand.

Fixes #1840