The-OpenROAD-Project/OpenROAD #1848: problem with abc

ripudamank2, 2022-05-10:
root@embedded:/home/embedded/Documents/OpenROAD/third-party# cd abc/
root@embedded:/home/embedded/Documents/OpenROAD/third-party/abc# ls
root@embedded:/home/embedded/Documents/OpenROAD/third-party/abc# ls -al
total 8
drwxr-xr-x 2 root root 4096 maj 10 23:33 .
drwxr-xr-x 3 root root 4096 maj 10 23:33 ..
root@embedded:/home/embedded/Documents/OpenROAD/third-party/abc# 

when  i tried to do git pull for openroad , it was showing empty folder for abc but in git repo there are files .
Can you please tell how i can get abc folder data .

maliberty, 2022-05-10:
git submodule update --init --recursive

ripudamank2, 2022-05-10:
yes i did it . Now its compiling . thank you .

Is there any tutorial how to use Openroad for ASIC design ?

maliberty, 2022-05-10:
You'll want to take a look at a flow that uses openroad.  You could try [OpenROAD-flow-scripts](https://openroad.readthedocs.io/en/latest/user/GettingStarted.html) or [OpenLane](https://openlane.readthedocs.io/en/latest/)

ripudamank2, 2022-05-10:
if i want to integrate other PDK , how i can do that like ASP7 or 45nm pdk?

maliberty, 2022-05-10:
https://openroad.readthedocs.io/en/latest/contrib/PlatformBringUp.html

closed 2022-05-10 by ripudamank2