The-OpenROAD-Project/OpenROAD #1853: DRT fails to find a path

VidyaChhabria, 2022-05-12:
Hi,

I modified ORFS to add fixes (repair_timing) post GR. In this case, the attached testcase fails to find a path between pin and source aps. See below. 

Testcase:

```
[INFO DRT-0195] Start 0th optimization iteration.
Failed to find a path between pin  and source aps:
( 299 195 5 ) (Idx) / ( 3780 60624 ) (coords)
extBBox (xDim, yDim, zDim) = (456, 710, 10)
Failed to find a path between pin  and source aps:
( 3 398 3 ) (Idx) / ( 12 85548 ) (coords)
( 6 398 3 ) (Idx) / ( 60 85548 ) (coords)
( 25 398 3 ) (Idx) / ( 300 85548 ) (coords)
extBBox (xDim, yDim, zDim) = (456, 710, 10)
[ERROR DRT-0255] Maze Route cannot find path of net clk_i in worker of routeBox ( 0 60480 ) ( 3780 64260 ).
[ERROR DRT-0255] Maze Route cannot find path of net clk_i in worker of routeBox ( 0 83160 ) ( 3780 86940 ).
```

VidyaChhabria, 2022-05-12:
@WenjingJiang21 FYI.

Stephanommg, 2022-05-16:
The problem is solved and the fix is going through some tests before it is merged into the master branch.

Stephanommg, 2022-05-18:
The fix is on the master branch now.

closed 2022-05-18 by Stephanommg