The-OpenROAD-Project/OpenROAD #1856: Resizer crash with set_max_delay

egorxe, 2022-05-13:
It seems that resizer can't correctly handle internal combinational paths constrained with set_max_delay. I am attaching OpenLane packaged example containing a line of SKY130 buffers constrained like this:
```
set_max_delay -from [get_pins {genblk1[1].buffer/X}] -to [get_pins {genblk1[6].buffer/A}] 0.1
```
This leads to a crash during *repair_timing -setup* command with the following stack trace:
```
 0# 0x000055DAC23FF819 in openroad
 1# 0x00007EFD5192C840 in /lib/x86_64-linux-gnu/libc.so.6
 2# std::vector<sta::TimingArc*, std::allocator<sta::TimingArc*> >::size() const in openroad
 3# sta::TimingArc::cornerArc(int) in openroad
 4# rsz::Resizer::repairSetup(sta::PathRef&, float) in openroad
 5# rsz::Resizer::repairSetup(float, int) in openroad
 6# rsz::repair_setup(float, int) in openroad
 7# 0x000055DAC26362A2 in openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007EFD538C33AF in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
13# 0x000055DAC2400138 in openroad
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# main in openroad
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in openroad
```

Crash comes from NULL dereference in the following line, but I'm not sure where to progress from here:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/8cd8fd14357244ec6b5d53b91a6249accdac2832/src/rsz/src/Resizer.cc#L2562

[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8689911/openroad_issue_reproducible.tar.gz)

maliberty, 2022-05-14:
@vvbandeira hit a similar issue with bp_dual.

jjcherry56, 2022-05-14:
Once you resubmit this as a tcl script that can be run with openroad I will look at it.  It should not depend on a shell script for variable state.

egorxe, 2022-05-14:
> Without running it I can see that the set_max_delay has invalid from/to pins (they are not valid start/end points). They would have to use -through to be valid. But that is certainly no excuse to seg fault.

Do you consider only input pins of combinational cells as valid start points and outputs as end points? Commercial EDAs I worked with supported inputs and outputs for both (as well as hierarchical pins in non flat designs). This limitation is not a problem, but there is no way to know about it, may be some kind of warning should be produced then SDC is loaded?

jjcherry56, 2022-05-15:
my bad, the start/endpoints are fine; it is mcp and false paths that have restrictions.
fixed in 50ec2e3c5e36b0c9fc0bf55b2a3f8b3a51936b43

closed 2022-05-15 by jjcherry56