The-OpenROAD-Project/OpenROAD #1868: AntennaRepair inserts diodes within blockage around hard macros

leon575777642, 2022-05-17:
### Problem

Function `AntennaRepair::insertDiode` places antenna-fixing diodes and avoids overlapping with existing insts. If a "legal" location is found, the inserted diode is placed in "FIRM" status ([src/grt/src/AntennaRepair.cpp:299](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/grt/src/AntennaRepair.cpp#L299)) and therefore will not be legalized by detailed placement. However, if the existing insts are hard macros, this function does not check for the blockages around the macros. As a result, the diodes may be inserted next to a hard macro where no power straps are available due to the blockage. Such diodes will fail placement checking later during the antenna fix.

### Temporary Solution

Comment out src/grt/src/AntennaRepair.cpp:297-300, 302 . All inserted diodes are thus always legalized by the detailed placer. I have a fork that implements this fix, but I assume this is not the ideal solution, so I did not create a PR.

### Context

I was working on a MPW-5 project (now carried over to MPW-6) which contains a 2-level hierarchical user design (not counting the caravel wrapper). In my openlane `config.tcl` for the top-level user design, I tried different antenna fixing strategies and none worked well. Strategy 3 using the FastRoute antenna fixer seems the most promising and reasonable strategy, yet it always fails and reports illegal placements during antenna fixing. The temporary solution worked for me, although there are still a few hundreds of violations left at the end of the flow (other strategies either end up with thousands of violations, or insert a ton of diodes so the design is not routable).

maliberty, 2022-05-17:
Please provide a test case demonstrating the issue.

leon575777642, 2022-05-17:
Hi @maliberty, sure I would be happy to create a reproducable test case. I am unsure what to include though. Could you give me some instructions on how to create a reproducable test case?

In addition, our repo is available publicly (https://github.com/getziadz/caravel_mpw5_prga). The problem can be reproduced by running `make prga` at the root of the repo.

maliberty, 2022-05-17:
See https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

leon575777642, 2022-05-17:
Great! Here is the auto-generated reproducible package. Let me know if you need any other files.

[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8711715/openroad_issue_reproducible.tar.gz)

maliberty, 2022-05-19:
The rows are cut around the macro according to the halo so there are no legal placement sites inside the halo.  I can't see any instances of an antenna diode inside the halo.  Would you provide an example of what you mean in your test case.

maliberty, 2022-05-19:
Diodes in yellow:
[image]

maliberty, 2022-05-19:
The diode must in a row to be marked FIRM:
```
  legally_placed = legally_placed && diodeInRow(inst_rect);
...
  if (core_area.contains(inst_rect) && !sink_inst->getMaster()->isBlock()
      && legally_placed) {
    antenna_inst->setPlacementStatus(odb::dbPlacementStatus::FIRM);
```

leon575777642, 2022-05-19:
Hi @maliberty thanks for looking into this. The error I got was the following:

```
[INFO GRT-0012] Antenna violations: 2181
[WARNING GRT-0054] Placement of diode ANTENNA_138 will be legalized by detailed placement.
...
[INFO GRT-0015] 2912 diodes inserted.
[WARNING DPL-0004] Placed in rows check failed (2).
[ERROR DPL-0033] detailed placement checks failed.
```

Due to the error I did not get the def with the inserted diodes, so I cannot verify if the diodes are indeed placed in the halo. Is there any way to get the def with the illegal diodes?

One thing to note is that I was using commit `8d53e9b018dec98fa63e907ddeb6c5406f035361` since that is what MPW-6 uses in their docker image.

maliberty, 2022-05-19:
8d53e9b018dec98fa63e907ddeb6c5406f035361 is from Feb 17.  When I run with the current OR head I don't see the above errors.  Why is MPW-6 so far behind?  @donn

maliberty, 2022-05-19:
I see the mpw-6b tag points at a slightly later version.

vijayank88, 2023-06-19:
@angl-dev 
Please test with latest commit, re-open an issue if problem repeats means with reproducible test case.

closed 2023-06-19 by vijayank88