The-OpenROAD-Project/OpenROAD #1869: segfault error

vijayank88, 2022-05-17:
Using currently: OpenROAD v2.0-3478-g1dcf04bbf 

Trying to use @antonblanchard design from https://github.com/antonblanchard/microwatt-mpw5/blob/main/verilog/rtl/multiply_add_64x64.v

Got failed during global routing, with below error:

```
[INFO GRT-0001] Minimum degree: 3
[INFO GRT-0002] Maximum degree: 3
[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 23326
[INFO GRT-0009] rerouting 2 nets.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace:
 0# 0x000000000102BD66 in openroad
 1# 0x00007F40900B3400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# sta::Table2::findValue(float, float, float) const in openroad
 5# sta::GateTableModel::gateDelay(sta::LibertyCell const*, sta::Pvt const*, float, float, float, bool, float&, float&) const in openroad
 6# sta::LumpedCapDelayCalc::gateDelay(sta::LibertyCell const*, sta::TimingArc*, float const&, float, sta::Parasitic*, float, sta::Pvt const*, sta::DcalcAnalysisPt const*, float&, float&) in openroad
 7# sta::DmpCeffDelayCalc::gateDelay(sta::LibertyCell const*, sta::TimingArc*, float const&, float, sta::Parasitic*, float, sta::Pvt const*, sta::DcalcAnalysisPt const*, float&, float&) in openroad
 8# rsz::Resizer::gateDelays(sta::LibertyPort*, float, sta::DcalcAnalysisPt const*, float*, float*) in openroad
 9# rsz::Resizer::bufferDelay(sta::LibertyCell*, sta::RiseFall const*, float, sta::DcalcAnalysisPt const*) in openroad
10# rsz::Resizer::addWireAndBuffer(sta::Vector<rsz::BufferedNet*>, rsz::SteinerTree*, int, int, int, double) in openroad
11# rsz::Resizer::rebufferBottomUp(rsz::SteinerTree*, int, int, int, double) in openroad
12# rsz::Resizer::rebufferBottomUp(rsz::SteinerTree*, int, int, int, double) in openroad
13# rsz::Resizer::rebuffer(sta::Pin const*) in openroad
14# rsz::Resizer::repairSetup(sta::PathRef&, float) in openroad
15# rsz::Resizer::repairSetup(float, int) in openroad
16# 0x000000000116C060 in openroad
17# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
18# 0x00007F409308B36C in /lib64/libtcl8.5.so
19# TclObjInterpProcCore in /lib64/libtcl8.5.so
20# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
21# 0x00007F409308B36C in /lib64/libtcl8.5.so
22# Tcl_ExprObj in /lib64/libtcl8.5.so
23# Tcl_ExprBooleanObj in /lib64/libtcl8.5.so
24# 0x00007F4093051F40 in /lib64/libtcl8.5.so
25# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
26# 0x00007F409308B36C in /lib64/libtcl8.5.so
27# 0x00007F4093093647 in /lib64/libtcl8.5.so
28# TclEvalObjEx in /lib64/libtcl8.5.so
29# 0x00007F40930CE27F in /lib64/libtcl8.5.so
30# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
31# 0x00007F409308B36C in /lib64/libtcl8.5.so
32# 0x00007F4093093647 in /lib64/libtcl8.5.so
33# TclEvalObjEx in /lib64/libtcl8.5.so
34# 0x00007F409304E1D0 in /lib64/libtcl8.5.so
35# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
36# 0x00007F409308B36C in /lib64/libtcl8.5.so
37# 0x00007F4093093647 in /lib64/libtcl8.5.so
38# TclEvalObjEx in /lib64/libtcl8.5.so
39# 0x00007F4093051F00 in /lib64/libtcl8.5.so
40# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
41# 0x00007F409308B36C in /lib64/libtcl8.5.so
42# TclObjInterpProcCore in /lib64/libtcl8.5.so
43# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
44# 0x00007F409308B36C in /lib64/libtcl8.5.so
45# TclObjInterpProcCore in /lib64/libtcl8.5.so
46# 0x00007F4093046EB2 in /lib64/libtcl8.5.so
47# 0x00007F4093047F1E in /lib64/libtcl8.5.so
48# Tcl_EvalEx in /lib64/libtcl8.5.so
49# Tcl_Eval in /lib64/libtcl8.5.so
50# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
51# ord::tclAppInit(Tcl_Interp*) in openroad
52# Tcl_Main in /lib64/libtcl8.5.so
53# main in openroad
54# __libc_start_main in /lib64/libc.so.6
55# 0x0000000001028727 in openroad
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/multiply_add_64x64/runs/RUN_2022.05.17_09.26.43/logs/routing/15-resizer.log
[ERROR]: Last 10 lines:
child process exited abnormally

```

Attached test case: 
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8707363/openroad_issue_reproducible.zip)

maliberty, 2022-05-17:
Note the seg fault is not from global routing but from repair_timing -setup

jjcherry56, 2022-05-18:
fixed by 1250124454c8741c67973cf5c96c75e73fbfe8c9

closed 2022-05-18 by jjcherry56