The-OpenROAD-Project/OpenROAD #1872: Support for MEOL layer?

maliberty, 2022-05-17:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/1871

<div type='discussions-op-text'>

<sup>Originally posted by **GuzTech** May 17, 2022</sup>
Hello everyone!

I'm in the process of bringing up a new platform which is going well. I am running into a problem with a local interconnect layer which is specified as a `MASTERSLICE` with `LEF58_TYPE` = `MEOL`, so it is not specified as a `ROUTING` layer as is the case with the sky130 PDK. OpenDB complains that there is a parse mismatch for this `lil` layer.

If I add this `lil` layer as a routing layer to `make_tracks.tcl` then OpenDB complains that it is not a routing layer. If I don't add `lil` as a routing layer, then TritonRoute gives an error that it cannot find layer `lil` for viarule that connects `lil` to the first metal layer.

Did anyone else run into a similar problem? If so, how did you solve it?</div>

maliberty, 2022-05-17:
@osamahammad21 we are missing a few LEF58_TYPE values including MEOL in lefin/lefTechLayerTypeParser.cpp.  Please guide Mohamed for adding all the missing values to odb.  (You can reassign this to him once he accepts the project invite).

GuzTech, 2022-05-18:
Just wanted to let you know that I added `MEOL` to the enum `LEF58_TYPE` in `db.h` and the following to `lefin/lefTechLayerTypeParser.cpp:74`:

```
             | lit("MEOL")[boost::bind(&odb::dbTechLayer::setLef58Type,
             							 layer,
             							 odb::dbTechLayer::LEF58_TYPE::MEOL)])
```

This did remove the warning but TritonRoute still cannot find the `lil` layer but I do not know if that if that is related to this issue.

osamahammad21, 2022-05-18:
That's a different issue. we should update TritonRoute to consider MEOL masterslice layers as well. for now, TritonRoute supports only one hard-coded masterslice layer with the name "FR_MASTERSLICE" I am fixing that now in parallel with supporting more LEF58_TYPE values.

GuzTech, 2022-05-24:
I just pulled the latest version of OpenROAD and I can confirm that at least OpenDB does not complain about the MEOL layer anymore.

closed 2022-05-30 by osamahammad21

GuzTech, 2022-05-31:
Thanks for updated version! :+1: I again pulled in the latest version of OpenROAD, but I still have the same problem. I'll quote from the discussion that lead to this issue:
```
> The section in the LEF file looks like this:
> 
> ```
> LAYER lil
> 	TYPE MASTERSLICE;
> 	PROPERTY LEF58_TYPE "TYPE MEOL;";
> END lil
> ```
> 
> But then TritonRoute gives me an error:
> 
> ```
> [ERROR DRT-0129] Unknown layer lil for viarule lil_m1.
> ```
> 
> The part in the LEF file:
> 
> ```
> VIARULE lil_m1 GENERATE 
> 	LAYER lil;
> 		XXX
> 	LAYER m1;
> 		XXX
> 	LAYER ctc;
> 		XXX
> END lil_m1 
> ```
> 
> where the `ctc` layer is of type `CUT`.

```

Is this related to this issue, or is it something else and should I create a different issue for it?

GuzTech, 2022-06-05:
Interestingly, after #1928 I now get a different error:
```
[ERROR DRT-0129] Unknown layer act for viarule Nact_m1.
```
where the relevant parts in the LEF file are:
```
LAYER act
    TYPE MASTERSLICE;
    PROPERTY LEF58_TYPE "TYPE DIFFUSION;";
    PROPERTY LEF58_SPACING "SPACING XXX;";
END act

VIARULE Nact_m1
    LAYER act;
        XXX;
    LAYER m1;
        XXX;
    LAYER cnt;
        XXX;
END Nact_m1
```

Since I saw that you removed the comparison with `LEF58_TYPE::MEOL` in the function `addMasterSliceLayer()`, I actually removed the comparison with NWELL, PWELL and DIFFUSION as well since I have master slices that are of these types too. But that also gives the same error.

osamahammad21, 2022-06-06:
@GuzTech as I told you before we only support one masterSlice layer per tech in tritonroute. So in the end you would have the layers stacked as following:
Masterslice layer
Cut layer
Routing layer
Cut layer
...
No other masterslice layer in between layers.

GuzTech, 2022-06-07:
@osamahammad21 Ah ok, I misunderstood when you first explained it then. That means that for now we cannot use our PDK with OpenROAD for P&R.

Thanks for the hard work! :+1:

osamahammad21, 2022-06-07:
@GuzTech Is your APK public?
If yes could you share it please?
If not, could you give me an idea on how the layers are stacked so that I could understand what needs to be supported?

GuzTech, 2022-06-08:
@osamahammad21 Thank you for wanting to look into it! It's a commercial PDK so unfortunately I cannot share it, but I can give an overview of how layers are stacked. There are two types of via rules that are like this:

```
Routing (metal layer number X)
Routing (metal layer number Y)
Cut (via layer number Z)
```
and
```
Masterslice (DIFFUSION / PWELL / NWELL / MEOL / POLY (typeless))
Routing (metal layer number 1)
Cut (metal contact layer)
```

Does that help? Just let me know if you need any more information :+1:

maliberty, 2022-06-14:
Are you actually routing signals outside the stdcells on the MEOL (or POLY) layers or is it just a matter of successfully ignoring them?

GuzTech, 2022-06-14:
@maliberty We just give it a Verilog design and call flow.tcl. The standard cells use a contact, metal #1, and nwell layer, but also have several statements that reference the lil layer which is the MEOL layer:

- ANTENNAPARTIALMETALAREA
- ANTENNADIFFAREA
- ANTENNAMAXAREACAR
- etc.

Again, thanks for taking the time. Much appreciated! If you need any more info, I'll be happy to help.

maliberty, 2022-06-15:
I'm trying to understand if we can just ignore that layer and associated vias or not.  

What layer are the std cell pins on in LEF?  What are the immediately above cut & routing layers.  Where is you MEOL layer in relation to those?

GuzTech, 2022-06-15:
The standard cell i/o pins are on the metal 1 and contact layers, and power pins are on the NWELL / PWELL layers. The lil layer (MEOL) is only referenced in the statements I mentioned before. In the stack, the MEOL layer apparently is between the highest two metal layers.