The-OpenROAD-Project/OpenROAD #1875: PDN has issues with thin areas of met1 rails

antonblanchard, 2022-05-18:
I wanted to see how OpenROAD handles very small areas between the perimeter of the core area and placed macros. I took a working design (40um macro halo with the macros 40um in from the left and right edges), and changed to a 20um halo, such that we have 20um of space for met1 rails. The design went from passing LVS to failing.

Although I haven't root caused it to the PDN stage, I do see some strange things. Firstly there are areas where some met4 straps appear to be missing:

[image]

Another question is how these small areas between the perimeter of the core and macros should be handled (should we avoid placing met1 rails or add met4 straps that are closer spaced than our PDN config to power it).

It looks like we run a met4 strap very close to the other one in order to power it:

[image]

Test case: 
[pdn.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8712941/pdn.tar.gz)

maliberty, 2022-05-18:
@arlpetergadfort comments?

antonblanchard, 2022-05-18:
I'm closer to understanding why LVS failed. It looks like extraction hasn't connected VPB to VPWR in these areas, eg:

```
Xoutput71 output71/A vssd1 vssd1 output71/VPB vccd1 io_out[34] sky130_fd_sc_hd__buf_2
```

I'm not yet sure if this is real, or a magic extraction issue.

antonblanchard, 2022-05-18:
I notice there are no well tap cells, just end caps and filler cells:

[image]

arlpetergadfort, 2022-05-18:
> @arlpetergadfort comments?

The grid looks okay when spot checking. The missing straps were removed because they didn't connect to anything so their presence is not need. I see connections down met1 for both VDD and VSS, so they should be connected. 

What I also see is PDNGEN added a huge number of ports for power and ground on the edges, which are probably not needed, so I might take a look at that later unless it is contributing to the issue.

antonblanchard, 2022-05-18:
> The grid looks okay when spot checking. The missing straps were removed because they didn't connect to anything so their presence is not need. I see connections down met1 for both VDD and VSS, so they should be connected.

The straps are indeed unused (There are 4 voltage domains and only 1 is used in this design). It is a bit interesting that in some areas all the unused met4 straps were removed (eg down the bottom), but at the top one unused strap is still there, and at the lower left 3 extra ones are left. In all cases the unused voltage domains are not connected to anything (apart from the power ring)

I guess the proper solution is to use the recent option to avoid laying down unused straps in the first place.

> What I also see is PDNGEN added a huge number of ports for power and ground on the edges, which are probably not needed, so I might take a look at that later unless it is contributing to the issue.

Yeah definitely not a high priority. I now think my LVS issue are the lack of well tap cells and not PDN generation.

antonblanchard, 2022-05-19:
I think this is the reason LVS is failing. you can see end cap cells, but no well tap cells:

[image]

I guess it's a bit of a corner case, since it's such a thin area, but shouldn't we be adding well tap cells?

Test case: [tap.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8728536/tap.tar.gz)

maliberty, 2022-05-19:
Any row should always have at least a tap cell or endcap (containing a welltap) otherwise it won't function.

antonblanchard, 2022-05-19:
I wasn't sure if the cells we use for end caps on sky130hd (`sky130_fd_sc_hd__decap_3`) provide a welltap or not (eg if the layout is just missing pin and label layers that magic can use for extraction).

Even if it did, there's no for OpenROAD to know is there? As such, It seems like it should have added tap cells.

I think I see why we aren't adding them. We try to, but the overlap check fails:

```
      const int site_width = row->getSite()->getWidth();
      for (int x = llx + offset; x < urx; x += pitch) {
        x = odb::makeSiteLoc(x, site_width, 1, llx);
        // Check if site is filled
        const int tap_width = tapcell_master->getWidth();
        odb::dbOrientType ori = row->getOrient();
        bool overlap = checkIfFilled(x, tap_width, ori, row_fill_check);
        if (!overlap) {
          const int lly = row_bb.yMin();
          makeInstance(block, tapcell_master, ori, x, lly, tap_prefix_);
        }
      }
```

maliberty, 2022-05-19:
I believe tapcell assumes that endcaps contain a welltap

maliberty, 2022-05-19:
There is no problem here.  You have set the decap as your endcap and they are within the 13um tapcell dist.  The only issue is choosing decap as a endcap which it is not really.

maliberty, 2022-05-19:
Btw, if you are going to push your macros close to the edge you should avoid putting top level pins in that tight routing channel if possible.
[image]

antonblanchard, 2022-05-23:
> There is no problem here. You have set the decap as your endcap and they are within the 13um tapcell dist. The only issue is choosing decap as a endcap which it is not really.

```
# welltap and endcap cells
set ::env(FP_WELLTAP_CELL) "sky130_fd_sc_hd__tapvpwrvgnd_1"
set ::env(FP_ENDCAP_CELL) "sky130_fd_sc_hd__decap_3"
```

@rtimothyedwards should we be using something other than `decap_3` for the endcap cells? (eg something with a tap)