The-OpenROAD-Project/OpenROAD #1877: How to use mpl2 module?

dearmrlv, 2022-05-18:
I build the OpenROAD app successfully, but when I ran mp_test1.tcl under ./src/mpl2/test/, I met errors. The log info is as follows:
$ openroad < mp_test1.tcl
OpenROAD v2.0-3615-gf4a99dadf
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./Nangate45/Nangate45_tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  ./Nangate45/Nangate45_tech.lef
[INFO ODB-0222] Reading LEF file: ./Nangate45/Nangate45.lef
[WARNING ODB-0217] duplicate VIARULE (Via1Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-3) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-4) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-3) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-4) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via4Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via5Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via6Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via7Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via8Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via9Array-0) ignoring...
[INFO ODB-0225]     Created 134 library cells
[INFO ODB-0226] Finished LEF file:  ./Nangate45/Nangate45.lef
[INFO ODB-0222] Reading LEF file: ./Nangate45/fake_macros.lef
[INFO ODB-0225]     Created 2 library cells
[INFO ODB-0226] Finished LEF file:  ./Nangate45/fake_macros.lef
[WARNING STA-0160] ./Nangate45/fake_macros.lib line 32, default_max_transition is 0.0.
[INFO ODB-0127] Reading DEF file: ./testcases/mp_test1_fp.def
[INFO ODB-0128] Design: mp_test1
[INFO ODB-0252]     Updated 8 pins.
[INFO ODB-0253]     Updated 10 components.
[INFO ODB-0134] Finished DEF file: ./testcases/mp_test1_fp.def
Running Partition Design...
[INFO PAR-0402] Traversed logical hierarchy
        Number of std cell instances: 0
        Total area: 160000
        Number of hard macros: 10
[INFO PAR-0490] Number of hops:  0
[INFO PAR-0490] Number of hops:  1
[INFO PAR-0490] Number of hops:  2
[INFO PAR-0490] Number of hops:  3
[INFO PAR-0403] Number of Clusters created: 17
Start RTL-MP
[INFO MPL-0009] RTL-MP param: min_aspect_ratio: 0.5.
[INFO MPL-0009] RTL-MP param: dead_space: 0.1.
[INFO MPL-0009] RTL-MP param: learning_rate: 0.
[INFO MPL-0009] RTL-MP param: shrink_factor: 0.997.
[INFO MPL-0009] RTL-MP param: shrink_freq: 0.1.
[INFO MPL-0009] RTL-MP param: halo_width: 0.
[INFO MPL-0009] RTL-MP param: region_file: macro_blockage.txt.
[INFO MPL-0009] RTL-MP param: location_file: location.txt.
[INFO MPL-0009] RTL-MP param: num_thread: 10.
[INFO MPL-0009] RTL-MP param: num_run: 5.
[INFO MPL-0009] RTL-MP param: heat_rate: 0.1.
[INFO MPL-0009] RTL-MP param: num_level: 3.
[INFO MPL-0009] RTL-MP param: num_worker: 10.
[INFO MPL-0009] RTL-MP param: alpha: 6.
[INFO MPL-0009] RTL-MP param: beta: 40.
[INFO MPL-0009] RTL-MP param: gamma: 100.
[INFO MPL-0009] RTL-MP param: boundary_weight: 2.
[INFO MPL-0009] RTL-MP param: macro_blockage_weight: 1.
[INFO MPL-0009] RTL-MP param: location_weight: 100.
[INFO MPL-0009] RTL-MP param: notch_weight: 212.
[INFO MPL-0009] RTL-MP param: resize_prob: 0.3.
[INFO MPL-0009] RTL-MP param: pos_swap_prob: 0.3.
[INFO MPL-0009] RTL-MP param: neg_swap_prob: 0.3.
[INFO MPL-0009] RTL-MP param: double_swap_prob: 0.1.
[INFO MPL-0009] RTL-MP param: init_prob: 0.95.
[INFO MPL-0009] RTL-MP param: rej_ratio: 0.95.
[INFO MPL-0009] RTL-MP param: k: 5000000.
[INFO MPL-0009] RTL-MP param: c: 1000.
[INFO MPL-0009] RTL-MP param: snap_layer: 4.
[INFO MPL-0009] RTL-MP param: max_num_step: 500.
[INFO MPL-0009] RTL-MP param: perturb_per_step: 60.
[INFO MPL-0009] RTL-MP param: seed: 0.
/tools/cluster-software/gcc/gcc-9.3.0/include/c++/9.3.0/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::__cxx11::basic_string<char>; _Alloc = std::allocator<std::__cxx11::basic_string<char> >; std::vector<_Tp, _Alloc>::reference = std::__cxx11::basic_string<char>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace:
 0# 0x0000559D06982B87 in openroad
 1# 0x00007F9E96813400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# std::char_traits<char>::assign(char&, char const&) in openroad
 5# std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::operator[](unsigned long) in openroad
 6# shape_engine::ParseBlockFile(std::vector<shape_engine::Cluster*, std::allocator<shape_engine::Cluster*> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float&, float&, float&, float&, utl::Logger*, float, float) in openroad
 7# shape_engine::ShapeEngine(float&, float&, float&, float&, float, float, float, utl::Logger*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int, unsigned int) in openroad
 8# mpl::rtl_macro_placer(char const*, utl::Logger*, odb::dbDatabase*, char const*, float, float, float, float, float, float, float, float, char const*, char const*, char const*) in openroad
 9# mpl::MacroPlacer2::place(char const*, char const*, float, float, float, float, float, float, float, float, char const*, char const*, char const*) in openroad
10# mpl2::rtl_macro_placer_cmd(char const*, char const*, float, float, float, float, float, float, float, float, char const*, char const*, char const*) in openroad
11# 0x0000559D07DE522F in openroad
12# 0x00007F9E98C22AD4 in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
13# 0x00007F9E98C6998F in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
14# TclObjInterpProcCore in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
15# 0x00007F9E98C22AD4 in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
16# 0x00007F9E98C6998F in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
17# 0x00007F9E98C6FA69 in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
18# TclEvalObjEx in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
19# Tcl_RecordAndEvalObj in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
20# Tcl_Main in /workspace/S/lvzhengyang/_install/lib/libtcl8.5.so
21# main in openroad
22# __libc_start_main in /lib64/libc.so.6
23# 0x0000559D06964262 in openroad

Appreciate for any help!

dearmrlv, 2022-05-18:
I also tried to built the project via gcc-8.5.0, but the same error appeared. Below are the versions of my dependencies:
tcltk-8.5
boost-1.76.0
spdlog-1.10.0

dearmrlv, 2022-05-18:
Another question, is mpl2 providing the same function as mpl?

maliberty, 2022-05-18:
mpl2 is intended to replace mpl once it is fully qualified.    Can you provide a test case for the crash?

I doubt those dependencies have any relation to the crash.

dearmrlv, 2022-05-18:
the test case is in the master branch of OpenROAD project, ./src/mpl2/test/mp_test1.tcl, 
https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/mpl2/test/mp_test1.tcl
and I built the project successfully with the dependencise mentioned above.
I got this crush when I run the following command under OpenROAD/src/mpl2/test: 
openroad < mp_test1.tcl

I found that in.the file OpenROAD/test/regression, line 46, the test of mpl2 is commented. (https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/test/regression)
I compiled the project with LOAD_PARTITIONERS OFF, maybe it's the reason of the crush.
But with LOAD_PARTITIONERS ON, cmake will report error for needing extra modules; Chaco, Metis.
I downloaded these packages and I don't know how to embed them into mpl2.

dearmrlv, 2022-05-19:
I located the bug. The bug is that the TCL command: `rtl_macro_placer -report_file` cannot correctly set user define value to var `report_file` in C++ code, and `report_file` remains its default value which is wrong in the test case.
I configure the corresponding code in OpenROAD/src/mpl2/src/mpl.tcl, line 70 manually, and re-compile the while project, and finally I ran mpl2 without error.
I'm not sure if it is due to my compile progress causing such error...

maliberty, 2022-05-19:
#1887 fixes -report_file

dearmrlv, 2022-05-20:
Another **ERROR**.
When running the test case from OpenROAD/test/aes_nangate45.tcl with mpl2 utilized, errors occured. The log attached below:
[logs.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/8735632/logs.txt)

And here is the test case
[partition.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8735642/partition.zip)

unzip the test case file and enter the "partition" folder, run `openroad < aes_nangate45_2.tcl`

I guess the error locates in OpenROAD/src/mpl2/src/block_placement.cpp, line 1746. When runing thread of FastSA, error occurs.

maliberty, 2022-05-20:
I'll take a look but not that aes has no macros so mpl2 is kinda pointless.

commit 87a9fb11 (2022-05-23): mpl2: skip with there are no macros to place

Fixes #1877
closed 2022-05-23 by maliberty