The-OpenROAD-Project/OpenROAD #1884: Flow failed to complete global routing with routing congestion

vijayank88, 2022-05-19:
`Description`
This is MPW-5 passed design. WIth latest openroad its failed to complete global routing.

MPW-5 repo: https://github.com/ISL-ECE-CEG-AU/GPS_Baseband/blob/main/openlane/user_project_wrapper/config.tcl

`Logs`

```
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 2128
[INFO GRT-0198] Via related Steiner nodes: 2
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 3510
[INFO GRT-0112] Final usage 3D: 98518
[ERROR GRT-0118] Routing congestion too high.
Error: groute.tcl, 55 GRT-0118

```

Attaching Test case here: 
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8727975/openroad_issue_reproducible.zip)

maliberty, 2022-05-19:
Macros should bring their pins to the outer edge and block their core (especially analog macros), not:
[image]

maliberty, 2022-07-05:
Is there any further update here or should this be closed?

vijayank88, 2022-07-06:
@maliberty 
Its a question. Whether Tritonroute will handle analog and digital blocks together to route.?
MPW-5 repo: https://github.com/ISL-ECE-CEG-AU/GPS_Baseband/blob/main/openlane/user_project_wrapper/config.tcl
Above the user repo and used openlane flow for routing the design. But now facing congestion issue.
I am not able to understand your statement: `Macros should bring their pins to the outer edge and block their core (especially analog macros), not:`

maliberty, 2022-07-06:
No TR is not an analog router.  Analog blocks should have a proper LEF

vijayank88, 2022-07-06:
Thanks @maliberty . Then closing this issue.

closed 2022-07-06 by vijayank88