The-OpenROAD-Project/OpenROAD #1885: Resizer routing timing failing with routing congestion

vijayank88, 2022-05-19:
`Description`
This is MPW-5 passed design. WIth latest openroad its failed to complete global routing.

MPW-5 repo: https://github.com/htfab/asicle/blob/main/openlane/user_project/config.tcl

`Logs`

```
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 294161
[INFO GRT-0198] Via related Steiner nodes: 37900
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 552784
[INFO GRT-0112] Final usage 3D: 3626431
[ERROR GRT-0118] Routing congestion too high.
Error: resizer_routing_timing.tcl, 53 GRT-0118

```

Reproducible test case: 
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8728060/openroad_issue_reproducible.zip)

maliberty, 2022-05-19:
We had this discussion elsewhere - this is not a helpful description of the problem.  In MPW-5 you may have generated different input to grt and the problem is elsewhere.    You need to identify where in the flow the divergence starts and file an issue about that.  I need a case where the old results and the new results for the same input differ substantially.

You have an area of very high placement density that looks suspect:
[image]

vijayank88, 2022-06-22:
@maliberty 
I tried with different density, still its failing.
Check CI for `asicle` in https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Funstable-small-1/detail/PR-51/1/pipeline/287

maliberty, 2022-06-22:
Same comment as before, please isolate the issue

vijayank88, 2022-07-21:
@maliberty 
For testing purpose I've increased the DIE_AREA but placement not spreading cells as expected. Increased CORE_UTIL
Attached latest test case 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9159569/issue_reproducible.zip)

maliberty, 2022-07-21:
Increasing the die area doesn't reduce placement density.   PLACE_DENSITY controls the placement density.

vijayank88, 2022-07-21:
PL_TARGET_DENSITY is `0.30`

maliberty, 2022-07-21:
@vijayank88 you test case is packaged at global routing and therefore isn't useful for debugging a global placement issue.

maliberty, 2022-07-21:
Since OL uses decap cells are fillers the congestion will always be unrealistic in the heatmap once they have been inserted (since decaps are not recognized as fill cells).  If I remove them I see:

[image]

still above 30% but more accurate.

vijayank88, 2022-10-11:
Yes. I used 0.30 as target density