The-OpenROAD-Project/OpenROAD #1888: OpenRoad fails to route ispd19_test6 with 1 macro pin corner spacing violation.

stefanottili, 2022-05-19:
OpenRoad fails to route ispd19_test6 with a macro pin corner spacing violation.
There is plenty of space to solve this pin access as can be seen on all the other pins of all the other macros.

get input lef/def from https://www.ispd.cc/contests/19/

OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -threads 8 -log log -gui ispd19_test6.or

ispd19_test6.or:

read_lef ispd19_test6.input.lef.gz
read_def ispd19_test6.input.def.gz
#set_thread_count 8
global_route -verbose
detailed_route -output_drc 5_route_drc.rpt -verbose 1
write_db route.db
exit

OpenROAD v2.0-3644-gc852a71d0 
....
[INFO DRT-0195] Start 64th optimization iteration.
[INFO DRT-0199]   Number of violations = 1.
[INFO DRT-0267] cpu time = 02:30:36, elapsed time = 00:20:51, memory = 8211.41 (MB), peak = 8287.22 (MB)

5_route_drc.rpt:

  violation type: CornerSpc
    srcs: net:net49701 inst:inst8396 
    bbox = ( 1207.35, 483.185 ) - ( 1207.63, 483.265 ) on Layer Metal6

maliberty, 2022-05-19:
We run this design nightly and it ends with 0 DRVs.  See https://jenkins.openroad.tools/job/OpenROAD-Nightly-Public/654/execution/node/478/log/

I'm not sure why you see a violation.

maliberty, 2022-05-19:
The tests are run with https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/drt/test/run-ispd.py

stefanottili, 2022-05-20:
I'm running on a M1 with 16GB and logs using 4,6,8 cores look nearly identical.

But comparing the 32-thread nightly log to my M1 logs, these are the differences I can spot:

a) OpenROAD v2.0-3644-gc852a71d0 vs OpenROAD v2.0-3674-g6bcd930ce
b) global_route was run without -verbose vs I ran with
c) Number of guides:     4718916  vs  Number of guides:     1906703
d) [INFO DRT-0176] GCELLGRID X -1 DO 905 STEP 3000 ; vs [INFO DRT-0176] GCELLGRID X -100 DO 905 STEP 3000 ;
[INFO DRT-0177] GCELLGRID Y -1 DO 883 STEP 3000 ; vs [INFO DRT-0177] GCELLGRID Y 0 DO 883 STEP 3000 ;
e) "no mention of this" vs [INFO DRT-0185] Post process initialize RPin region query.     
f) [INFO DRT-0184] Done with 1360212 vertical wires in 19 frboxes and 949165 vs 1286915 vertical wires in 19 frboxes and 852326      
Done with 145885 vertical wires in 19 frboxes and 122642 vs 174782 vertical wires in 19 frboxes and 175988

I get completely different guides/offset from groute and the "Post process initialize RPin region query" message is new.

maliberty, 2022-05-20:
The ispd benchmark comes with guides so you shouldn't be running global route at all.

stefanottili, 2022-05-20:
Why not ?
Starting point is a legal placement that is known to route.
And it’s not a congestion problem, just pin access to one of the macro pins.
The pin is even wide enough to be covered by multiple tracks, so no offgrid pin access necessary.

maliberty, 2022-05-21:
That contest was for detailed routing so the global routing was fixed.  We run it as it was run in the contest.  We could try to do what you are doing but that would be a separate effort and isn't currently a priority (more of a grt test at that point).

stefanottili, 2022-05-21:
loading the guide still leaves me with this one macro pin access violation, even with --latest OpenROAD v2.0-3684-ga120f938e

read_lef ispd19_test6/ispd19_test6.input.lef.gz
read_def ispd19_test6/ispd19_test6.input.def.gz
detailed_route -guide ispd19_test6/ispd19_test6.guide -output_drc 5_route_drc.rpt -verbose 1
exit

comparing the 32-thread linux regression log file against the 8-thread M1 Mac log shows that the first discrepancies are in the "guide region query size" messages.

PS: It would be very helpful if the log file would contain [CMD] lines showing which commands were run and the runtime of these that can be easily grep'ed for. Let me know whether I should make that a feature request.

[log25.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8748196/log25.gz)

[INFO DRT-0036] FR_MASTERSLICE guide region query size = 0.     [INFO DRT-0036] FR_MASTERSLICE guide region query size = 0.
[INFO DRT-0036] FR_VIA guide region query size = 0.             [INFO DRT-0036] FR_VIA guide region query size = 0.
[INFO DRT-0036] Metal1 guide region query size = 763939.      | [INFO DRT-0036] Metal1 guide region query size = 764122.
[INFO DRT-0036] Via1 guide region query size = 0.               [INFO DRT-0036] Via1 guide region query size = 0.
[INFO DRT-0036] Metal2 guide region query size = 774176.      | [INFO DRT-0036] Metal2 guide region query size = 774169.
[INFO DRT-0036] Via2 guide region query size = 0.               [INFO DRT-0036] Via2 guide region query size = 0.
[INFO DRT-0036] Metal3 guide region query size = 509727.      | [INFO DRT-0036] Metal3 guide region query size = 509715.
[INFO DRT-0036] Via3 guide region query size = 0.               [INFO DRT-0036] Via3 guide region query size = 0.
[INFO DRT-0036] Metal4 guide region query size = 71887.       | [INFO DRT-0036] Metal4 guide region query size = 71901.
[INFO DRT-0036] Via4 guide region query size = 0.               [INFO DRT-0036] Via4 guide region query size = 0.
[INFO DRT-0036] Metal5 guide region query size = 9673.          [INFO DRT-0036] Metal5 guide region query size = 9673.
[INFO DRT-0036] Via5 guide region query size = 0.               [INFO DRT-0036] Via5 guide region query size = 0.
[INFO DRT-0036] Metal6 guide region query size = 4591.          [INFO DRT-0036] Metal6 guide region query size = 4591.
[INFO DRT-0036] Via6 guide region query size = 0.               [INFO DRT-0036] Via6 guide region query size = 0.
[INFO DRT-0036] Metal7 guide region query size = 3429.        | [INFO DRT-0036] Metal7 guide region query size = 3431.
[INFO DRT-0036] Via7 guide region query size = 0.               [INFO DRT-0036] Via7 guide region query size = 0.
[INFO DRT-0036] Metal8 guide region query size = 1672.        | [INFO DRT-0036] Metal8 guide region query size = 1671.
[INFO DRT-0036] Via8 guide region query size = 0.               [INFO DRT-0036] Via8 guide region query size = 0.
[INFO DRT-0036] Metal9 guide region query size = 147.           [INFO DRT-0036] Metal9 guide region query size = 147.
[INFO DRT-0179] Init gr pin query.                              [INFO DRT-0179] Init gr pin query.

maliberty, 2022-05-23:
You can always file an issue if you think something needs improving.

vijayank88, 2023-06-19:
Please re-open if you're facing an issue with test case again.

stefanottili, 2023-06-19:
I kept this bug open because OR failed to fix 1 violation both with and without guides.
It shouldn't just be closed because it's "old", but thanks for reminding me to revisit this.
Detail route doesn't have the option to load guides any more, so I just reran with groute/detail-route.

With a June 18 OR executable detailed routing finishes with 0 violations after the 8th iteration.
Great !!

But .... the runtime increased from 20min to 2h40min !!
8x the runtime to fix 1 violation ...

I'm curious whether you would consider that a regression or just a side effect of the router doing checking more rules.
Is ispd16_test6 still part of the nightly regression and if so is this runtime to be expected ?
This seems excessive for 170k instances and a very routable design (according to the GRT congestion report).

May 22 2022:
[INFO DRT-0195] Start 64th optimization iteration.
[INFO DRT-0199] Number of violations = 1.
[INFO DRT-0267] cpu time = 02:30:36, elapsed time = 00:20:51, memory = 8211.41 (MB), peak = 8287.22 (MB)

June 19 20223:
[INFO DRT-0195] Start 6th optimization iteration.
    Completing 10% with 21 violations. ....
[INFO DRT-0195] Start 7th optimization iteration
    Completing 10% with 1 violations. ....
[INFO DRT-0195] Start 8th optimization iteration.
[INFO DRT-0199]   Number of violations = 0.
[INFO DRT-0267] cpu time = 16:41:41, elapsed time = 02:40:50, memory = 7391.95 (MB), peak = 9197.98 (MB)

[INFO GRT-0096] Final congestion report:
Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
---------------------------------------------------------------------------------------
Metal1         4325352        361063            8.35%             0 /  0 /  0
Metal2         6412381       1743220           27.19%             0 /  0 /  0
Metal3         7418829       1524345           20.55%             0 /  0 /  0
Metal4         7423116        496914            6.69%             0 /  0 /  0
Metal5         4807805        120909            2.51%             0 /  0 /  0
Metal6         3316595         50165            1.51%             0 /  0 /  0
Metal7         5444456         60756            1.12%             0 /  0 /  0
Metal8         5578624          4261            0.08%             0 /  0 /  0
Metal9          799092            84            0.01%             0 /  0 /  0
---------------------------------------------------------------------------------------
Total         45526250       4361717            9.58%             0 /  0 /  0

maliberty, 2023-06-19:
There is a read_guides command.  I think the runtime is worth looking at.

stefanottili, 2023-06-19:
Using read_guides the runtime is much better at 50min, which is still 2.5x slower then in May 2022.
This really boils down to a groute test case, the M1-M2 wire length is a lot larger then with the read_guides.

groute
```
1402 Total wire length on LAYER Metal1 = 96338 um.
1403 Total wire length on LAYER Metal2 = 2552239 um.
1404 Total wire length on LAYER Metal3 = 2780384 um.
```
read_guide
```
338 Total wire length on LAYER Metal1 = 3895 um.
339 Total wire length on LAYER Metal2 = 1566612 um.
340 Total wire length on LAYER Metal3 = 2240879 um.
```

PS: can somebody please explain the DRC violation number display for "completing xx%" to me ?

```
1370 [INFO DRT-0195] Start 0th optimization iteration.
1371     Completing 10% with 54891 violations.
...
1389     Completing 100% with 495540 violations.
1390     elapsed time = 01:06:44, memory = 4796.30 (MB).
1391 [INFO DRT-0199]   Number of violations = 1080579.
...
1428 [INFO DRT-0195] Start 1st optimization iteration.
1429     Completing 10% with 986059 violations.
...
1447     Completing 100% with 64751 violations.
1448     elapsed time = 00:34:46, memory = 6473.91 (MB).
1449 [INFO DRT-0199]   Number of violations = 452590.
```
groute

```
1363 [INFO DRT-0184] Done with 982157 vertical wires in 19 frboxes and 631114 horizontal wires in 18 frboxes.
1364 [INFO DRT-0186] Done with 338675 vertical wires in 19 frboxes and 268880 horizontal wires in 18 frboxes.
1365 [INFO DRT-0182] Complete track assignment.
1366 [INFO DRT-0267] cpu time = 00:04:25, elapsed time = 00:00:56, memory = 3772.45 (MB), peak = 6942.59 (MB)
1367 [INFO DRT-0187] Start routing data preparation.
1368 [INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 4431.84 (MB), peak = 6942.59 (MB)
1369 [INFO DRT-0194] Start detail routing.
1370 [INFO DRT-0195] Start 0th optimization iteration.
1371     Completing 10% with 54891 violations.
1372     elapsed time = 00:08:11, memory = 5830.45 (MB).
1373     Completing 20% with 154296 violations.
1374     elapsed time = 00:14:11, memory = 4220.42 (MB).
1375     Completing 30% with 154320 violations.
1376     elapsed time = 00:18:59, memory = 5508.48 (MB).
1377     Completing 40% with 249650 violations.
1378     elapsed time = 00:26:17, memory = 4764.81 (MB).
1379     Completing 50% with 282048 violations.
1380     elapsed time = 00:29:40, memory = 5145.11 (MB).
1381     Completing 60% with 327995 violations.
1382     elapsed time = 00:39:49, memory = 6646.06 (MB).
1383     Completing 70% with 404091 violations.
1384     elapsed time = 00:46:52, memory = 4552.16 (MB).
1385     Completing 80% with 404102 violations.
1386     elapsed time = 00:52:55, memory = 6016.72 (MB).
1387     Completing 90% with 474020 violations.
1388     elapsed time = 01:02:26, memory = 4899.11 (MB).
1389     Completing 100% with 495540 violations.
1390     elapsed time = 01:06:44, memory = 4796.30 (MB).
1391 [INFO DRT-0199]   Number of violations = 1080579.
1392 Viol/Layer      Metal1   Via1 Metal2   Via2 Metal3   Via3 Metal4   Via4 Metal5   Via5 Metal6   Via6 Metal7 Metal8
1393 Corner Spacing       0      0   2597      0    199      0     21      0      3      0      2      0      0      0
1394 Cut Spacing          0  10303      0  14171      0   1729      0     22      0      0      0      0      0      0
1395 EOL Spacing      17325      0  48249      0   8801      0    111      0     11      0      0      0      0      0
1396 Metal Spacing    77336      0  25436      0    732      0     36      0     18      0      2      0      4      0
[... 197 lines cut ...]
592 [INFO DRT-0195] Start 5th optimization iteration.
593     Completing 10% with 189 violations.
594     elapsed time = 00:00:00, memory = 7384.34 (MB).
595     Completing 20% with 8 violations.
596     elapsed time = 00:00:00, memory = 7384.48 (MB).
597     Completing 30% with 8 violations.
598     elapsed time = 00:00:00, memory = 7384.48 (MB).
599     Completing 40% with 7 violations.
600     elapsed time = 00:00:00, memory = 7384.52 (MB).
601     Completing 50% with 6 violations.
602     elapsed time = 00:00:00, memory = 7384.59 (MB).
603     Completing 60% with 6 violations.
604     elapsed time = 00:00:00, memory = 7384.59 (MB).
605     Completing 70% with 3 violations.
606     elapsed time = 00:00:00, memory = 7384.64 (MB).
607     Completing 80% with 3 violations.
608     elapsed time = 00:00:00, memory = 7384.64 (MB).
609     Completing 90% with 1 violations.
610     elapsed time = 00:00:00, memory = 7384.67 (MB).
611     Completing 100% with 0 violations.
612     elapsed time = 00:00:00, memory = 7384.67 (MB).
613 [INFO DRT-0199]   Number of violations = 0.
614 [INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 7384.67 (MB), peak = 8466.27 (MB)

```

maliberty, 2023-06-19:
If you don't use the guides then you should use the global router in OR and not rely on the one in drt without guides.  It is more of an academic legacy.

stefanottili, 2023-06-20:
given that the std cells are completely blocked in M1 for either PINS or OBS, there shouldn't be any M1 routing ...

maliberty, 2023-06-20:
When you ran grt what did you set for the min routing layer?

stefanottili, 2023-06-20:
I didn't set a min routing layer. I was (wrongly ?) assuming that global route would take M1 pins/obs of std cells into account when computing routing resources, but when I do set Metal2 as min layer I get a coredump ...
```
read_lef ispd19_test6/ispd19_test6.input.lef.gz
read_def ispd19_test6/ispd19_test6.input.def.gz
set_thread_count 8
set_routing_layers -signal Metal2-Metal9 -clock Metal2-Metal9
global_route -verbose
#read_guides ispd19_test6/ispd19_test6.guide 
detailed_route -output_drc 5_route_drc.rpt -verbose 1
write_db route.db
```
```
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 29018 violations.
    elapsed time = 00:03:48, memory = 6417.78 (MB).
    Completing 20% with 80829 violations.
    elapsed time = 00:06:52, memory = 4275.94 (MB).
    Completing 30% with 80846 violations.
    elapsed time = 00:08:46, memory = 6192.88 (MB).
    Completing 40% with 131632 violations.
    elapsed time = 00:12:26, memory = 4434.80 (MB).
    Completing 50% with 150025 violations.
    elapsed time = 00:14:18, memory = 4577.17 (MB).
Signal 11 received
Thread 0::  Dispatch queue: com.apple.main-thread
0   openroad                      	       0x1007c30b0 fr::FlexGridGraph::getNextPathCost(fr::FlexWavefrontGrid const&, fr::frDirEnum const&) const + 716
1   openroad                      	       0x1007c2330 fr::FlexGridGraph::expand(fr::FlexWavefrontGrid&, fr::frDirEnum const&, fr::FlexMazeIdx const&, fr::FlexMazeIdx const&, odb::Point const&) + 232
2   openroad                      	       0x1007c3f08 fr::FlexGridGraph::expandWavefront(fr::FlexWavefrontGrid&, fr::FlexMazeIdx const&, fr::FlexMazeIdx const&, odb::Point const&) + 372
3   openroad                      	       0x1007c6fe0 fr::FlexGridGraph::search(std::__1::vector<fr::FlexMazeIdx, std::__1::allocator<fr::FlexMazeIdx>>&, fr::drPin*, std::__1::vector<fr::FlexMazeIdx, std::__1::allocator<fr::FlexMazeIdx>>&, fr::FlexMazeIdx&, fr::FlexMazeIdx&, odb::Point const&, std::__1::map<fr::FlexMazeIdx, fr::frBox3D*, std::__1::less<fr::FlexMazeIdx>, std::__1::allocator<std::__1::pair<fr::FlexMazeIdx const, fr::frBox3D*>>>&) + 1700
4   openroad                      	       0x1007b8394 fr::FlexDRWorker::routeNet(fr::drNet*) + 420
5   openroad                      	       0x1007b69c0 fr::FlexDRWorker::route_queue_main(std::__1::queue<fr::FlexDRWorker::RouteQueueEntry, std::__1::deque<fr::FlexDRWorker::RouteQueueEntry, std::__1::allocator<fr::FlexDRWorker::RouteQueueEntry>>>&) + 648
6   openroad                      	       0x1007b6274 fr::FlexDRWorker::route_queue() + 456
7   openroad                      	       0x100776c64 fr::FlexDRWorker::main(fr::frDesign*) + 2668
```

maliberty, 2023-06-21:
I just ran with your script and no crash using aed70e329e5b7c072efcc92a370c68dfe5b5a54d

stefanottili, 2023-06-21:
OpenROAD v2.0-8799-gfb20dbe66 (Jun 21 2023) finishes.
Merge pull request https://github.com/The-OpenROAD-Project/OpenROAD/pull/3487 from The-OpenROAD-Project-staging/gui-race
seems to have done the trick for OR on Apple Silicon too.

So I this ticket can be closed, since OR routes ispd16_test6, but should there be new issues opened ?
Please comment.

A) the 2.5x detailed router runtime regression (compared to May 2022)
B) the 8x (7x blocking M1) router regression compared to "other tool route guides" 

1) OR May 2022, "other tools" route guide: 20min, 1 Violation (lets take this a baseline)
2) OR June 2023, "other tools" route guide: 50min, 0 Violations (2.5x slowdown)
3) OR June 2023, groute: 2h40, 0 Violations (8x slowdown)
4) OR June 2023, groute M2-M9: 2h15, 0 Violations (7x slowdown)

a) the detailed router slowed down by 2.5x from May 2022 to June 2023, finishes without the 1 violation.
b) the global router doesn't seem to take the M1/M2 std cell PIN/OBS into account,
so the detail router starts with a lot more routes/violations on these layers then by loading "other tools" route guides.
c) not using M1 at all helps, but speeds up the overall runtime from 2:40 to 2:15.

While looking at the logs I was slightly confused by the # of violations being reported:
It seems to me that the "Recheck" (violations ?) get added to the # of violations counting down.
```
391     Completing 100% with 275348 violations.    
...
393 [INFO DRT-0199]   Number of violations = 711042.        
...
400 Recheck           2864      0 108070      0 315882      0   6426      0    751      0   1182      0    516      3
...
430 [INFO DRT-0195] Start 1st optimization iteration.
431     Completing 10% with 649708 violations.
```
d) What are Recheck violations ? Why are they added to the # of violations for the next iteration ?

```
read_lef ispd19_test6/ispd19_test6.input.lef.gz
read_def ispd19_test6/ispd19_test6.input.def.gz
set_thread_count 8
set_routing_layers -signal Metal2-Metal9 -clock Metal2-Metal9
global_route -verbose
#read_guides ispd19_test6/ispd19_test6.guide 
detailed_route -output_drc 5_route_drc.rpt -verbose 1
write_db route.db
```
```
 61 [INFO GRT-0096] Final congestion report:
 62 Layer         Resource        Demand        Usage (%)    Max H / Max V / Total Overflow
 63 ---------------------------------------------------------------------------------------
 64 Metal1               0             0            0.00%             0 /  0 /  0
 65 Metal2         6412381       1755855           27.38%             0 /  0 /  0
 66 Metal3         7418829       1834935           24.73%             0 /  0 /  0
 67 Metal4         7423116        480628            6.47%             0 /  0 /  0
 68 Metal5         4807805        150773            3.14%             0 /  0 /  0
 69 Metal6         3316595         51309            1.55%             0 /  0 /  0
 70 Metal7         5444456         77481            1.42%             0 /  0 /  0
 71 Metal8         5578624          6766            0.12%             0 /  0 /  0
 72 Metal9          799092           826            0.10%             0 /  0 /  0
 73 ---------------------------------------------------------------------------------------
 74 Total         41200898       4358573           10.58%             0 /  0 /  0
 75 
 76 [INFO GRT-0018] Total wirelength: 8001264 um

372 [INFO DRT-0195] Start 0th optimization iteration.

393 [INFO DRT-0199]   Number of violations = 711042.
394 Viol/Layer      Metal1   Via1 Metal2   Via2 Metal3   Via3 Metal4   Via4 Metal5   Via5 Metal6   Via6 Metal7 Metal8
395 Corner Spacing       0      0   2612      0    201      0      9      0      8      0      0      0      0      0
396 Cut Spacing          0    641      0   6973      0   1214      0     59      0      0      0      0      0      0
397 EOL Spacing        604      0  38542      0   7872      0     92      0      5      0      1      0      0      0
398 Metal Spacing     1749      0  15726      0    456      0     31      0     95      0      1      0      9      0
399 NS Metal             2      0      9      0      0      0      0      0      0      0      0      0      0      0
400 Recheck           2864      0 108070      0 315882      0   6426      0    751      0   1182      0    516      3
401 Short              125     22 172727   1400  22716    417    462      5    468     12     17      2     64      0
402 [INFO DRT-0267] cpu time = 04:22:31, elapsed time = 00:57:24, memory = 7106.08 (MB), peak = 8475.70 (MB)
403 Total wire length = 6707639 um.

789 [INFO DRT-0199]   Number of violations = 0.
790 [INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 6361.03 (MB), peak = 8475.70 (MB)
791 Total wire length = 6674246 um.
792 Total wire length on LAYER Metal1 = 8373 um.
793 Total wire length on LAYER Metal2 = 2343428 um.
794 Total wire length on LAYER Metal3 = 2768303 um.
795 Total wire length on LAYER Metal4 = 1095236 um.
796 Total wire length on LAYER Metal5 = 249966 um.
797 Total wire length on LAYER Metal6 = 77871 um.
798 Total wire length on LAYER Metal7 = 116688 um.
799 Total wire length on LAYER Metal8 = 13146 um.
800 Total wire length on LAYER Metal9 = 1231 um.
801 Total number of vias = 2166082.

846 [INFO DRT-0267] cpu time = 13:50:26, elapsed time = 02:15:52, memory = 6361.03 (MB), peak = 8475.70 (MB)

```

stefanottili, 2023-06-26:
I'm closing this since drt 2023 does route without violations. Filed two follow up issues.

closed 2023-06-26 by stefanottili