The-OpenROAD-Project/OpenROAD #1894: GPL questions of a newbie

mirkat1206, 2022-05-22:
Hello, I am studying LEF/DEF format and trying to put them into RePlAce to get a global placement result and its HPWL.

However, RePlAce runs very slow with my baby LEF/DEF files compared with the much complex gcd testcase.
Is there any problems with my LEF/DEF files?

Here is my LEF file:
```
VERSION 5.6 ;
BUSBITCHARS "[]" ;
DIVIDERCHAR "/" ;
UNITS
  DATABASE MICRONS 100 ;
END UNITS
MANUFACTURINGGRID 0.01 ;

LAYER metal1
  TYPE ROUTING ;
  DIRECTION HORIZONTAL ;
  PITCH 0 ;
  WIDTH 0 ;
  SPACING 0 ;
END metal1

SITE CoreSite
  SYMMETRY x y ;
  CLASS core ;
  SIZE 0.01 BY 0.10 ;
END CoreSite

MACRO MC1
  CLASS core ;
  FOREIGN C1 0.0 0.0 ;
  ORIGIN 0 0 ;
  SITE CoreSite ;
  SIZE 0.10 BY 0.10 ;
  PIN P1
    DIRECTION INPUT ;
    PORT
      LAYER metal1 ;
        RECT 0 0 0 0 ;
    END
  END P1
END MC1

MACRO MC2
  CLASS core ;
  FOREIGN C2 0.0 0.0 ;
  ORIGIN 0 0 ;
  SITE CoreSite ;
  SIZE 0.10 BY 0.10 ;
  PIN P1
    DIRECTION INPUT ;
    PORT
      LAYER metal1 ;
        RECT 0 0 0 0 ;
    END
  END P1
END MC2
```

And here is my DEF file:
```
VERSION 5.6 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
DESIGN design_name ;
UNITS DISTANCE MICRONS 100 ;
DIEAREA ( 0 0 ) ( 20 10 ) ;
ROW r0 CoreSite 0 0 N DO 20 BY 1 STEP 1 0 ;
COMPONENTS 2 ;
- C1 MC1 ;
- C2 MC2 ;
END COMPONENTS
NETS 1 ;
- N1
	( C1 P1 ) ( C2 P1 ) ;
END NETS
END DESIGN
```

mirkat1206, 2022-05-22:
And I met *NEGATIVE HPWL ERROR* in the final result.
Any suggestions for this problem?

[image]

Appreciate your help!

maliberty, 2022-05-22:
Would you package a test case

maliberty, 2022-12-15:
No activity; please reopen if still relevant.

closed 2022-12-15 by maliberty