The-OpenROAD-Project/OpenROAD #1898: clock_tree_synthesis is causing Openroad to crash

gkamendje, 2022-05-23:
I am using the latest version of ORFS and  the call to clock_tree_synthesis in scripts/cts.tcl is systematically causing Openroad to crash pretty early in the process. I have no clue on how to start debugging this. Not sure what exactly is causing the issue and how to guide someone else to reproduce it. I have double checked the synthesized netlist as well as the pin placement and everything seems to be ok. 
```
OpenROAD v2.0-3729-g713ef39fc
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO CTS-0049] Characterization buffer is: CLKBUFHDV1.
[INFO CTS-0039] Number of created patterns = 49104.
[INFO CTS-0084] Compiling LUT.
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           34          1           572
[WARNING CTS-0043] 3168 wires are pure wire and no slew degradation.
TritonCTS forced slew degradation on these wires.
[INFO CTS-0046]     Number of wire segments: 48888.
[INFO CTS-0047]     Number of keys in characterization LUT: 1990.
[INFO CTS-0048]     Actual min input cap: 1.
Stack trace:
 0# 0x0000558132766CB1 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F791924B0C0 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::Vertex::bfsInQueue(sta::BfsIndex) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# sta::BfsIterator::enqueue(sta::Vertex*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# 0x0000558132B77BC5 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# sta::ClkNetwork::findClkPins() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# sta::dbSta::findClkNets() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# cts::TritonCTS::populateTritonCTS() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# cts::TritonCTS::runTritonCts() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# 0x0000558132FECCB6 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# TclNRRunCallbacks in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007F791AF88924 in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
14# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# ord::tclAppInit(Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# Tcl_MainEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
17# main in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# _start in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command exited with non-zero status 1
Elapsed time: 0:13.92[h:]min:sec. CPU time: user 13.73 sys 0.17 (99%). Peak memory: 307832KB.

```

0962221289, 2022-05-23:
[image]

0962221289, 2022-05-23:
> I am using the latest version of ORFS and  the call to clock_tree_synthesis in scripts/cts.tcl is systematically causing Openroad to crash pretty early in the process. I have no clue on how to start [...]

gkamendje, 2022-05-23:
The `make cts_isse`  command includes Foundry files in the tar bundle it creates. What would be the best way to share this file without violating any NDA?

maliberty, 2022-05-23:
You can't transfer a testcase with private PDK data.  Which PDK are you using?  If we have it in house I could take the test case without the PDK.

gkamendje, 2022-05-23:
I am using SMIC180. I will try to see if I can reproduce the issue with a public PDK.

maliberty, 2022-05-23:
If you can't reproduce it on a public PDK then @tspyrou might be able to help with NDAs.

gkamendje, 2022-05-23:
I have tried reproducing it on an public PDK but no luck so far. 
Is there any recommended procedure for compiling Openroad in debug mode? I could try firing up GBD and see how far I can get.

maliberty, 2022-05-23:
I usually use from the top of ORFS:
```
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_FLAGS_DEBUG="-g -O0" tools/OpenROAD -B tools/OpenROAD/build -D CMAKE_INSTALL_PREFIX=`pwd`/tools/install/OpenROAD
```
then 
```
make -C tools/OpenROAD/build/  install -j45
```

gkamendje, 2022-05-25:
@tspyrou any update regarding the NDA?

gkamendje, 2022-05-31:
I attached gdb to a run of `make cts`. it looks like the call to  `sta::Vertex::bfsInQueue (this=0x0, index=sta::BfsIndex::other)` `OpenROAD/src/sta/graph/Graph.cc:1292` is causing the segmentation fault. I am attaching the full gdb log to this message. 

I noticed the following but I am not sure if it has anything to do with the issue. 
```
#6  0x0000555559861e80 in sta::dbSta::findClkNets (this=0x55555f7674c0) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/dbSta/src/dbSta.cc:340
        clk_nets = std::set with 140737488339664 elements<error reading variable: Cannot access memory at address 0xcbee8c78958>
```
@tspyrou any update regarding the NDA?

[gdb.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/8801552/gdb.log)

QuantamHD, 2022-06-11:
@gkamendje In your PDK what signal type is the input of your buffer? I'm also using a proprietary PDK, and ran into a segfault because OpenROAD didn't support clock type for the clock buffer cells. I submitted https://github.com/The-OpenROAD-Project/OpenROAD/pull/1952 if you want to give that a try.

gkamendje, 2022-06-12:
@QuantamHD Thanks for notifying. In my case, buffer cells have `SIGNAL` as clock input type. As I mentioned previously, the problem is related to a call to `sta::Vertex::bfsInQueue (this=0x0, index=sta::BfsIndex::other)` in the file `OpenROAD/src/sta/graph/Graph.cc:1292` while it is trying to find the clock pins of the clock network (see the gdb log below). I need some further investigation to understand exactly what is happening. 

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
[34m-----------------------------------------------------------------------------------------------------------------------[34m[1m[regs]
[0m  [32mRAX:[30m 0x0000000000000000  [32mRBX:[31m 0x000055555F7C9790  [32mRBP:[31m 0x00007FFFFFFFBE30  [32mRSP:[31m 0x00007FFFFFFFBE30  [1m[4m[31mo d I t s Z a P c 
[0m  [32mRDI:[30m 0x0000000000000000  [32mRSI:[31m 0x0000000000000003  [32mRDX:[31m 0x0000000000000003  [32mRCX:[30m 0x0000000000000000  [32mRIP:[31m 0x0000555559887431
[32mR8 :[31m 0x000055555F7BA348  [32mR9 :[31m 0x0000000000000001  [32mR10:[31m 0x000055555F656010  [32mR11:[31m 0x0000000000000001  [32mR12:[30m 0x0000000000000000
  [32mR13:[31m 0x000055555F75B090  [32mR14:[31m 0x000055555F7B5B90  [32mR15:[31m 0x0000000000000001
  [32mCS:[30m 0033  [32mDS:[30m 0000  [32mES:[30m 0000  [32mFS:[30m 0000  [32mGS:[30m 0000 [32mSS:[30m 002B[0m				
[36m[0m[34m-----------------------------------------------------------------------------------------------------------------------[34m[1m[code]
[0m=> 0x555559887431 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+19>:	movzx  eax,BYTE PTR [rax+0x1f]
   0x555559887435 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+23>:	and    eax,0xf
   0x555559887438 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+26>:	movzx  edx,al
   0x55555988743b <sta::Vertex::bfsInQueue(sta::BfsIndex) const+29>:	mov    eax,DWORD PTR [rbp-0xc]
   0x55555988743e <sta::Vertex::bfsInQueue(sta::BfsIndex) const+32>:	mov    ecx,eax
   0x555559887440 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+34>:	sar    edx,cl
   0x555559887442 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+36>:	mov    eax,edx
   0x555559887444 <sta::Vertex::bfsInQueue(sta::BfsIndex) const+38>:	and    eax,0x1
[34m-----------------------------------------------------------------------------------------------------------------------------
[0m0x0000555559887431 in sta::Vertex::bfsInQueue (this=0x0, index=sta::BfsIndex::other) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/graph/Graph.cc:1292
1292	  return (bfs_in_queue_ >> unsigned(index)) & 1;
#0  0x0000555559887431 in sta::Vertex::bfsInQueue (this=0x0, index=sta::BfsIndex::other) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/graph/Graph.cc:1292
No locals.
#1  0x00005555599b792d in sta::BfsIterator::enqueue (this=0x7fffffffbf40, vertex=0x0) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/search/Bfs.cc:257
No locals.
#2  0x0000555559b5ee10 in sta::ClkNetwork::findClkPins (this=0x55555f7ba230, ideal_only=0x0, pin_clks_map=...) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/search/ClkNetwork.cc:123
        vertex = 0x0
        bidirect_drvr_vertex = 0x0
        pin = 0x55556a3b1670
        __for_range = @0x55556a719728: {<std::set<sta::Pin*, std::less<sta::Pin*>, std::allocator<sta::Pin*> >> = std::set with 1 element = {[0] = 0x55556a3b1670}, <No data fields>}
        __for_begin = 0x55556a3b1670
        __for_end = 0x1
        clk_pins = @0x55556be1b358: {<std::set<sta::Pin*, std::less<sta::Pin*>, std::allocator<sta::Pin*> >> = std::set with 0 elements, <No data fields>}
        clk = 0x55556a7196e0
        __for_range = @0x55555f8f66e0: {<std::vector<sta::Clock*, std::allocator<sta::Clock*> >> = std::vector of length 33, capacity 64 = {0x55555f73d500, 0x55556a713200, 0x55556a7135c0, 0x55556a713a10, 0x55556a713e90, 0x55556a714260, 0x55556a714650, 0x55556a714a90, 0x55556a714ea0, 0x55556a7152b0, 0x55556a7156c0, 0x55556a715ad0, 0x55556a715ee0, 0x55556a716320, 0x55556a716730, 0x55556a716c60, 0x55556a7170a0, 0x55556a7174e0, 0x55556a717920, 0x55556a717d60, 0x55556a7181a0, 0x55556a7185e0, 0x55556a718a20, 0x55556a718e60, 0x55556a7192a0, 0x55556a7196e0, 0x55556a719b20, 0x55556a719f60, 0x55556a71a3a0, 0x55556a71a7e0, 0x55556a71ac20, 0x55556a71b240, 0x55556a71b680}, <No data fields>}
        __for_begin = 0x55556a7196e0
        __for_end = 0x55556a7108b0
        srch_pred = {<sta::ClkTreeSearchPred> = {<sta::SearchPred1> = {<sta::SearchPred0> = {<sta::SearchPred> = {_vptr.SearchPred = 0x55555f1846c0 <vtable for sta::ClkSearchPred+16>}, sta_ = 0x55555f7ba230}, <No data fields>}, <No data fields>}, <No data fields>}
        bfs = {<sta::BfsIterator> = {<sta::StaState> = {_vptr.StaState = 0x55555f181010 <vtable for sta::BfsFwdIterator+16>, report_ = 0x55555f73a550, debug_ = 0x55555f778050, units_ = 0x55555f768f40, network_ = 0x55555f73f250, sdc_network_ = 0x55555f7ba380, cmd_network_ = 0x55555f7ba380, sdc_ = 0x55555f8f65f0, corners_ = 0x55555f74e7c0, graph_ = 0x55555f8a0ae0, levelize_ = 0x55555f77a7f0, parasitics_ = 0x55555f7491f0, arc_delay_calc_ = 0x55555f791d80, graph_delay_calc_ = 0x55555f9ad7e0, sim_ = 0x55555f793610, search_ = 0x55555f736d60, latches_ = 0x55555f8f7770, clk_network_ = 0x55555f7ba230, thread_count_ = 0x1, dispatch_queu [... cut ...]
No symbol table info available.
#19 0x00005555594c1ab7 in main (argc=0x4, argv=0x7fffffffcac8) at /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:237
        orig_sigint_handler = {__sigaction_handler = {sa_handler = 0x0, sa_sigaction = 0x0}, sa_mask = {__val = {0x0, 0x0, 0x0, 0x18, 0x5555594c3668, 0x55555c651a1b, 0x7fffffffd0c9, 0x7fffffffc8b0, 0x55555987bda6, 0x0, 0x55555c651a1b, 0x7fffffffcac8, 0x7fffffffc8cc, 0x5b0000006e, 0x600000004, 0x7fffffffd0c9}}, sa_flags = 0x0, sa_restorer = 0x0}

```

jjcherry56, 2022-07-21:
The problem has nothing to do with LEF.
It is failing in opensta, which uses liberty, not LEF.
The issue is that the pin it is searching from has no graph vertex.
Can you show me the SDC for the clock definitions and the verilog lines that reference the clock pins?
I suspect there is some issue with missing liberty cell definitions.

gkamendje, 2022-07-31:
@jjcherry56 Thanks for looking into this. 
Unfortunately I cannot publicly share the SDC. 
I used the following script snippet to load the `.odb`  and the `.sdc` files generated after placement. 

```
read_lef ${TECH_LEF}
read_lef ${SC_LEF}
read_db  ./results/${TECH}/${DESIGN}/base/3_place.odb
read_sdc ./results/${TECH}/${DESIGN}/base/3_place.sdc

check_setup -verbose
report_clock_properties 
report_checks -from [get_clocks *]  -path_delay max -unconstrained -group_count 100 -endpoint_count 100
```
- I do not get any error nor warning while reading the `.sdc` file. 
- `check_setup-verbose` does not report any issue.
- `report_clock_properties` shows all the clock as defined in the `.sdc` file 
- `report_checks -from [get_clocks *] -unconstrained ` report that all the paths in the design are unconstrained. This is rather puzzling given that the log file `3_5_opendp.log` does report multiple constrained paths and `report_disable_edges` does not report any disabled edge that could explain why the paths are unconstrained. Is there a command that could help me understand why the paths are seen as unconstrained?

jjcherry56, 2022-07-31:
Honestly, there isn't much I can do to help you without any data. I am not going to play 20 questions in english to debug your design.

It sounds like check_setup should have  reported unconstrained endpoints. Something is very strange.

Are there set_false_path commands between the clocks? Because that would make the paths unconstrained.

Pick a register in the design and use report_arrival on the data and clock pins. The data arrival will be with respect to some clock. The clock arrival should have arrivals from both the rising and falling edges of the clock source. If you don't see the arrivals, trace back through the logic to them using report_edges -to|-from to see the graph and gate/net delays.

maliberty, 2022-12-15:
There doesn't appear to be any way to make further progress on this.

closed 2022-12-15 by maliberty