The-OpenROAD-Project/OpenROAD #1903: Facing segfault during global placement RePlace

vijayank88, 2022-05-24:
`Issue Description:`

With `set ::env(PL_ROUTABILITY_DRIVEN) {1};` configuration facing abnormal exit by tool.

`Log:`
```
[STEP 7]
[INFO]: Running Global Placement...
[ERROR]: during executing openroad script /openlane/scripts/openroad/replace.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/syntacore/runs/RUN_2022.05.23_11.55.13/logs/placement/7-global.log
[ERROR]: Last 10 lines:
33# 0x00007F78CAA9EF1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000001028727 in openroad
child process exited abnormally
```
`Workaround Used:`
If I disable `set ::env(PL_ROUTABILITY_DRIVEN) {0};` tool complete the global placement stage.

`Test case: `
[openroad_issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/8762071/openroad_issue_reproducible.zip)

eder-matheus, 2022-08-16:
@vijayank88 I've checked your test case with the latest OpenROAD version, and it ends with no errors. Can we close this issue?

vijayank88, 2022-08-17:
@eder-matheus 
Have you changed anything wrt., `openlane/scripts/openroad`? 
Still I am seeing same crash with test case attached

vijayank88, 2022-08-17:
Considering original design able to pass https://github.com/Univisiontechnocon-com/yifive_a2/tree/master/openlane/syntacore global placement stage now with latest OR.

closed 2022-08-17 by vijayank88