The-OpenROAD-Project/OpenROAD #1908: make gui_* causing OR to crash in ORFS

gkamendje, 2022-05-25:
I am using ORFS and updated my git directory a couple of days ago to OpenROAD v2.0-3729-g713ef39fc. Now, running `make gui_2_floorplan.odb`  for example will cause the tool to crash (see the stack trace below). 
I do not see this behavior with the previous version of OR I was running OpenROAD v2.0-3236-ge9d9ebe5f. 
I am running Ubuntu 20.04 in WSL2
```
 make gui_2_floorplan.odb
[INFO][FLOW] Using platform directory ./platforms/smic018
ODB_FILE=./results/smic018/puric/base/2_floorplan.odb /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  ./scripts/gui.tcl
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-gkamendje'
OpenROAD v2.0-3729-g713ef39fc
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad: /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/db/dbTable.h:116: T* odb::dbTable<T>::getPtr(odb::dbId<T>) const [with T = odb::_dbInst]: Assertion `p->_oid & DB_ALLOC_BIT' failed.
Stack trace:
 0# 0x000055581CBA05F2 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FCD9FD870C0 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x00007FCD9FD66729 in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007FCD9FD78006 in /lib/x86_64-linux-gnu/libc.so.6
 6# odb::dbTable<odb::_dbInst>::getPtr(odb::dbId<odb::_dbInst>) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# odb::dbModuleInstItr::getObject(unsigned int, ...) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# odb::dbSetIterator<odb::dbInst>::operator*() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# gui::BrowserWidget::populateModule(odb::dbModule*, QStandardItem*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# gui::BrowserWidget::addModuleItem(odb::dbModule*, QStandardItem*, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# gui::BrowserWidget::updateModel() in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# gui::BrowserWidget::setBlock(odb::dbBlock*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# gui::MainWindow::setBlock(odb::dbBlock*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# 0x000055581D986F9B in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# QMetaObject::activate(QObject*, int, int, void**) in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
16# gui::MainWindow::designLoaded(odb::dbBlock*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# gui::MainWindow::postReadDb(odb::dbDatabase*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# ord::OpenRoad::readDb(char const*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
19# read_db_cmd(char const*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
20# 0x000055581CBBB790 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
21# TclNRRunCallbacks in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
22# 0x00007FCDA1AC4924 in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
23# Tcl_EvalEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
24# Tcl_Eval in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
25# gui::ScriptWidget::executeTclCommand(QString const&) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
26# gui::ScriptWidget::executeSilentCommand(QString const&) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
27# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
28# 0x000055581CBA0CCF in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
29# ord::tclAppInit(Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
30# Tcl_MainEx in /usr/lib/x86_64-linux-gnu/libtcl8.6.so
31# main in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
32# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
33# _start in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad

```

maliberty, 2022-05-25:
You need to update again.  That bug was fixed a few commits later.

maliberty, 2022-05-25:
> @maliberty Is this new feature added to open `.odb` in gui directly? `make gui_2_floorplan.odb`
> 
> I used only `make gui_final` to load final layout view in `openroad GUI`

That feature was added when we switched to odb files (make gui_<step>.odb).  It was similar when we use def files between steps.