The-OpenROAD-Project/OpenROAD #1921: Segmentation fault when running GRT

YihangQiu, 2022-05-28:
For tinyRocket_nangate45, I can run through the full flow with OpenRoad, but when I use my own design to turn on routability in the global placement phase, I get an error due to a call to GRT. The error is reported as shown in the figure. I don't know what caused the segment error and would appreciate a reply.

[image]

maliberty, 2022-05-28:
Can you provide a test case?

YihangQiu, 2022-05-28:
> Can you provide a test case?

Okay, but the design file is not convenient for public. Can we contact by email? My email address is edayhq@126.com

vijayank88, 2022-05-30:
@YihangQiu 
Run `make global_routing_issue` and share test case here.