The-OpenROAD-Project/OpenROAD #1927: RSZ: hold repair creates setup violations

antonblanchard, 2022-05-30:
I'm struggling to resolve setup violations while taping out Microwatt. While the design has a lot of setup slack, it appears that hold repair is consuming all that slack and creating setup violations.

There are no complaints from the resizer:

```
Repair setup violations...
[DEBUG RSZ-0001] worst_slack = 6.827
Repair hold violations...
[INFO RSZ-0046] Found 2447 endpoints with hold violations.
[INFO RSZ-0032] Inserted 7683 hold buffers.
Placement Analysis
```

But we end up with setup violations. As an aside - I wonder if we need a way to specify the setup slack margin when doing hold repair. At the moment we have a single option:

```
repair_timing -hold -slack_margin 0.2
```

What does that mean for the setup margin? Do we use the same 0.2 value? I might want to maintain a relatively large setup margin together with a small hold margin.

```
Startpoint: _122817_ (rising edge-triggered flip-flop clocked by ext_clk)
Endpoint: soc0.bram.bram0.ram_0.memory_0
          (rising edge-triggered flip-flop clocked by ext_clk)
Path Group: ext_clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                          0.00    0.00   clock ext_clk (rise edge)
                          0.00    0.00   clock source latency
                  0.70    0.51    0.51 ^ ext_clk (in)
     1    0.16                           ext_clk (net)
                  0.70    0.00    0.51 ^ repeater15/A (sky130_fd_sc_hd__buf_16)
                  0.37    0.44    0.95 ^ repeater15/X (sky130_fd_sc_hd__buf_16)
     1    0.35                           net7077 (net)
                  0.37    0.01    0.95 ^ clkbuf_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.08    0.26    1.21 ^ clkbuf_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.03                           clknet_0_ext_clk (net)
                  0.08    0.00    1.21 ^ clkbuf_1_1_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.23    0.29    1.50 ^ clkbuf_1_1_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.13                           clknet_1_1_0_ext_clk (net)
                  0.23    0.00    1.50 ^ clkbuf_2_2_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.23    0.35    1.85 ^ clkbuf_2_2_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.13                           clknet_2_2_0_ext_clk (net)
                  0.23    0.00    1.85 ^ clkbuf_3_4_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.24    0.35    2.20 ^ clkbuf_3_4_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     4    0.14                           clknet_3_4_0_ext_clk (net)
                  0.24    0.00    2.20 ^ clkbuf_5_19_0_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.13    0.27    2.47 ^ clkbuf_5_19_0_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     2    0.07                           clknet_5_19_0_ext_clk (net)
                  0.13    0.00    2.47 ^ clkbuf_6_38__f_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.33    0.38    2.85 ^ clkbuf_6_38__f_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
    13    0.19                           clknet_6_38__leaf_ext_clk (net)
                  0.33    0.00    2.85 ^ clkbuf_leaf_308_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.16    0.31    3.17 ^ clkbuf_leaf_308_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
    19    0.09                           clknet_leaf_308_ext_clk (net)
                  0.16    0.00    3.17 ^ _122817_/CLK (sky130_fd_sc_hd__dfxtp_2)
                  0.12    0.43    3.60 ^ _122817_/Q (sky130_fd_sc_hd__dfxtp_2)
     5    0.02                           soc0._308_ (net)
                  0.12    0.00    3.60 ^ hold1562/A (sky130_fd_sc_hd__dlygate4sd3_1)
                  0.08    0.60    4.20 ^ hold1562/X (sky130_fd_sc_hd__dlygate4sd3_1)
     1    0.01                           net8639 (net)
                  0.08    0.00    4.20 ^ repeater5039/A (sky130_fd_sc_hd__buf_6)
                  0.09    0.15    4.35 ^ repeater5039/X (sky130_fd_sc_hd__buf_6)
     3    0.04                           net5039 (net)
                  0.09    0.00    4.36 ^ hold1563/A (sky130_fd_sc_hd__dlygate4sd3_1)
                  0.38    0.84    5.19 ^ hold1563/X (sky130_fd_sc_hd__dlygate4sd3_1)
     5    0.04                           net8640 (net)
[... 85 lines cut ...]
                  0.03    0.00   17.21 v clkbuf_opt_122_1_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.04    0.13   17.34 v clkbuf_opt_122_1_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     1    0.02                           clknet_opt_122_1_ext_clk (net)
                  0.04    0.00   17.34 v clkbuf_opt_122_2_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.04    0.13   17.47 v clkbuf_opt_122_2_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     1    0.02                           clknet_opt_122_2_ext_clk (net)
                  0.04    0.00   17.47 v clkbuf_opt_122_3_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.04    0.13   17.60 v clkbuf_opt_122_3_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     1    0.02                           clknet_opt_122_3_ext_clk (net)
                  0.04    0.00   17.60 v clkbuf_opt_122_4_ext_clk/A (sky130_fd_sc_hd__clkbuf_8)
                  0.05    0.14   17.74 v clkbuf_opt_122_4_ext_clk/X (sky130_fd_sc_hd__clkbuf_8)
     1    0.03                           clknet_opt_122_4_ext_clk (net)
                  0.05    0.00   17.74 v soc0.bram.bram0.ram_0.memory_0/CLK (RAM512)
                         -0.25   17.49   clock uncertainty
                          0.14   17.63   clock reconvergence pessimism
                         -4.00   13.63   library setup time
                                 13.63   data required time
-----------------------------------------------------------------------------
                                 13.63   data required time
                                -14.25   data arrival time
-----------------------------------------------------------------------------
                                 -0.62   slack (VIOLATED)
```

Test case: [hold-fixup-setup-violation.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8800844/hold-fixup-setup-violation.tar.gz)

msaligane, 2022-05-30:
@minghungumich FYI.

antonblanchard, 2022-05-30:
Note that @jjcherry56 fixed a similar issue in https://github.com/The-OpenROAD-Project/OpenROAD/issues/1729. In that case we'd fail to fix the hold violations after consuming all the setup slack.

This looks a bit different since we resolved all the hold violations and there were no errors from the resizer. Perhaps we are pushing the design too close to zero setup slack, and going just over.

jjcherry56, 2022-05-31:
You have an interesting point that the commands do not account for. Hold fixing uses available setup slack, which isn't going to respect your setup margin. I need to add separate margins to handle it. In in any event hold fixing should not cause setup violations unless -allow_setup_violations is on.

jjcherry56, 2022-05-31:
I haven't unpacked this yet but I will point out that any design that takes 7000 buffers to fix hold violations probably has something seriously wrong with it. Examples that I have seen are broken liberty ram models, input/output delays that ignore the latency of the clock, or gated clocks that CTS is clueless about handling. If the hold violations were not so horrendous the setup slack would be safe.

antonblanchard, 2022-05-31:
I've been trying to build up the liberty files from STA output and I'm sure I'm getting a lot wrong. Do you see anything in the liberty files that are obviously bad?

An example of what I'm doing, for the a[*] input to the multiplier:

```
Startpoint: a[23] (input port clocked by clk)
Endpoint: _370_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clk (rise edge)
   0.00    0.00   clock network delay (propagated)
   4.00    4.00 v input external delay
   0.01    4.01 v a[23] (in)
   0.14    4.15 v input16/X (sky130_fd_sc_hd__buf_4)
   0.17    4.32 v repeater1436/X (sky130_fd_sc_hd__buf_6)
   0.20    4.52 v repeater1435/X (sky130_fd_sc_hd__buf_4)
   0.17    4.69 v U$$1510/X (sky130_fd_sc_hd__and2_1)
   0.34    5.03 v U$$1511/X (sky130_fd_sc_hd__a32o_4)
   0.22    5.25 v repeater604/X (sky130_fd_sc_hd__clkbuf_8)
   0.20    5.46 v repeater603/X (sky130_fd_sc_hd__clkbuf_8)
   0.27    5.73 v U$$1611/X (sky130_fd_sc_hd__a22o_1)
   0.19    5.91 ^ U$$1612/X (sky130_fd_sc_hd__xor2_1)
   0.47    6.38 v dadda_fa_0_71_2/SUM (sky130_fd_sc_hd__fa_1)
   0.45    6.83 v dadda_fa_1_71_8/COUT (sky130_fd_sc_hd__fa_1)
   0.54    7.37 v dadda_fa_2_72_3/SUM (sky130_fd_sc_hd__fa_1)
   0.57    7.93 ^ dadda_fa_3_72_3/SUM (sky130_fd_sc_hd__fa_1)
   0.46    8.39 v dadda_fa_4_72_2/SUM (sky130_fd_sc_hd__fa_1)
   0.42    8.81 v dadda_fa_5_72_1/COUT (sky130_fd_sc_hd__fa_1)
   0.51    9.31 v dadda_fa_6_73_0/SUM (sky130_fd_sc_hd__fa_1)
   0.40    9.71 v dadda_fa_7_73_0/COUT (sky130_fd_sc_hd__fa_1)
   0.00    9.71 v _370_/D (sky130_fd_sc_hd__dfxtp_1)
           9.71   data arrival time

  10.00   10.00   clock clk (rise edge)
   0.00   10.00   clock source latency
   0.18   10.18 ^ clk (in)
   0.20   10.38 ^ clkbuf_0_clk/X (sky130_fd_sc_hd__clkbuf_16)
   0.15   10.53 ^ clkbuf_2_1_0_clk/X (sky130_fd_sc_hd__clkbuf_8)
   0.17   10.70 ^ clkbuf_3_2__f_clk/X (sky130_fd_sc_hd__clkbuf_16)
   0.00   10.70 ^ _370_/CLK (sky130_fd_sc_hd__dfxtp_1)
  -0.25   10.45   clock uncertainty
   0.00   10.45   clock reconvergence pessimism
  -0.11   10.34   library setup time
          10.34   data required time
---------------------------------------------------------
          10.34   data required time
          -9.71   data arrival time
---------------------------------------------------------
           0.63   slack (MET)
```

hold delay: 4ns
setup delay: 10 (next clock) - 4 (hold delay) - 0.63 (excess setup slack) =  5.37 (rounded up to 5.5)

```
            timing() {
                timing_type : setup_rising;
                related_pin : "clk";
                rise_constraint(scalar) {
                    values("5.5000");
                }
                fall_constraint(scalar) {
                    values("5.5000");
                }
            }

            timing() {
                timing_type : hold_rising;
                related_pin : "clk";
                rise_constraint(scalar) {
                    values("4.0000");
                }
                fall_constraint(scalar) {
                    values("4.0000");
                }
            }

```

antonblanchard, 2022-05-31:
Makes sense, let me package everything up and send it your way.

donnie-j, 2022-06-02:
> I've been trying to build up the liberty files from STA output...

Anton, are you writing a tool to do this?  This is IMHO one absolutely key piece of infrastructure that is missing.

jjcherry56, 2022-06-02:
Liberty files are inadequate to support hierarchical flows. They have no way to express boundary parasitics or sdc exceptions that cross the block boundary or transparent latch behavior. ILM models were developed for this reason
but even those have issues and have been supplanted in commercial flows. It is an extremely difficult problem.

maliberty, 2022-06-02:
Even a basic ILM generator would be better than nothing.  We don't have SI or CCS to deal with yet so I think it would suffice for most skywater users.

jjcherry56, 2022-06-03:
fixed in ed7cb4e7e rsz repair_hold

closed 2022-06-03 by jjcherry56