The-OpenROAD-Project/OpenROAD #1933: After run the flow successfully, the DRC is not clean

Elon-J, 2022-06-04:
When I try to increase the output load ability of my block, after passing the OpenRoad flow successfully, the DRC is not clean. What should I adjust?
Attach my block repository:
https://github.com/Elon-J/OpenROAD-flow-scripts/tree/master/flow/designs/sky130hd/NON_CLK_GEN
Attach my magic DRC report and layout DRC report
[sky130_drc.txt](https://github.com/The-OpenROAD
[6_final_drc.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/8837213/6_final_drc.txt)
-Project/OpenROAD/files/8837212/sky130_drc.txt)

Elon-J, 2022-06-04:
[image]

Elon-J, 2022-06-04:
I also have try many verisons including adjust different core areas.

vijayank88, 2022-06-04:
@Elon-J 
Please execute following commands from `flow` directory and attach zip file.
```
make detail_route_issue
```

Elon-J, 2022-06-04:
OK

Elon-J, 2022-06-04:
I have updated my first comment.

vijayank88, 2022-06-04:
@Elon-J 
Detail routing completed with 0 DRC errors.
```
[INFO DRT-0195] Start 57th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:00:00, memory = 149.02 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:00:00, memory = 149.02 (MB).
    Completing 30% with 0 violations.
    elapsed time = 00:00:00, memory = 149.03 (MB).
    Completing 40% with 0 violations.
    elapsed time = 00:00:00, memory = 149.03 (MB).
[INFO DRT-0199]   Number of violations = 0.
```
Its something error coming from KLayout.

Please try to load and check those overlapping error.

Elon-J, 2022-06-04:
@vijayank88 , Thank you very much! Let me check it!

Elon-J, 2022-06-04:
@vijayank88 , By the way, actually I also check it using magic, there are also some DRC errors.

msaligane, 2022-06-04:
@Elon-J can you add the drc report and maybe zoomed in screenshots. That would be helpful to understand what is wrong.

vijayank88, 2022-06-04:
@msaligane please check this slack channel conversation: https://skywater-pdk.slack.com/archives/C0161A4A59V/p1654326142362839

msaligane, 2022-06-04:
@vijayank88 thanks! 
@Elon-J what is your fix for the buf_16?

Elon-J, 2022-06-04:
@msaligane , I just try to increase the drive ability to generate a new CLK, and this problem has been fixed.

maliberty, 2022-06-13:
Is this resolved?

closed 2022-06-13 by Elon-J