The-OpenROAD-Project/OpenROAD #1934: cmake does not check for readline.h header package

titan73, 2022-06-05:
I installed libreadline-dev package in ubuntu 22.04 to fix the problem but that would be better to catch it during cmake.

[  6%] Building CXX object third-party/abc/CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o
cd /home/titan/OpenROAD/build/third-party/abc && /usr/bin/c++  -I/home/titan/OpenROAD/third-party/abc/src -O3 -DNDEBUG -Wall -Wno-array-bounds -Wno-nonnull -Wno-maybe-uninitialized -Wno-format-overflow -Wno-unused-variable -Wno-unused-function -Wno-write-strings -Wno-sign-compare -Wno-deprecated -Wno-register -Wno-format -DLIN64 -DSIZEOF_VOID_P=8 -DSIZEOF_LONG=8 -DSIZEOF_INT=4 -DABC_NAMESPACE=abc -fpermissive -DABC_USE_CUDD=1 -DABC_USE_READLINE -DABC_USE_PTHREADS -Wno-unused-but-set-variable -std=c++17 -MD -MT third-party/abc/CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o -MF CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o.d -o CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o -c /home/titan/OpenROAD/third-party/abc/src/base/main/mainUtils.c
/home/titan/OpenROAD/third-party/abc/src/base/main/mainUtils.c:25:10: fatal error: readline/readline.h: Aucun fichier ou dossier de ce type
   25 | #include <readline/readline.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [third-party/abc/CMakeFiles/libabc.dir/build.make:2109 : third-party/abc/CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o] Erreur 1
make[2] : on quitte le répertoire « /home/titan/OpenROAD/build »
make[1]: *** [CMakeFiles/Makefile2:1134 : third-party/abc/CMakeFiles/libabc.dir/all] Erreur 2
make[1] : on quitte le répertoire « /home/titan/OpenROAD/build »
make: *** [Makefile:139 : all] Erreur 2

vvbandeira, 2022-06-21:
@titan73 the issue was with the third-party ABC submodule. We have disabled the use of readline for ABC (since we use the C++ API) as per their documentation. If the problem persists, please file an issue with the upstream repo. Thanks.

closed 2022-06-21 by vvbandeira

titan73, 2022-07-05:
Morover defining the variable must apparently be done before "add_subdirectory(abc)".
This works:

if (NOT USE_SYSTEM_ABC)

  set(ABC_USE_NAMESPACE "abc")

  # abc does not have "platform" support for macos so force it to use stdint.h
  if (CMAKE_OSX_SYSROOT)
     set(ABC_USE_STDINT_H 1)
  endif()

  set(READLINE_FOUND FALSE)
  add_subdirectory(abc)

endif()

vvbandeira, 2022-07-05:
> Morover defining the variable must apparently be done before "add_subdirectory(abc)".

Can you try on your system and if it works, please file a PR.

> Is the upstream abc repo "https://github.com/berkeley-abc/abc"?

Yes. If there are changes upstream to be added to OpenROAD, please file an issue.

titan73, 2022-07-05:
Yep sure. Will do.

PR #1973 (2022-06-21): Disable readline includes on abc code
Closes #1934