The-OpenROAD-Project/OpenROAD #1940: DRT: met3 min area violation

antonblanchard, 2022-06-06:
I'm seeing a single met3 minimum area violation when taping out Microwatt using a recent OpenROAD (commit 0b8b7ae255f8fbbbefa57d443949b84e73eed757).

The issue is at 1147.305, 965.775

[image]

The DRC issue was found with magic, but I was able to reproduce it with the klayout DRC deck.

Test case: [met3_min_area.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/8842337/met3_min_area.tar.gz)

maliberty, 2022-06-06:
Does the router show any drcs?

antonblanchard, 2022-06-06:
> Does the router show any drcs?

It completes without any DRC errors:

```
[INFO DRT-0199]   Number of violations = 0.
```

Stephanommg, 2022-06-14:
The fix is on master branch now.

closed 2022-06-14 by Stephanommg