The-OpenROAD-Project/OpenROAD #1943: When I try to connect every macro block into a big one I meet this problem

Elon-J, 2022-06-07:
[image]

How can I set the pdn.cfg correctly to make the VPWR and VGND of all my macro blocks connect to the power ring?

msaligane, 2022-06-07:
@Elon-J Your GitHub issue has no details and no test case. Jianwei: Please provide as much details as possible so others could help you!

arlpetergadfort, 2022-06-08:
@msaligane @Elon-J based on the file it appears they are using the deprecated version of PDNgen, so they should probably start by updating to the current version

maliberty, 2022-06-13:
@Elon-J do you still need help?  There is nothing further we can do without action from you.

closed 2022-06-13 by Elon-J