The-OpenROAD-Project/OpenROAD #1945: Tapcell: Close Source PDK Feature Request

QuantamHD, 2022-06-08:
I'm using a PDK + Standard Cell Library that has special end cap cells for both the top, bottom, left and right cells. It also has corner cells for each corner of the macro rectangles (TR, TL, BR, BL). It doesn't seem like tapcell supports this. Can we add this feature.

QuantamHD, 2022-06-08:
So upon closer inspection it looks like the cells in the PDK are technically named differently, but are really just rotations of each other. I feel like we should probably support the differently named cells, but it's not incorrect per say since OpenROAD seems to be doing to the rotating.

maliberty, 2025-08-25:
I think this has been resolved - reopen if you disagree

closed 2025-08-25 by maliberty